realsenseai/realsense-ros #131: Auto exposure not working in r200 cameras

tulku, 2016-10-18:
### System Configuration

| Version               | Your Configuration   |
|:--------------------- |:---------------------|
| Operating System      | Ubuntu 16.04                |
| Kernel                | 4.4.0-38                |
| Backend               | v4l                    |
| ROS                   |  kinetic                   |
| ROS RealSense         |  1.5.0               |
| librealsense          | 1.11.0                |
| Camera Type-Firmware  | r200-1.0.72.06            |

---
### Expected Behavior

When setting auto_exposure to 1, the camera should adjust the exposure value so that the image is well exposed (avoid under/over exposure). Ideally, getting rid of holes due to too much IR reflections.

### Actual Behavior

Enabling the auto_exposure value does not change the exposure setting of the camera.

### Steps to Reproduce

To produce and under exposed image: Set auto exposure to 0, and manually adjust the exposure value to 50, you should see a dull IR image. Then re-enable auto_exposure, like in the picture:
[image]

To produce and over exposed image: Set auto exposure to 0, and manually adjust the exposure value to 164, you should see a very bright IR image. Then re-enable auto_exposure:
[image]

If I understand auto_exposure correctly, whenever it is active, it should adjust the exposure to whatever is the best value at that moment, so the previously, manually set values, should have no effect when re enabling auto_exposure.

For the pictures in the issue, the camera is facing a wall, at a distance between 80 to 90cm.

rjingar, 2016-10-31:
Hi @tulku ,

Yes we are able to reproduce it. We will work on fixing this issue and will notify once it is fixed.
Thank you for filling an issue!

stwirth, 2016-11-21:
I can reproduce the same issue with an LR200. The `r200_lr_auto_exposure_enabled` setting doesn't seem to have any effect.

| Version               | Your Configuration   |
|:--------------------- |:---------------------|
| Operating System      | Ubuntu 14.04.5 LTS                |
| Kernel                | 4.4.0-040400-generic                |
| ROS                   | indigo                    |
| ROS RealSense         | 1.5.0                |
| librealsense          | 1.11.0               |
| Camera Type-Firmware  | LR200-2.0.71.24            |

tulku, 2016-12-15:
I tested this again with a downgraded firmware to see if it solved the problem:

* os: Ubuntu 16.04
* librealsense: 1.11.1
* realsense-camera: 1.6.0
* camera type - firmware: r200-1.0.72.04

But autoexposure is still not working.

pfuersattel, 2017-02-08:
Hi,
Is this still an issue in the 1.9.7. release?

evanbeard, 2017-02-14:
This still does not seem to work with the latest release

mdhorn, 2017-04-03:
Resolved in release 1.8.0
Available now by pulling tag 1.8.0 and building from sources.
Released to ROS Build Farm.  Available in [Shadow Repository](http://wiki.ros.org/ShadowRepository) once [PR 14443 for Indigo](https://github.com/ros/rosdistro/pull/14443) and [PR 14444 for Kinetic](https://github.com/ros/rosdistro/pull/14444l) are merged.
Public ROS repository packages available after next sync (1-3 weeks) by [OSRF](https://www.osrfoundation.org/osrf-projects/).

closed 2017-04-03 by mdhorn