realsenseai/realsense-ros #135: F200 Depth Image with multiple images

rarrais, 2016-10-19:
### System Configuration

| Version               | Your Configuration   |
|:--------------------- |:---------------------|
| Operating System      | Ubuntu 14.04.5 LTS                |
| Kernel                |4.4.0-42-generic       |
| Backend               | video4linux                 |
| ROS                   | indigo  |
| ROS RealSense         | 1.4.0          |
| librealsense          | 0.9.1               |
| Camera Type-Firmware  | 2.60.0.0           |

---

### Expected Behavior

We were expecting a depth image that were not multiplied as shown in the "Actual Behaviour". Also, the PointCloud2 points do not appear correctly. 

We are using the default configuration and the default launch file. We also tried to do some testing with turning on the point cloud generation and the PointCloud2 generated point cloud did not appeared right also. We suspect that this is related with the odd depth image we are observing. 

Can someone give us some input on what we might be doing wrong?

### Actual Behavior

Check the screenshots 

[image]

### Steps to Reproduce

After installation, 

`roscore`

`roslaunch realsense_camera f200_nodelet_default.launch`

`rviz`

reaganlo, 2016-10-28:
@rarrais We were not able to reproduce this issue. One reason could be that "librealsense" and its corresponding udev rules and patches were not installed/applied correctly. Could be please try installing the latest librealsense and realsense_camera packages by following the instructions at http://wiki.ros.org/realsense_camera

Also, we recommend not to use the native pointcloud of the nodelet. Instead, generate the pointcloud using **rgbd_launch** package. I am attaching a sample launch file for the F200 camera that generates an rgbd pointcloud. [f200_nodelet_rgbd.launch.txt] (https://github.com/intel-ros/realsense/files/559485/f200_nodelet_rgbd.launch.txt)

After launching the attached file:

- For depth image, subscribe to **/camera/depth/image_raw**
- For color image, subscribe to **/camera/rgb/image_raw**
- For rgbd pointcloud, subscribe to **/camera/depth_registered/points**

Also attaching a screenshot of how the images typically look like in rviz
[image]

rarrais, 2016-11-03:
Dear @reaganlo ,

We managed to configure the camera following your advice. The problem was indeed related with a bad installation of "librealsense". Our point cloud in RVIz looks good now!

Thank you very much for your assistance. 

(You can close the issue)

Best regards,
Rafael Arrais

closed 2017-02-13 by reaganlo

weili7, 2020-03-05:
@rarrais do u mind to share the system configuration for your success build? Having similar problem right now. Thank

rarrais, 2020-03-06:
@weili7 I apologize, but I do not have this system nor its configuration anymore. The experiment was done back in 2016. Nevertheless, I remember managing to have it working following the suggestion on this thread.