realsenseai/realsense-ros #136: 'RS_CAPABILITIES_ADAPTER_BOARD' was not declared in this scope

jacksonkr, 2016-10-21:
ros-indigo, ubuntu 14.04 arm, librealsense installed

```
$ catkin_make
Base path: /home/ubuntu/catkin_ws
Source space: /home/ubuntu/catkin_ws/src
Build space: /home/ubuntu/catkin_ws/build
Devel space: /home/ubuntu/catkin_ws/devel
Install space: /home/ubuntu/catkin_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/ubuntu/catkin_ws/build"
####
####
#### Running command: "make -j1 -l1" in "/home/ubuntu/catkin_ws/build"
####
[  3%] Built target realsense_camera_gencfg
[  3%] Built target _realsense_camera_generate_messages_check_deps_IsPowered
[  3%] Built target _realsense_camera_generate_messages_check_deps_CameraConfiguration
[  3%] Built target _realsense_camera_generate_messages_check_deps_ForcePower
[  3%] Built target _realsense_camera_generate_messages_check_deps_IMUInfo
[  3%] Built target std_msgs_generate_messages_lisp
[  3%] Built target _realsense_camera_generate_messages_check_deps_SetPower
[  3%] Built target _realsense_camera_generate_messages_check_deps_GetIMUInfo
[  8%] Built target realsense_camera_generate_messages_lisp
[  8%] Built target std_msgs_generate_messages_cpp
[ 12%] Built target realsense_camera_generate_messages_cpp
[ 12%] Built target std_msgs_generate_messages_py
[ 17%] Built target realsense_camera_generate_messages_py
[ 17%] Built target realsense_camera_generate_messages
[ 17%] Built target topic_tools_generate_messages_py
[ 17%] Built target nodelet_generate_messages_lisp
[ 17%] Built target nodelet_generate_messages_cpp
[ 17%] Built target dynamic_reconfigure_generate_messages_cpp
[ 17%] Built target dynamic_reconfigure_generate_messages_lisp
[ 17%] Built target dynamic_reconfigure_generate_messages_py
[ 17%] Built target nodelet_generate_messages_py
[ 17%] Built target roscpp_generate_messages_cpp
[ 17%] Built target dynamic_reconfigure_gencfg
[ 17%] Built target roscpp_generate_messages_lisp
[ 17%] Built target roscpp_generate_messages_py
[ 17%] Built target sensor_msgs_generate_messages_lisp
[ 17%] Built target sensor_msgs_generate_messages_py
[ 17%] Built target rosgraph_msgs_generate_messages_py
[ 17%] Built target rosgraph_msgs_generate_messages_lisp
[ 17%] Built target sensor_msgs_generate_messages_cpp
[ 17%] Built target rosgraph_msgs_generate_messages_cpp
[ 17%] Built target bond_generate_messages_cpp
[ 17%] Built target bond_generate_messages_lisp
[ 17%] Built target bond_generate_messages_py
[ 17%] Built target tf_generate_messages_cpp
[ 17%] Built target tf_generate_messages_py
[ 17%] Built target tf_generate_messages_lisp
[ 17%] Built target geometry_msgs_generate_messages_cpp
[ 17%] Built target geometry_msgs_generate_messages_lisp
[ 17%] Built target geometry_msgs_generate_messages_py
[ 17%] Built target pcl_msgs_generate_messages_cpp
[ 17%] Built target actionlib_msgs_generate_messages_cpp
[ 17%] Built target topic_tools_generate_messages_lisp
[ 17%] Built target tf2_msgs_generate_messages_lisp
[ 17%] Built target actionlib_generate_messages_lisp
[ 17%] Built target actionlib_generate_messages_cpp
[ 17%] Built target actionlib_generate_messages_py
[ 17%] Built target actionlib_msgs_generate_messages_lisp
[ 17%] Built target actionlib_msgs_generate_messages_py
[ 17%] Built target pcl_ros_gencfg
[ 17%] Built target topic_tools_generate_messages_cpp
[ 17%] Built target pcl_msgs_generate_messages_py
[ 17%] Built target pcl_msgs_generate_messages_lisp
[ 17%] Built target tf2_msgs_generate_messages_cpp
[ 17%] Built target tf2_msgs_generate_messages_py
[ 18%] Building CXX object realsense/realsense_camera/CMakeFiles/realsense_camera_nodelet.dir/src/base_nodelet.cpp.o
/home/ubuntu/catkin_ws/src/realsense/realsense_camera/src/base_nodelet.cpp: In member function 'virtual std::vector<int> realsense_camera::BaseNodelet::listCameras(int)':
/home/ubuntu/catkin_ws/src/realsense/realsense_camera/src/base_nodelet.cpp:241:43: error: 'RS_CAPABILITIES_ADAPTER_BOARD' was not declared in this scope
       if (rs_supports(rs_detected_device, RS_CAPABILITIES_ADAPTER_BOARD, &rs_error_))
                                           ^
/home/ubuntu/catkin_ws/src/realsense/realsense_camera/src/base_nodelet.cpp:241:84: error: 'rs_supports' was not declared in this scope
       if (rs_supports(rs_detected_device, RS_CAPABILITIES_ADAPTER_BOARD, &rs_error_))
                                                                                    ^
/home/ubuntu/catkin_ws/src/realsense/realsense_camera/src/base_nodelet.cpp:243:74: error: 'RS_CAMERA_INFO_ADAPTER_BOARD_FIRMWARE_VERSION' was not declared in this scope
         const char * adapter_fw = rs_get_device_info(rs_detected_device, RS_CAMERA_INFO_ADAPTER_BOARD_FIRMWARE_VERSION,
                                                                          ^
/home/ubuntu/catkin_ws/src/realsense/realsense_camera/src/base_nodelet.cpp:244:23: error: 'rs_get_device_info' was not declared in this scope
             &rs_error_);
                       ^
/home/ubuntu/catkin_ws/src/realsense/realsense_camera/src/base_nodelet.cpp:249:43: error: 'RS_CAPABILITIES_MOTION_EVENTS' was not declared in this scope
       if (rs_supports(rs_detected_device, RS_CAPABILITIES_MOTION_EVENTS, &rs_error_))
                                           ^
/home/ubuntu/catkin_ws/src/realsense/realsense_camera/src/base_nodelet.cpp:249:84: error: 'rs_supports' was not declared in this scope
       if (rs_supports(rs_detected_device, RS_CAPABILITIES_MOTION_EVENTS, &rs_error_))
                                                                                    ^
/home/ubuntu/catkin_ws/src/realsense/realsense_camera/src/base_nodelet.cpp:251:80: error: 'RS_CAMERA_INFO_MOTION_MODULE_FIRMWARE_VERSION' was not declared in this scope
         const char * motion_module_fw = rs_get_device_info(rs_detected_device, RS_CAMERA_INFO_MOTION_MODULE_FIRMWARE_VERSION,
                                                                                ^
/home/ubuntu/catkin_ws/src/realsense/realsense_camera/src/base_nodelet.cpp:252:23: error: 'rs_get_device_info' was not declared in this scope
             &rs_error_);
                       ^
make[2]: *** [realsense/realsense_camera/CMakeFiles/realsense_camera_nodelet.dir/src/base_nodelet.cpp.o] Error 1
make[1]: *** [realsense/realsense_camera/CMakeFiles/realsense_camera_nodelet.dir/all] Error 2
make: *** [all] Error 2
Invoking "make -j1 -l1" failed
```

mdhorn, 2016-10-21:
Please provide details about version of librealsense and realsense_camera.

| Version               | Your Configuration   |
|:--------------------- |:---------------------|
| Kernel                | ?-?-?                |
| Backend               | ?                    |
| ROS RealSense         | ?.?.?                |
| librealsense          | ?.?.?                |

jacksonkr, 2016-10-21:
This is all happening on a Jetson TK1 running the grinch kernel

Kernel: 3.10.40-grinch-21.3.4
Backend: video4linux
ROS Realsense: v0.9.2 (tried most recent as well but ultimately reverted)
librealsense: 1.4.0 (tried most recent as well but ultimately reverted)

Also, I noticed that if I try running `cpp-capture` I receive
```
RealSense error calling rs_create_context(api_version:11100):
    api version mismatch
```

mdhorn, 2016-10-21:
This seems to be duplicate of https://github.com/intel-ros/realsense/issues/128

Is there really a separate issue here?

jacksonkr, 2016-10-21:
Good eye, these seem to have enough in common that I can close this ticket.

closed 2016-10-21 by jacksonkr