realsenseai/realsense-ros #139: linux-headers-4.4.0-040400-generic is not supported

jacksonkr, 2016-10-26:
### System Configuration
| Version               | Your Configuration   |
|:--------------------- |:---------------------|
| Operating System      | Ubuntu 14.04                |
| Kernel                | 4.4.0-040400-generic                |
| Backend               | video4linux                    |
| ROS                   | indigo                    |
| ROS RealSense         | 1.5.0-0trusty-20161020-182325-0700                |
| librealsense          | v0.9.2                |
| Camera Type-Firmware  | r200-?ros

---

### Expected Behavior
Install `ros-indigo-realsense-camera` using apt-get

### Actual Behavior
Install from apt-get is unsuccessful.

```
$ sudo apt-get install ros-indigo-realsense-camera 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following extra packages will be installed:
  ros-indigo-librealsense
The following NEW packages will be installed:
  ros-indigo-librealsense ros-indigo-realsense-camera
0 upgraded, 2 newly installed, 0 to remove and 0 not upgraded.
Need to get 0 B/481 kB of archives.
After this operation, 2,451 kB of additional disk space will be used.
Do you want to continue? [Y/n] y
Selecting previously unselected package ros-indigo-librealsense.
(Reading database ... 363126 files and directories currently installed.)
Preparing to unpack .../ros-indigo-librealsense_1.11.0-1trusty-20161020-181438-0700_amd64.deb ...
Unpacking ros-indigo-librealsense (1.11.0-1trusty-20161020-181438-0700) ...
Selecting previously unselected package ros-indigo-realsense-camera.
Preparing to unpack .../ros-indigo-realsense-camera_1.5.0-0trusty-20161020-182325-0700_amd64.deb ...
Unpacking ros-indigo-realsense-camera (1.5.0-0trusty-20161020-182325-0700) ...
Setting up ros-indigo-librealsense (1.11.0-1trusty-20161020-181438-0700) ...

Creating symlink /var/lib/dkms/uvcvideo/1.1.1-3-realsense/source ->
                 /usr/src/uvcvideo-1.1.1-3-realsense

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Running the pre_build script:

dkms: Pre-build script started...

Downloading kernel sources...
Reading package lists...
Building dependency tree...
Reading state information...
Picking 'linux' as source package instead of 'linux-image-4.4.0-040400-generic'
NOTICE: 'linux' packaging is maintained in the 'Git' version control system at:
http://kernel.ubuntu.com/git-repos/ubuntu/ubuntu-trusty.git
Need to get 126 MB of source archives.
Get:1 http://us.archive.ubuntu.com/ubuntu/ trusty-updates/main linux 3.13.0-100.147 (dsc) [8,127 B]
Get:2 http://us.archive.ubuntu.com/ubuntu/ trusty-updates/main linux 3.13.0-100.147 (tar) [116 MB]
Get:3 http://us.archive.ubuntu.com/ubuntu/ trusty-updates/main linux 3.13.0-100.147 (diff) [9,323 kB]
gpgv: Signature made Tue 18 Oct 2016 12:40:11 PM EDT using RSA key ID CBEECEA3
gpgv: Can't check signature: public key not found
dpkg-source: warning: failed to verify signature on ./linux_3.13.0-100.147.dsc
dpkg-source: info: extracting linux in linux-3.13.0
dpkg-source: info: unpacking linux_3.13.0.orig.tar.gz
dpkg-source: info: applying linux_3.13.0-100.147.diff.gz
dpkg-source: info: upstream files that have been modified: 
Fetched 126 MB in 1min 12s (1,737 kB/s)
INFO: No Intel RealSense(TM) cameras are currently supported.
Patching uvcvideo sources...
patching file drivers/media/usb/uvc/Makefile
patching file drivers/media/usb/uvc/uvc_driver.c
Hunk #1 succeeded at 122 (offset -26 lines).
patching file drivers/media/usb/uvc/uvcvideo.h
Hunk #1 succeeded at 104 (offset -15 lines).
patching file include/uapi/linux/videodev2.h
Hunk #1 succeeded at 430 with fuzz 2 (offset -191 lines).
  HOSTCC  scripts/basic/fixdep
  HOSTCC  scripts/kconfig/conf.o
  SHIPPED scripts/kconfig/zconf.tab.c
  SHIPPED scripts/kconfig/zconf.lex.c
  SHIPPED scripts/kconfig/zconf.hash.c
  HOSTCC  scripts/kconfig/zconf.tab.o
  HOSTLD  scripts/kconfig/conf
scripts/kconfig/conf --olddefconfig Kconfig
.config:934:warning: symbol value 'm' invalid for BRIDGE_NETFILTER
.config:1423:warning: symbol value 'm' invalid for OPENVSWITCH_GRE
.config:1424:warning: symbol value 'm' invalid for OPENVSWITCH_VXLAN
.config:1685:warning: symbol value 'm' invalid for NFC_NCI_SPI
.config:2707:warning: symbol value 'm' invalid for STMMAC_PLATFORM
.config:4421:warning: symbol value 'm' invalid for MFD_WM8994
.config:5536:warning: symbol value 'm' invalid for SND_HDA_CODEC_REALTEK
.config:5537:warning: symbol value 'm' invalid for SND_HDA_CODEC_ANALOG
.config:5538:warning: symbol value 'm' invalid for SND_HDA_CODEC_SIGMATEL
.config:5539:warning: symbol value 'm' invalid for SND_HDA_CODEC_VIA
.config:5540:warning: symbol value 'm' invalid for SND_HDA_CODEC_HDMI
.config:5541:warning: symbol value 'm' invalid for SND_HDA_CODEC_CIRRUS
.config:5542:warning: symbol value 'm' invalid for SND_HDA_CODEC_CONEXANT
.config:5543:warning: symbol value 'm' invalid for SND_HDA_CODEC_CA0110
.config:5544:warning: symbol value 'm' invalid for SND_HDA_CODEC_CA0132
.config:5546:warning: symbol value 'm' invalid for SND_HDA_CODEC_CMEDIA
.config:5547:warning: symbol value 'm' invalid for SND_HDA_CODEC_SI3054
.config:5548:warning: symbol value 'm' invalid for SND_HDA_GENERIC
.config:6634:warning: symbol value 'm' invalid for COMEDI_PCI_DRIVERS
.config:6689:warning: symbol value 'm' invalid for COMEDI_PCMCIA_DRIVERS
.config:6697:warning: symbol value 'm' invalid for COMEDI_USB_DRIVERS
.config:7105:warning: symbol value 'm' invalid for IIO_BUFFER_CB
.config:7150:warning: symbol value 'm' invalid for LP8788_ADC
#
# configuration written to .config
#
scripts/kconfig/conf --silentoldconfig Kconfig
  WRAP    arch/x86/include/generated/asm/clkdev.h
  HOSTCC  scripts/genksyms/genksyms.o
  SHIPPED scripts/genksyms/lex.lex.c
  SHIPPED scripts/genksyms/keywords.hash.c
  SHIPPED scripts/genksyms/parse.tab.h
  HOSTCC  scripts/genksyms/lex.lex.o
  SHIPPED scripts/genksyms/parse.tab.c
  HOSTCC  scripts/genksyms/parse.tab.o
  HOSTLD  scripts/genksyms/genksyms
  CC      scripts/mod/empty.o
  HOSTCC  scripts/mod/mk_elfconfig
  MKELF   scripts/mod/elfconfig.h
  CC      scripts/mod/devicetable-offsets.s
  GEN     scripts/mod/devicetable-offsets.h
  HOSTCC  scripts/mod/file2alias.o
  HOSTCC  scripts/mod/modpost.o
  HOSTCC  scripts/mod/sumversion.o
  HOSTLD  scripts/mod/modpost
  HOSTCC  scripts/selinux/genheaders/genheaders
  HOSTCC  scripts/selinux/mdp/mdp
  HOSTCC  scripts/kallsyms
  HOSTCC  scripts/conmakehash
  HOSTCC  scripts/recordmcount
  HOSTCC  scripts/sortextable
  HOSTCC  scripts/asn1_compiler
  SYSHDR  arch/x86/syscalls/../include/generated/uapi/asm/unistd_32.h
  SYSHDR  arch/x86/syscalls/../include/generated/uapi/asm/unistd_64.h
  SYSHDR  arch/x86/syscalls/../include/generated/uapi/asm/unistd_x32.h
  SYSTBL  arch/x86/syscalls/../include/generated/asm/syscalls_32.h
  SYSHDR  arch/x86/syscalls/../include/generated/asm/unistd_32_ia32.h
  SYSHDR  arch/x86/syscalls/../include/generated/asm/unistd_64_x32.h
  SYSTBL  arch/x86/syscalls/../include/generated/asm/syscalls_64.h
  HOSTCC  arch/x86/tools/relocs_32.o
  HOSTCC  arch/x86/tools/relocs_64.o
  HOSTCC  arch/x86/tools/relocs_common.o
  HOSTLD  arch/x86/tools/relocs
  CHK     include/config/kernel.release
  UPD     include/config/kernel.release
  CHK     include/generated/uapi/linux/version.h
  UPD     include/generated/uapi/linux/version.h
  CHK     include/generated/utsrelease.h
  UPD     include/generated/utsrelease.h
  CC      kernel/bounds.s
  GEN     include/generated/bounds.h
  CC      arch/x86/kernel/asm-offsets.s
  GEN     include/generated/asm-offsets.h
  CALL    scripts/checksyscalls.sh

Building module:
cleaning build area....
make KERNELRELEASE=4.4.0-040400-generic -C /var/lib/dkms/uvcvideo/1.1.1-3-realsense/build/linux-src M=drivers/media/usb/uvc/....(bad exit status: 2)
ERROR (dkms apport): kernel package linux-headers-4.4.0-040400-generic is not supported
Error! Bad return status for module build on kernel: 4.4.0-040400-generic (x86_64)
Consult /var/lib/dkms/uvcvideo/1.1.1-3-realsense/build/make.log for more information.
WARNING: No Intel RealSense(TM) cameras are supported!
    Visit https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md
    to enable Intel RealSense (TM) camera types.
Setting up ros-indigo-realsense-camera (1.5.0-0trusty-20161020-182325-0700) ...

### Steps to Reproduce

`sudo apt-get install --reinstall ros-indigo-librealsense`

mdhorn, 2016-10-26:
@jacksonkr The kernel you are using is not support for ROS, only the kernels which also provide source Debian packages. The one you have there is from the old librealsense directions and should be removed.

Have you tried just installing the prebuilt packages?
Do you have a specific problem where you need to use sources instead of the ROS Debian packages?

The only prerequisite for using the ROS Debian packages is ensure you have enabled the Debian sources for the your kernel.
See: 	http://wiki.ros.org/librealsense#Enable_Kernel_Sources

jacksonkr, 2016-10-27:
@mdhorn 

Mark, yes, I doubled back and cleaned out EVERY reference of *realsense* I could find on one of my machines with a supported kernel. I'll close this assuming the debian packages will work just fine on my other machine after I restore an old kernel.

Cheers

closed 2016-10-27 by jacksonkr

jacksonkr, 2016-10-28:
FYI for future visitors. Use apt-get to move to a different kernel.

eg. `sudo apt-get install linux-image-` then hit tab a couple times to see a completely listing. My advice it stick with when looks like a popular image

Then after installing the image, make sure you install the matching headers because librealsense will need that and realsense-camera needs librealsense.

`sudo apt-get install linux-headers-[kernel]`
`sudo apt-get install linux-headers-4.4.0-45-generic`

You may need to `sudo apt-get install --reinstall ros-[distro]-librealsense` before being able to correctly install realsense-camera

mdhorn, 2016-10-28:
You might also consider using the latest kernel under 14.04 by upgrading to the latest dist 14.0.4.5.
`sudo apt-get update`
`sudo apt-get dist-upgrade`
14.04.5 switched to using the latest Xenial kernel (linux-image-generic-lts-xenial).
This just ensures you are getting security patches and other critical kernel updates for your system.