realsenseai/realsense-ros #141: Static transformations missing when running the multiple camera example

tulku, 2016-10-26:
### System Configuration

| Version               | Your Configuration   |
|:--------------------- |:---------------------|
| Operating System      | Ubuntu 16.04.1 LTS                |
| Kernel                | 4.4.0-45-generic #66-Ubuntu SMP               |
| Backend               | v4l                    |
| ROS                   | kinetic                    |
| ROS RealSense         | 1.5.0                |
| librealsense          | 1.11.0                |
| Camera Type-Firmware  | r200-1.0.72.06            |

### Expected Behavior

The static transformations for both cameras should be found in the tf tree.

### Actual Behavior

When running the `r200_nodelet_multiple_cameras.launch` only one of the cameras is able to publish it's static transforms. This probably happens because the nodelet manager will have to latched publishers to tf_static, one per camera, and that does not work correctly, as explained here:

http://wiki.ros.org/tf2/Migration#Addition_of_.2BAC8-tf_static_topic

### Steps to Reproduce

Start the `r200_nodelet_multiple_cameras.launch` and check the tf tree. This can be done doing:

```
$ rosrun tf tf_echo camera2_link camera2_depth_optical_frame
$ rosrun tf tf_echo camera1_link camera1_depth_optical_frame
```
You will see the camera2 frames being correctly resolved but the camera1 frame missing.

tulku, 2016-10-26:
This is related (or even a duplicate of) https://github.com/intel-ros/realsense/issues/120

reaganlo, 2016-10-28:
@tulku As you figured out, since this is a duplicate of #120, I am closing this one. Please look out for updates on the other thread.

closed 2016-10-28 by reaganlo