realsenseai/realsense-ros #142: fails to be be launched

robotpilot, 2016-10-27:
### System Configuration
Please complete Your Configuration detail below. Refer to the BKC at [Configuration](../realsense_camera/README.md#configuration).

| Version               | Your Configuration   |
|:--------------------- |:---------------------|
| Operating System      | Ubuntu 16.04.1 LTS                |
| Kernel                | 4.4.0-31-generic               |
| Backend               | ?                    |
| ROS                   | kinetic                  |
| ROS RealSense         | 1.6.0               |
| librealsense          | 1.11.0             |
| Camera Type-Firmware  | Name: Intel RealSense ZR300, Camera FW: 2.0.71.18, Adapter FW: 1.21.0.0           |

### Expected Behavior

launched

### Actual Behavior

https://gist.github.com/robotpilot/0328100922b33df481c964282d3ab721

### Steps to Reproduce

I try to find the Depth camera manager for ZR300, but I can't.

mdhorn, 2016-10-27:
@robotpilot It looks like your firmware is old; see https://github.com/IntelRealSense/librealsense/#compatible-devices
I would direct you to the contact that supplied the camera for updated firmware.

mdhorn, 2016-10-27:
RealSense ZR300
Camera Firmware 2.0.71.18 and up
USB Adapter Board Firmware 1.27.0.0 and up
Motion Module Firmware 1.23.0.0 and up

robotpilot, 2016-10-27:
Thanks, @mdhorn 
My email address is "pyo at robotis.com".

mdhorn, 2016-10-28:
The problem you reported was a known error using that old firmware.
Once you have updated firmware and the camera is calibrated, please open a new Issue if you continue to have issues.

thank you.

closed 2016-10-28 by mdhorn