realsenseai/realsense-ros #145: Nodelet for ZR300 doesn't launch with default settings

rojkov, 2016-10-31:
### System Configuration

| Version               | Your Configuration   |
|:--------------------- |:---------------------|
| Operating System      | Ostro XT                |
| Kernel                |  4.4.20-yocto-standard               |
| ROS                   |      indigo               |
| ROS RealSense         |   1.6.0              |
| librealsense          | 1.10.0                |
| Camera Type-Firmware  |  Name: Intel RealSense ZR300, Camera FW: 2.0.71.13, Adapter FW: 0.0.1.16           |

### Expected Behavior

`roslaunch realsense_camera zr300_nodelet_default.launch` starts successfully

### Actual Behavior

The command `roslaunch realsense_camera zr300_nodelet_default.launch` reports the error:

    [ERROR] [1477912810.031223078]: /camera/driver - Error calling rs_get_device_extrinsics (device:0x555f955f7860, from:FISHEYE, to:COLOR, extrin:0x7fff56df48f0 ): 
    The extrinsic from FISHEYE to COLOR is not valid

Then it segfaults in the code handling the error:

    #0  0x00007ffff6033038 in __GI_raise (sig=sig@entry=6) at /usr/src/debug/glibc/2.24-r0/git/sysdeps/unix/sysv/linux/raise.c:54
    #1  0x00007ffff603448a in __GI_abort () at /usr/src/debug/glibc/2.24-r0/git/stdlib/abort.c:89
    #2  0x00007ffff607075b in __libc_message (do_abort=do_abort@entry=2, fmt=fmt@entry=0x7ffff6163dc8 "*** Error in `%s': %s: 0x%s ***\n") at /usr/src/debug/glibc/2.24-r0/git/sysdeps/posix/libc_fatal.c:175
    #3  0x00007ffff6076a36 in malloc_printerr (action=3, str=0x7ffff6160a11 "free(): invalid size", ptr=<optimized out>, ar_ptr=<optimized out>) at /usr/src/debug/glibc/2.24-r0/git/malloc/malloc.c:5046
    #4  0x00007ffff607723e in _int_free (av=0x7ffff6397b00 <main_arena>, p=<optimized out>, have_lock=0) at /usr/src/debug/glibc/2.24-r0/git/malloc/malloc.c:3902
    #5  0x00007ffff02780dc in __gnu_cxx::new_allocator<char>::deallocate (this=0x555555ac5ef0, __p=<optimized out>) at /usr/include/c++/5.4.0/ext/new_allocator.h:110
    #6  std::allocator_traits<std::allocator<char> >::deallocate (__a=..., __n=<optimized out>, __p=<optimized out>) at /usr/include/c++/5.4.0/bits/alloc_traits.h:517
    #7  std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_destroy (__size=<optimized out>, this=0x555555ac5ef0) at /usr/include/c++/5.4.0/bits/basic_string.h:185
    #8  std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_dispose (this=0x555555ac5ef0) at /usr/include/c++/5.4.0/bits/basic_string.h:180
    #9  std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::~basic_string (this=0x555555ac5ef0, __in_chrg=<optimized out>) at /usr/include/c++/5.4.0/bits/basic_string.h:543
    #10 rs_error::~rs_error (this=0x555555ac5ef0, __in_chrg=<optimized out>) at src/rs.cpp:17
    #11 rs_free_error (error=0x555555ac5ef0) at src/rs.cpp:680
    #12 0x00007ffff052f01b in realsense_camera::BaseNodelet::checkError (this=0x555555a18f30) at /home/rojkov/work/ostro-os-xt/build/workspace/sources/realsense-camera/realsense_camera/src/base_nodelet.cpp:1035
    #13 0x00007ffff052f8c7 in realsense_camera::BaseNodelet::prepareTopics (this=0x555555a18f30) at /home/rojkov/work/ostro-os-xt/build/workspace/sources/realsense-camera/realsense_camera/src/base_nodelet.cpp:726
    #14 0x00007ffff536a275 in boost::(anonymous namespace)::thread_proxy (param=<optimized out>) at /usr/src/debug/boost/1.61.0-r0/boost_1_61_0/libs/thread/src/pthread/thread.cpp:171
    #15 0x00007ffff579f3c4 in start_thread (arg=0x7fffb93e2700) at /usr/src/debug/glibc/2.24-r0/git/nptl/pthread_create.c:333
    #16 0x00007ffff60e75ef in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:105

### Steps to Reproduce

1. In your Yocto-based distro (e.g. Ostro XT) enable the meta-ros layer (https://github.com/bmwcarit/meta-ros);
2. Make a bitbake recipe for `realsense_camera`;
3. Build an image with ROS and `realsense_camera` included;
4. Follow the instructions at https://github.com/bmwcarit/meta-ros on how to start ROS.
5. Run the command `roslaunch realsense_camera zr300_nodelet_default.launch`

I get this log
```
root@intel-corei7-64:~# roslaunch realsense_camera zr300_nodelet_default.launch
... logging to /home/root/.ros/log/ede297dc-9f5b-11e6-aa79-00133b4908aa/roslaunch-intel-corei7-64-814.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://intel-corei7-64:42622/

SUMMARY
========

PARAMETERS
 * /camera/driver/base_frame_id: camera_link
 * /camera/driver/camera_type: ZR300
 * /camera/driver/color_frame_id: camera_rgb_frame
 * /camera/driver/color_optical_frame_id: camera_rgb_optica...
 * /camera/driver/depth_frame_id: camera_depth_frame
 * /camera/driver/depth_optical_frame_id: camera_depth_opti...
 * /camera/driver/fisheye_frame_id: camera_fisheye_frame
 * /camera/driver/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/driver/imu_frame_id: camera_imu_frame
 * /camera/driver/imu_optical_frame_id: camera_imu_optica...
 * /camera/driver/ir2_frame_id: camera_ir2_frame
 * /camera/driver/ir2_optical_frame_id: camera_ir2_optica...
 * /camera/driver/ir_frame_id: camera_ir_frame
 * /camera/driver/ir_optical_frame_id: camera_ir_optical...
 * /camera/driver/serial_no: 
 * /camera/driver/usb_port_id: 
 * /rosdistro: indigo
 * /rosversion: 1.11.20

NODES
  /camera/
    driver (nodelet/nodelet)
    nodelet_manager (nodelet/nodelet)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[camera/nodelet_manager-1]: started with pid [833]
process[camera/driver-2]: started with pid [834]
[ INFO] [1477912803.880309101]: Initializing nodelet with 4 worker threads.
[ INFO] [1477912805.790287099]: /camera/driver - Detected the following cameras:
				- Serial No: 3141800472, USB Port ID: 2-2, Name: Intel RealSense ZR300, Camera FW: 2.0.71.13, Adapter FW: 0.0.1.16
[ INFO] [1477912805.790473814]: /camera/driver - Connecting to camera with Serial No: 3141800472, USB Port ID: 2-2
[ INFO] [1477912809.840375295]: /camera/driver - Setting static camera options
[ INFO] [1477912809.873601191]: /camera/driver - Enabling Color in preset mode
[ INFO] [1477912809.999323129]: /camera/driver - Enabling Depth in preset mode
[ INFO] [1477912809.999774014]: /camera/driver - Enabling IR in preset mode
[ INFO] [1477912809.999981759]: /camera/driver - Enabling IR2 in preset mode
[ INFO] [1477912810.000172606]: /camera/driver - Enabling Fisheye in preset mode
[ INFO] [1477912810.000468422]: /camera/driver - Starting camera
[ INFO] [1477912810.030548340]: /camera/driver - Publishing camera transforms
[ERROR] [1477912810.031223078]: /camera/driver - Error calling rs_get_device_extrinsics ( device:0x555f955f7860, from:FISHEYE, to:COLOR, extrin:0x7fff56df48f0 ): 
The extrinsic from FISHEYE to COLOR is not valid 

*** Error in `/opt/ros/indigo/lib/nodelet/nodelet': munmap_chunk(): invalid pointer: 0x0000555f956a8850 ***
======= Backtrace: =========
/lib/libc.so.6(+0x3c94070756)[0x7fbaebb9c756]
/lib/libc.so.6(+0x3c94076a36)[0x7fbaebba2a36]
/usr/lib/librealsense.so(rs_free_error+0x2c)[0x7fbaddd800dc]
/opt/ros/indigo/lib//librealsense_camera_nodelet.so(_ZN16realsense_camera11BaseNodelet10checkErrorEv+0x6b)[0x7fbae405701b]
/opt/ros/indigo/lib//librealsense_camera_nodelet.so(_ZN16realsense_camera11BaseNodelet13prepareTopicsEv+0x447)[0x7fbae40578c7]
/usr/lib/libboost_thread.so.1.61.0(+0x10275)[0x7fbaeae96275]
/lib/libpthread.so.0(+0x3c944073c4)[0x7fbaeb2cb3c4]
/lib/libc.so.6(clone+0x5f)[0x7fbaebc135ef]

```

Also I tried to disable Fisheye by setting `enable_fisheye` to `false`, but I'm still getting the same error.

mkhansen-intel, 2016-10-31:
@rojkov It looks like your firmware versions are old. We're running Camera FW: 2.0.71.20, Adapter FW: 1.27.0.0, Motion Module FW: 1.23.0.0. You should get updated FW from your supplier. Also, you should verify this works on Ubuntu before trying it on Ostro, as our supported OS is Ubuntu. I hope this helps.

tripzero, 2016-10-31:
Do the librealsense examples run with the zr300 camera on Ostro?  The crash is in librealsense, but it's on rs_free_error().  It could be we are not checking for a null error message before trying to free...  If the librealsense examples run, it may be a bug in how the node is handling error messages.

mdhorn, 2016-10-31:
The problem you noted is a known issue running the older, outdated FW.  Please update to the latest documented FW.
https://github.com/IntelRealSense/librealsense/#compatible-devices

rojkov, 2016-11-01:
Right, the FW I use seems to be very old.

@tripzero all examples run with no segfaults on Ostro and seem to work fine except `cpp-enumerate-devices`, it reports failures in fisheye too due to the outdated FW apparently:
```
 Stream FISHEYE - 2 modes:
  640   x 480   @ 60Hz  RAW8rs.warn: Intrinsics validation ofFISHEYE failed, because the reading of calibration table failed
rs.error: The intrinsic of FISHEYE is not valid
        180 x 180 degrees
  640   x 480   @ 30Hz  RAW8rs.warn: Intrinsics validation ofFISHEYE failed, because the reading of calibration table failed
rs.error: The intrinsic of FISHEYE is not valid
        180 x 180 degrees
```
So, there must be something fishy in the way how error messages are freed.

andrepfr, 2016-11-04:
I get a very similar error using the newest firmware. Running ROS Kinetic on Ubuntu 16.04. Any ideas on how to solve this? Thanks.

```
roslaunch realsense_camera zr300_nodelet_default.launch 
... logging to /home/andreas/.ros/log/6bf846fa-a29d-11e6-81fa-7c7a918b90c0/roslaunch-carbonX1-20280.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://carbonX1:39227/

SUMMARY
========

PARAMETERS
 * /camera/driver/base_frame_id: camera_link
 * /camera/driver/camera_type: ZR300
 * /camera/driver/color_frame_id: camera_rgb_frame
 * /camera/driver/color_optical_frame_id: camera_rgb_optica...
 * /camera/driver/depth_frame_id: camera_depth_frame
 * /camera/driver/depth_optical_frame_id: camera_depth_opti...
 * /camera/driver/fisheye_frame_id: camera_fisheye_frame
 * /camera/driver/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/driver/imu_frame_id: camera_imu_frame
 * /camera/driver/imu_optical_frame_id: camera_imu_optica...
 * /camera/driver/ir2_frame_id: camera_ir2_frame
 * /camera/driver/ir2_optical_frame_id: camera_ir2_optica...
 * /camera/driver/ir_frame_id: camera_ir_frame
 * /camera/driver/ir_optical_frame_id: camera_ir_optical...
 * /camera/driver/serial_no: 
 * /camera/driver/usb_port_id: 
 * /rosdistro: kinetic
 * /rosversion: 1.12.5

NODES
  /camera/
    driver (nodelet/nodelet)
    nodelet_manager (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [20291]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 6bf846fa-a29d-11e6-81fa-7c7a918b90c0
process[rosout-1]: started with pid [20304]
started core service [/rosout]
process[camera/nodelet_manager-2]: started with pid [20318]
process[camera/driver-3]: started with pid [20322]
[ INFO] [1478270769.129851792]: Initializing nodelet with 4 worker threads.
[ INFO] [1478270769.663466342]: /camera/driver - Detected the following cameras:
				- Serial No: 3181802298, USB Port ID: 3-2, Name: Intel RealSense ZR300, Camera FW: 2.0.71.20, Adapter FW: 1.27.0.0, Motion Module FW: 1.23.0.0
[ INFO] [1478270769.663544259]: /camera/driver - Connecting to camera with Serial No: 3181802298, USB Port ID: 3-2
[ INFO] [1478270770.428076851]: /camera/driver - Setting static camera options
[ INFO] [1478270770.433625471]: /camera/driver - Enabling Color in preset mode
[ INFO] [1478270770.434014807]: /camera/driver - Enabling Depth in preset mode
[ INFO] [1478270770.434206876]: /camera/driver - Enabling IR in preset mode
[ INFO] [1478270770.434342596]: /camera/driver - Enabling IR2 in preset mode
[ INFO] [1478270770.434481561]: /camera/driver - Enabling Fisheye in preset mode
[ INFO] [1478270770.434673083]: /camera/driver - Starting camera
[ INFO] [1478270770.532455836]: /camera/driver - Publishing camera transforms
[ERROR] [1478270770.532742177]: /camera/driver - Error calling rs_get_device_extrinsics ( device:0x11cb290, from:FISHEYE, to:COLOR, extrin:0x7ffeaf1e62c0 ): 
The extrinsic from FISHEYE to COLOR is not valid 

[FATAL] [1478270770.720449333]: Failed to load nodelet '/camera/driver` of type `realsense_camera/ZR300Nodelet` to manager `nodelet_manager'
*** Error in `/opt/ros/kinetic/lib/nodelet/nodelet': munmap_chunk(): invalid pointer: 0x000000000110a6d0 ***
======= Backtrace: =========
/lib/x86_64-linux-gnu/libc.so.6(+0x777e5)[0x7f5881f2c7e5]
/lib/x86_64-linux-gnu/libc.so.6(cfree+0x1a8)[0x7f5881f38ae8]
/opt/ros/kinetic/lib/x86_64-linux-gnu/librealsense.so(rs_free_error+0x2c)[0x7f5871d8654c]
/home/andreas/irs_ws/devel/lib//librealsense_camera_nodelet.so(_ZN16realsense_camera11BaseNodelet10checkErrorEv+0x6b)[0x7f587815bfab]
/home/andreas/irs_ws/devel/lib//librealsense_camera_nodelet.so(_ZN16realsense_camera11BaseNodelet13prepareTopicsEv+0x447)[0x7f587815c857]
/usr/lib/x86_64-linux-gnu/libboost_thread.so.1.58.0(+0x115d5)[0x7f588143c5d5]
/lib/x86_64-linux-gnu/libpthread.so.0(+0x770a)[0x7f588121570a]
/lib/x86_64-linux-gnu/libc.so.6(clone+0x6d)[0x7f5881fbb82d]

```

mkhansen-intel, 2016-11-14:
@andrepfr - It looks like your camera doesn't contain calibration information. Check with your supplier to find out how to calibrate it.

mkhansen-intel, 2016-12-08:
I believe the original issue has been resolved. I'm closing this issue.

closed 2016-12-08 by mkhansen-intel