realsenseai/realsense-ros #148: Debfile compatibility issue

josephduchesne, 2016-11-03:
### System Configuration

| Version               | Your Configuration   |
|:--------------------- |:---------------------|
| Operating System      |  Ubuntu 14.04              |
| Kernel                | 4.4 w/ patch              |
| ROS                   | Indigo                   |
| ROS RealSense         | 1.3.0-0trusty-20160824-174135-0700               |
| librealsense          | 0.9.2-3trusty-20160824-173529-0700                |
| Camera Type-Firmware  | Intel R200-?.?.?.?            |

### Expected Behavior
We have created our own internal deb package that deends on ros-indigo-librealsense and ros-indigo-realsense-camera. We have set version compatibility to ros-indigo-librealsense=0.9.2 and ros-indigo-realsense-cameras=1.3.0. We expect to be able to install our software.

### Actual Behavior
The software install fails because 0.9.2 and 1.3.0 are no longer hosted on the deb files.

### Steps to Reproduce
Try to install any version of librealsense but the latest one.

Listing available versions shows:
```
> apt-cache madison ros-indigo-librealsense
ros-indigo-librealsense | 1.11.0-1trusty-20161020-181438-0700 | http://packages.ros.org/ros/ubuntu/ trusty/main amd64 Packages
```

From our end, we will begin caching deb files so that this doesn't happen going forward, but it's pretty inconvenient to have compatibility breaking versions entirely replace existing ones without the option to downgrade or specify the desired version.

mdhorn, 2016-11-03:
Not caching files isn't a choice by any package developer, it is ROS standard from their distribution repository.

Is this isn't a realsense_camera or librealsense problem, it is a ROS problem that our team won't be able to solve.

mdhorn, 2016-11-03:
BTW, I'd highly suggest moving to the latest versions of librealsense 1.11.0 and realsense_camera 1.5.0 as the new librealsense package handles installing the udev rules and using DKMS to patch the kernel.

closed 2016-11-03 by mdhorn

josephduchesne, 2016-11-04:
I attempted to upgrade, however the new package failed to install properly in our environment (DKMS failed). We may be able to devote more resource to troubleshooting this in the future, but for the time being we have to stick with the version that we have working.