realsenseai/realsense-ros #151: r200 rgbd and default launch file

fw-timku, 2016-11-15:
Hi all, 

I would like to know what is the difference between the r200_nodelet_default.launch and r200_nodelet_rgbd.launch? I have tried running both launch file and it seems that both produces different rostopic.

Particularly for the rgbd.launch, some of the rostopic includes:
/camera/depth/image_rect
/camera/ir/image_rect_ir
/camera/rgb/image_rect_color
/camera/rgb/image_rect_mono

Is the above few topics the rectified version for the depth, ir and rgb stream?

While the default.launch only produces the raw stream:
/camera/depth/image_raw
/camera/ir/image_raw
/camera/color/image_raw

mkhansen-intel, 2016-11-15:
The rgbd launch file is meant to be compatible with the rgbd launch package: http://wiki.ros.org/rgbd_launch

The default launch file turns on the bare minimum image streams only.

Keep in mind these are both for examples only.

mkhansen-intel, 2016-11-18:
@fw-timku - does that answer your question?

fw-timku, 2016-11-18:
@mkhansen-intel 
I have managed to run the ros calibration and I would like to know how do I go about loading the camera calibration file into the respective camera_info topic?

mkhansen-intel, 2016-12-08:
I'm sorry our camera doesn't currently support loading the ROS calibration files. Contact the librealsense team for info on how to calibrate your camera.

closed 2016-12-08 by mkhansen-intel