realsenseai/realsense-ros #152: Unable to change frame rate

MathiasCiarlo, 2016-11-15:
### System Configuration

| Version               | Your Configuration   |
|:--------------------- |:---------------------|
| Operating System      | Ubuntu 16.04 |
| Kernel                | 4.4.0-47-generic |
| ROS                   | kinetic |
| ROS RealSense         | 1.6.0 |
| librealsense          | 1.11.1 |
| Camera Type-Firmware  | Intel RealSense F200, Camera FW: 2.60.0.0 |

---
I have been trying to change the fps for a week, without results. The first thing I tried was to change the static parameters: depth_fps, color_fps and fisheye_fps, without results.

After looking at librealsense I discovered a dynamic parameter called f200_dynamic_fps. This was default set to 0 when I called

    $ rosservice call /camera/driver/get_settings

I tried to change the settings with

    $ rosrun rqt_reconfigure rqt_reconfigure

but f200_dynamic_fps did not show up under /camera/driver

To get around this issue I added the following line to the parameter generator in realsense_camera/cfg/f200_params.cfg:

    gen.add("f200_dynamic_fps", int_t, 0, "Dynamic fps", 2, 2, 60)

I also added the following line to realsense_camera/src/f200_nodelet.cpp

    rs_set_device_option(rs_device_, RS_OPTION_F200_DYNAMIC_FPS, config.f200_dynamic_fps, 0); 

This makes the option appear in rqt_reconfigure, but changing the fps does not change the actual fps.
I validated the fps with

    $ rostopic hz /camera/depth/camera_info

This gives around 53 fps for me.

### Expected Behavior
The fps is changed to 2

### Actual Behavior
The fps stays at 60 (Unchanged)

### Steps to Reproduce
1. In realsense_camera/cfg/f200_params.cfg add the line
```c++
gen.add("f200_dynamic_fps", int_t, 0, "Dynamic fps", 2, 2, 60)
```
2. In realsense_camera/src/f200_nodelet.cpp add the line
``` c++
rs_set_device_option(rs_device_, RS_OPTION_F200_DYNAMIC_FPS, config.f200_dynamic_fps, 0);
```
3. Build

4. Run the node
``` bash
$ roslaunch realsense_camera f200_nodelet_default.launch
```
3. Run  rqt_reconfigure and change the value f200_dynamic_fps to 2 (will be 2 default after the config above)
``` bash
$ rosrun rqt_reconfigure rqt_reconfigure
```
4. Validate fps with
``` bash
$ rostopic hz /camera/depth/camera_info
```

mdhorn, 2016-11-16:
There are a few problems here.

First, the proper way to change from the default Frames Per Second is to switch from the default of mode:=preset to using the mode:=manual.
See **mode** in the [Parameters section of the wiki doc.](http://wiki.ros.org/realsense_camera#Parameters)

The realsense_camera package has an example launch file for the R200 camera, but I'm attaching a version for the F200 setting the FPS to 5.
[f200_nodelet_modify_params.launch.txt](https://github.com/intel-ros/realsense/files/593675/f200_nodelet_modify_params.launch.txt)
`roslaunch f200_nodelet_modify_params.launch`

The list of [valid FPS for resolutions is published on the librealsense GitHub](https://github.com/IntelRealSense/librealsense/blob/master/doc/supported_video_formats.pdf).

In creating the example launch file for you, I ran into two problems.

1.  The existing realsense_camera package (1.6.0) only allow the setting of the depth and IR streams to the same values. For the F200 camera that overlap is only 30 or 60fps. This will be resolved in the next release of realsense_camera package as the IR streams will be disable by default allowing the selection of a lower fps for the depth stream. See PR #150 

2. librealsense appears to have a bug which locks the Depth stream to 30fps for both the F200 and SR300 cameras regardless of the requested FPS. See [librealsense Issue 361](https://github.com/IntelRealSense/librealsense/issues/361) 

The attached launch file will allow you to get the color stream at the lower rate, but the depth appears to be stuck at 30fps until we can get a fix from librealsense team.

mdhorn, 2016-11-21:
The librealsense bug appears to not be a real issue.

Were you able to use the attached launch file to change the FPS on your camera?

MathiasCiarlo, 2016-11-22:
I see, I am now able to change fps, thank you for the help! I did not understand that I needed to set the mode to change the fps! I see it is written vaguely under **mode** in the list of static parameters, but maybe it would be smart to write it at the very top under static parameters?

I tried all fps settings, all works except for color fps set to 2 (supposed to be supported), which crashes with this error:

[ERROR] [1479818696.536289742]: /camera/driver - Error calling rs_wait_for_frames ( device:0x2187e00 ): 
Timeout waiting for frames. 

[camera/nodelet_manager-1] process has died [pid 9449, exit code 1, cmd /opt/ros/kinetic/lib/nodelet/nodelet manager __name:=nodelet_manager __log:=/home/mathiact/.ros/log/399d2a96-b0b1-11e6-a0eb-7c7a911c589a/camera-nodelet_manager-1.log].
log file: /home/mathiact/.ros/log/399d2a96-b0b1-11e6-a0eb-7c7a911c589a/camera-nodelet_manager-1*.log
[Logfile - camera-driver-2-stdout.txt](https://github.com/intel-ros/realsense/files/606397/camera-driver-2-stdout.txt)

Thank you so much for the help! :)

mdhorn, 2016-11-22:
I've made some updates to the [ROS wiki page](http://wiki.ros.org/realsense_camera#ROS_API) to try to highlight the default of 'preset' mode.  Also added a short Tutorial on how to make a modified launch file for modifying parameters.

closed 2016-11-22 by mdhorn

MathiasCiarlo, 2016-11-23:
This is great! Well done :)

MathiasCiarlo, 2016-12-05:
Hi, can you get back to me here, @mdhorn?

I built this on a new computer, and the fps worked the day I installed it, but not after. It is stuck to 30 again. I use your launch file. Launching does not give any errors, it says depth_fps is set to 5, but in reality it is set to 30. Can you please help me on debugging this?

To summarize: This worked the first day, but not after reboot. I have tried to rebuild it several times, no change. What can I do?

mdhorn, 2016-12-05:
@MathiasCiarlo this has not been 100% reproducible. I did see an F200 stuck at 30 fps for depth, but have not been able to reproduce this problem for a while.
What USB devices are attached?  I would try running the attached launch file with just the camera attached and remote log into the system.
My guess it there is some USB traffic issue causing the problem.

[f200_nodelet_modify_params.launch.txt](https://github.com/intel-ros/realsense/files/632526/f200_nodelet_modify_params.launch.txt)

MathiasCiarlo, 2016-12-06:
Thanks. I was using a keyboard and mouse through a monitor hub, but nothing changed when I unplugged them and remote logged in. I think I will have to use /depth/image_raw/compressedDepth instead and manually build the point cloud.. Hopefullly I will figure out how to do it by looking at the rviz source code.