realsenseai/realsense-ros #156: Cannot start two drivers at the same time

tulku, 2016-11-21:
### System Configuration

| Version               | Your Configuration   |
|:--------------------- |:---------------------|
| Operating System      | Ubuntu 16.04.1 LTS                |
| Kernel                | 4.4.0-47                |
| ROS                   | Kinetic                    |
| ROS RealSense         |  1.6.0              |
| librealsense          | 1.11.1                |
| Camera Type-Firmware  |      1.0.72.06 / 1.0.72.04      |

### Expected Behavior

When using more than one camera, it would be good to use one nodelet manager per camera, and start both nodes from one launchfile.

### Actual Behavior

Many times (but not always), one of the nodes dies when trying to open the camera, as can be seen in the following log:

```
[ INFO] [1479760830.605049212]: Initializing nodelet with 8 worker threads.
[ INFO] [1479760830.613274391]: Initializing nodelet with 8 worker threads.
[ERROR] [1479760830.909300637]: /camera_two/driver - Error calling rs_create_context ( api_version:11100 ): 
UVCIOC_CTRL_QUERY:UVC_SET_CUR error 5, Input/output error 

[ INFO] [1479760830.916027551]: /camera_one/driver - Detected the following cameras:
				- Serial No: 2361000476, USB Port ID: 2-1, Name: Intel RealSense R200, Camera FW: 1.0.72.06
				- Serial No: 26, USB Port ID: 2-2, Name: Intel RealSense R200, Camera FW: 1.0.72.04
[ERROR] [1479760830.916115296]: /camera_one/driver - Couldn't find camera to connect with serial_no = 2191300209, usb_port_id = 
[ INFO] [1479760830.916928998]: /camera_one/driver - Sleeping 5 seconds then retrying to connect
[FATAL] [1479760831.048640380]: Failed to load nodelet '/camera_two/driver` of type `realsense_camera/R200Nodelet` to manager `camera_two_nodelet_manager'
[camera_two/camera_two_nodelet_manager-2] process has died [pid 15667, exit code 1, cmd /opt/ros/kinetic/lib/nodelet/nodelet manager __name:=camera_two_nodelet_manager __log:=/home/tulku/.ros/log/bddbf512-b02a-11e6-8662-001fc69bb339/camera_two-camera_two_nodelet_manager-2.log].
log file: /home/tulku/.ros/log/bddbf512-b02a-11e6-8662-001fc69bb339/camera_two-camera_two_nodelet_manager-2*.log
[camera_two/driver-3] process has died [pid 15668, exit code 255, cmd /opt/ros/kinetic/lib/nhttps://github.com/intel-ros/realsense/issues/120odelet/nodelet load realsense_camera/R200Nodelet camera_two_nodelet_manager depth:=depth color:=color ir:=ir ir2:=ir2 fisheye:=fisheye imu:=imu __name:=driver __log:=/home/tulku/.ros/log/bddbf512-b02a-11e6-8662-001fc69bb339/camera_two-driver-3.log].
log file: /home/tulku/.ros/log/bddbf512-b02a-11e6-8662-001fc69bb339/camera_two-driver-3*.log
^C[camera_one/driver-5] killing on exit
[camera_one/camera_one_nodelet_manager-4] killing on exit
```

The current workaround is to start both cameras in the same nodelet manager, but that triggers https://github.com/intel-ros/realsense/issues/120

### Steps to Reproduce

Plug two cameras, look for their serial numbers, download [this launchfile](https://gist.github.com/tulku/db7f867304082fab7c30cc1ac8482983) and run:

```
export CAMERA_ONE_ID=<id camera 1>
export CAMERA_TWO_ID=<id camera 2>
roslaunch dual_cameras.launch
```

kevincwells, 2016-12-05:
Hi @tulku,

This is a known issue with a race condition in librealsense over rs_create_context.  The current work-around is to launch the cameras in separate launch files.  This gets around the timing by introducing a delay between the calls to rs_create_context.

I believe this is technically a duplicate of #22, but that issue was closed at the time because the single-nodelet-manager workaround was deemed acceptable and issue #120 didn't yet exist. Because the workaround in that issue is no longer viable, I will leave this open and not mark it as a duplicate.

tulku, 2016-12-07:
Hi @kevincwells. Is the upstream (librealsense) issue created? I could not find it in github.

Your propose workaround (using two nodelets) does not work as you cannot control the time roslaunch takes between starting the two drivers. That approach results on unreliable system startup, which can work or not intermittently.

My current 'solution' is to use only one nodelet, disabling the tf_publishing and workaround #120 manually publishing the static transforms needed using tf static publishers. This solution is obviously less than ideal.

kevincwells, 2017-03-09:
Hi @tulku,

Hopefully you're no longer having this issue.

1. The underlying librealsense race condition has since been fixed for R200 cameras (and remains only for F200/SR300 cameras), so with the current editions of the code you should be able to run the nodelets in separate nodelet managers in a single launch file.
2. If you're running the two camera nodelets in the same nodelet manager (as in the workaround originally presented in #22), the static transform issue mentioned in #120 can now easily be addressed by enabling dynamic instead of static transforms.

Please let us know if you're still experiencing any additional issues.

Cheers,
Kevin

closed 2017-03-09 by kevincwells