realsenseai/realsense-ros #161: Cant get depth image and ir, but can get color image

Playfish, 2016-12-07:
### System Configuration
Please complete Your Configuration detail below. Refer to the BKC at [Configuration](../realsense_camera/README.md#configuration).

| Version               | Your Configuration   |
|:--------------------- |:---------------------|
| Operating System      |  Ubuntu 14.04.4 LTS             |
| Kernel                | 4.4.0-040400-generic               |
| ROS                   | indigo                    |
| ROS RealSense         | 1.6.1               |
| librealsense          | 1.11.1                |
| Camera Type-Firmware  | 1.0.72.06             |

---
#### How to collect Configuration Data
*This section can be deleted before submission.*

| Version               | Method |
|:--------------------- |:------------ |
| Operating System      | `grep DISTRIB_DESCRIPTION /etc/*elease*` or `lsb_release -d` |
| Kernel                | `uname -r` |
| ROS                   | `rosversion -d` |
| ROS RealSense         | `rosversion realsense_camera` |
| librealsense          | `rosversion librealsense` |
| Camera Type-Firmware  | View the ROS log from running nodelet |

---

### Expected Behavior
I want get topics of depth images, but the depth image was black and color images was ok.

### Actual Behavior
Everything was well when I used tutorials c-tutorials-streams, but when I run with ros will get error with there is nothing in topics with depth images, and color topics is adverse.
next is using tutorials get images.
[image]
and follow this is the result when I launch r200_nodelet_default.launch
[image]

### Steps to Reproduce
roslaunch realsense_camera r200_nodelet_default.launch 

How should I do can get depth topics?
Thanks!

mkhansen-intel, 2016-12-08:
The INFO statement says it is enabling depth in preset mode. How are you checking for the depth image? Try this: rostopic hz /camera/depth/image_raw
Also try: rostopic echo /camera/depth/camera_info

Playfish, 2016-12-09:
Hi, Thanks! Now its working well, I think is uvcvideo's problem, because I got wrong installation step that script of patch-uvcvideo-4.4.sh v4.4-wily shown me configure(sorry about that I forgot that content ) something and I skipped that. Yesterday I run this script of patch-uvcvideo-4.4.sh v4.4-wily with root, not get configure, and get depth image.

mkhansen-intel, 2016-12-12:
Glad to hear it. I'm closing this issue!

closed 2016-12-12 by mkhansen-intel