realsenseai/realsense-ros #167: IR image frames are flashing between dark and bright

mehdish89, 2016-12-16:
### System Configuration
Please complete Your Configuration detail below. Refer to the BKC at [Configuration](../realsense_camera/README.md#configuration).

| Version                            | Your Configuration                            |
|:-------------------------------- |:-------------------------------------------------|
| Operating System           | Ubuntu 14.04.5 LTS                          |
| Kernel                             | 4.4.0-040400-generic                        |
| ROS                                | indigo                                                |
| ROS RealSense             | 1.6.0                                                  |
| librealsense                    | 1.11.1                                                |
| Camera Type-Firmware  | Intel RealSense SR300 - 3.10.10.0  |

---

### Expected Behavior### Actual Behavior

### Steps to Reproduce

The camera is setup correctly, I ran kernel update patch from the original librealsense repository, and Its working fine when you open the examples provided. I enabled the kernel sources and installed ros-indigo-librealsense and ros-indigo-realsense-camera. I expected it to work the same using ROS. But running the camera using realsense_camera package in ROS, presents flashing IR stream. The IR image frames frequently jump back and forth between dark and bright. So the depth image is also flashing. But the color image is fine.

mdhorn, 2016-12-16:
For ROS, you should not be running anything from the librealsense repository -- no patching.

The default kernel with Ubuntu 14.04.5 LTS will automatically be correctly patched as part of installing the ros-indigo-realsense-camera package (which requires the ros-indigo-librealsense package which handles all of the patching automatically).

You should remove the 4.4.0-040400-generic kernel from your system, and reboot into the default latest kernel (currently 4.4.0-47-generic).

Follow the direction on the ROS wiki for the Installation prerequisites.
http://wiki.ros.org/librealsense#Installation_Prerequisites
(This enables the download of kernel sources for automatic patching of all future kernel updates.)

And then just install the ros realsense camera package
http://wiki.ros.org/realsense_camera#Installation

superjax, 2017-01-17:
I am seeing similar behavior with my SR300.  Slightly different though.  After running `roslaunch realsense_camera sr300_nodelet_rgbd.launch`, I get a flashing `/camera/depth/image_raw` (as viewed with rviz) but my `/camera/ir/image_raw` only has the topmost left pixel illuminated while the rest is black.

I can also confirm that the camera works in the official Intel SDK windows utility.  I tried updating the firmware through Intel's utility, so I am running FW 3.21.0.0 but that didn't seem to change behavior at all.  I am running kernel 4.4.0.59-generic, xenial and kinetic, and I'm pretty sure I have followed the instructions on the previous link in setting up librealsense by using the provided script and `apt-get`

Is there anything that I am missing with the SR300?  The IR images are particularly confusing to me.

superjax, 2017-02-05:
As an update, it seems to have been fixed by compiling realsense from source after cloning master.

mattcurfman, 2017-02-15:
@superjax Thanks for letting us now :)

closed 2017-02-15 by mdhorn