realsenseai/realsense-ros #171: Not able to install ROS Debian Package

ank4444, 2016-12-18:
| Version               | Your Configuration   |
|:--------------------- |:---------------------|
| Operating System      | Ubuntu 16.04.1 LTS                |
| Kernel                | 4.4.0-53-generic                |
| ROS                   | N.A.                    | -->  I want to install it, but error during unable to locate package ros-*-realsense-camera

PS: I have followed prerequisite from http://wiki.ros.org/librealsense

mdhorn, 2016-12-18:
Have you already followed the Wiki to install ROS.
http://wiki.ros.org/ROS/Installation

mdhorn, 2016-12-18:
If you have setup the ROS environment, are you running the command including the quotes?
The single quote are needed to prevent the asterisk from being expanded by the shell. 

`sudo apt-get install 'ros-*-realsense-camera'`

mdhorn, 2016-12-21:
@ank4444 Please provide an update or more information if you are still having problems.

ank4444, 2016-12-21:
Yes, I am running a command using single quotes, but I am getting error unable to locate package ros-*-realsense-camera

mdhorn, 2016-12-21:
Since you are on 16.04, you would be using Kinetic, so you could try the exact package name:
`sudo apt-get update`
`sudo apt-get install ros-kinetic-realsense-camera`

If you aren't seeing the package, then you have not completed all of the installation steps for ROS:
http://wiki.ros.org/ROS/Installation

ank4444, 2016-12-22:
Yes, It solves.

closed 2016-12-22 by ank4444