realsenseai/realsense-ros #178: 720p color video not working

josephduchesne, 2016-12-22:
### System Configuration
Please complete Your Configuration detail below. Refer to the BKC at [Configuration](../realsense_camera/README.md#configuration).

| Version               | Your Configuration   |
|:--------------------- |:---------------------|
| Operating System      | Ubuntu 14.04                |
| Kernel                | 4.4.0-47-generic               |
| ROS                   | Indigo                   |
| ROS RealSense         | 1.5.0                |
| librealsense          | 1.11.0                |
| Camera Type-Firmware  | 1.0.72.06            |

### Expected Behavior
I would like to launch the node and read 1280x720 color video at 30fps. 

### Actual Behavior
```
[ INFO] [1482448098.000314925]: /camera_left/driver - Enabling Color in manual mode
[ERROR] [1482448098.001060156]: /camera_left/driver - Error calling rs_get_stream_intrinsics ( device:0xe64308, stream:COLOR, intrin:0x7ffff8d33980 ): 
uvc subdevice 2 cannot provide 1280x720:RGB8@30Hz COLOR 

```

### Steps to Reproduce

Launch node with these settings:
```
driver:
    camera_type: "R200"
    enable_depth: true
    enable_color: true
    enable_pointcloud: false
    enable_tf: false
    mode: "manual"
    color_width: 1280
    color_height: 720
    color_fps: 30
    depth_width: 480 
    depth_height: 360
    depth_fps: 30
```

mdhorn, 2016-12-22:
The error message is from the underlying librealsense library which has the camera driver.
1280x720@30fps is not a valid format for the R200.
See this sheet and refer to the R200 column for valid formats:
https://github.com/IntelRealSense/librealsense/blob/master/doc/supported_video_formats.pdf

josephduchesne, 2016-12-23:
I searched for documentation, but apparently not hard enough. Thanks!

Both of these pages were incorrect:
http://reconstructme.net/qa_faqs/intel-realsense-r200-review/
https://software.intel.com/en-us/articles/realsense-r200-camera

closed 2016-12-23 by josephduchesne

mdhorn, 2017-01-03:
@josephduchesne Thank you Joseph, I'll try to find the page owner and get that corrected/removed.