realsenseai/realsense-ros #184: The rotation between Depth and color image coordinate system

laofanfan, 2017-01-07:
### System Configuration
Please complete Your Configuration detail below. Refer to the BKC at [Configuration](../realsense_camera/README.md#configuration).

| Version               | Your Configuration   |
|:--------------------- |:---------------------|
| Operating System      | Ubuntu 16.04.1 LTS                |
| Kernel                | 4.4.0-31-generic               |
| ROS                   | kinetic                  |
| ROS RealSense         | 1.6.0               |
| librealsense          | 1.12              |
| Camera Type-Firmware  | Serial No: 2391016074, USB Port ID: 2-4, Name: Intel RealSense R200, Firmware: 1.0.72.06          |

---

### Expected Behavior

Last time I got a small registered error in /camera/depth_registered/points
Issue Link:  https://github.com/intel-ros/realsense/issues/115

I found maybe the problem is about the  rotation between Depth and Color image.

In the base_nodelet.cpp, the  rotation between Depth and Color image coordinate system was set to identity matrix. as follows
///////
    b2d_msg.header.stamp = transform_ts_;
    b2d_msg.header.frame_id = base_frame_id_;
    b2d_msg.child_frame_id = frame_id_[RS_STREAM_DEPTH];
    b2d_msg.transform.translation.x =  color2depth_extrinsic_.translation[2];
    b2d_msg.transform.translation.y = -color2depth_extrinsic_.translation[0];
    b2d_msg.transform.translation.z = -color2depth_extrinsic_.translation[1];
    b2d_msg.transform.rotation.x = 0;
    b2d_msg.transform.rotation.y = 0;
    b2d_msg.transform.rotation.z = 0;
    b2d_msg.transform.rotation.w = 1;
static_tf_broadcaster_.sendTransform(b2d_msg);
///////
However in my r200 device the really rotation matrix is not a identity matrix. as follows
//////
  depth_to_rgb(0,0) = 0.999996;
  depth_to_rgb(1,0) = 0.002805;
  depth_to_rgb(2,0) = -0.000393;
  depth_to_rgb(0,1) = -0.002796;
  depth_to_rgb(1,1) = 0.999757;
  depth_to_rgb(2,1) = 0.021851;
  depth_to_rgb(0,2) = 0.000454;
  depth_to_rgb(1,2) = -0.021850;
  depth_to_rgb(2,2) = 0.999761;
//////  
So I changed the color image stream from RS_STREAM_COLOR to RS_STREAM_RECTIFIED_COLOR, the registered error was gone.

Does this a bug or something else?

### Actual Behavior

### Steps to Reproduce

skohlbr, 2017-01-30:
Can you clarify what you did to fix the issue? I get a compile error when just changing to RS_STREAM_RECTIFIED_COLOR, but the define is used in librealsense. Before I start digging deeper it probably would be easier if you could provide more details.

The other approach for fixing things would be to convert the rotation matrix to a quaternion and then publish it to tf at the lines you indicated.

laofanfan, 2017-02-07:
dear skohlbr,

     I change two place of the base_nodelet.cpp as follows: you can try it

1，///////
  void BaseNodelet::getStreamCalibData(rs_stream stream_index)
  {
    rs_intrinsics intrinsic;
      if(stream_index == RS_STREAM_COLOR)

rs_get_stream_intrinsics(rs_device_, RS_STREAM_RECTIFIED_COLOR, &intrinsic, &rs_error_);
      else
      rs_get_stream_intrinsics(rs_device_, stream_index, &intrinsic, &rs_error_);

    checkError();

2，////////
    else
    {
if(stream_index == RS_STREAM_COLOR)
image_[stream_index].data = (unsigned char *) (rs_get_frame_data(rs_device_, RS_STREAM_RECTIFIED_COLOR, 0));
      else
      image_[stream_index].data = (unsigned char *) (rs_get_frame_data(rs_device_, stream_index, 0));
    }
  }

  /*
   * Prepare and publish topics.
   */
  void BaseNodelet::prepareTopics()

mkhansen-intel, 2017-03-13:
@mattcurfman - does it make sense to you to change RS_STREAM_COLOR to RS_STREAM_RECTIFIED_COLOR?

mattcurfman, 2017-04-25:
@mkhansen-intel yes.  would welcome a PR for it.

icarpis, 2018-01-31:
Fixed by #261

closed 2018-01-31 by icarpis