realsenseai/realsense-ros #185: R200 always do not working on linux

BlazingForests, 2017-01-08:
### System Configuration
Please complete Your Configuration detail below. Refer to the BKC at [Configuration](../realsense_camera/README.md#configuration).

| Version               | Your Configuration   |
|:--------------------- |:---------------------|
| Operating System      |Ubuntu 14.04.5 LTS                |
| Kernel                |  4.4.0-58-generic #79~14.04.1-Ubuntu                |
| ROS                   | indigo                    |
| ROS RealSense         | 1.7.0                |
| librealsense          | 1.12.0               |
| Camera Type-Firmware  | R200 -  1.0.72.10      |

### Expected Behavior
PointCloud2 msg come from R200

### Actual Behavior
NO data

### Steps to Reproduce

When i power on my computer, plug-in R200, the `dmesg` output is

```
[ 1027.491773] usb 4-3: new SuperSpeed USB device number 2 using xhci_hcd
[ 1027.510455] usb 4-3: New USB device found, idVendor=8086, idProduct=0a80
[ 1027.510461] usb 4-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1027.510465] usb 4-3: Product: Intel RealSense 3D Camera R200
[ 1027.510481] usb 4-3: Manufacturer: Intel Corp
[ 1027.510484] usb 4-3: SerialNumber: SN_2361000097
[ 1027.552433] media: Linux media interface: v0.10
[ 1027.566863] Linux video capture interface: v2.00
[ 1027.598452] uvcvideo: Found UVC 1.10 device Intel RealSense 3D Camera R200 (8086:0a80)
[ 1027.600217] uvcvideo: Found UVC 1.10 device Intel RealSense 3D Camera R200 (8086:0a80)
[ 1027.601842] uvcvideo: Unable to create debugfs 4-2 directory.
[ 1027.601944] input: Intel RealSense 3D Camera R200 as /devices/pci0000:00/0000:00:1c.6/0000:07:00.0/usb4/4-3/4-3:1.2/input/input16
[ 1027.602037] uvcvideo: Found UVC 1.10 device Intel RealSense 3D Camera R200 (8086:0a80)
[ 1027.603758] uvcvideo: Unable to create debugfs 4-2 directory.
[ 1027.603886] input: Intel RealSense 3D Camera R200 as /devices/pci0000:00/0000:00:1c.6/0000:07:00.0/usb4/4-3/4-3:1.4/input/input17
[ 1027.603960] usbcore: registered new interface driver uvcvideo
[ 1027.603962] USB Video Class driver (1.1.1-3-realsense)
[ 1027.620506] ------------[ cut here ]------------
[ 1027.620525] WARNING: CPU: 6 PID: 3614 at /build/linux-lts-xenial-OvqN2x/linux-lts-xenial-4.4.0/drivers/media/v4l2-core/v4l2-ioctl.c:1272 v4l_enum_fmt+0xd3e/0x10c0 [videodev]()
[ 1027.620534] Unknown pixelformat 0x2036315a
[ 1027.620535] Modules linked in: uvcvideo(OE) videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_core v4l2_common videodev media ipt_MASQUERADE nf_nat_masquerade_ipv4 iptable_nat nf_nat_ipv4 nf_nat nf_conntrack_ipv4 nf_defrag_ipv4 xt_conntrack nf_conntrack ipt_REJECT nf_reject_ipv4 xt_CHECKSUM iptable_mangle xt_tcpudp bridge stp llc xt_multiport ip6table_filter ip6_tables iptable_filter ip_tables ebtable_nat ebtables x_tables vmw_vsock_vmci_transport vsock vmw_vmci nvidia_drm(POE) nvidia_modeset(POE) nvidia(POE) input_leds snd_hda_codec_hdmi snd_hda_codec_realtek bnep rfcomm snd_hda_codec_generic bluetooth snd_hda_intel snd_hda_codec snd_hda_core snd_hwdep x86_pkg_temp_thermal intel_powerclamp coretemp kvm_intel kvm drm_kms_helper drm snd_pcm irqbypass crct10dif_pclmul crc32_pclmul ghash_clmulni_intel aesni_intel snd_seq_midi snd_seq_midi_event snd_rawmidi snd_seq snd_seq_device snd_timer snd aes_x86_64 lrw gf128mul glue_helper ablk_helper cryptd serio_raw sb_edac edac_core fb_sys_fops mei_me syscopyarea sysfillrect mei sysimgblt wmi soundcore shpchp 8250_fintek lpc_ich mac_hid parport_pc ppdev lp parport binfmt_misc nls_iso8859_1 hid_generic usbhid hid uas usb_storage isci psmouse e1000e libsas ahci ptp libahci scsi_transport_sas pps_core pata_acpi fjes
[ 1027.620615] CPU: 6 PID: 3614 Comm: AudioThread Tainted: P           OE   4.4.0-58-generic #79~14.04.1-Ubuntu
[ 1027.620617] Hardware name: LENOVO 98002825/LENOVO, BIOS A2KT44AUS 03/13/2014
[ 1027.620619]  0000000000000000 ffff88042279bbe8 ffffffff813dbd6c ffff88042279bc30
[ 1027.620623]  ffffffffc060bab0 ffff88042279bc20 ffffffff8107e9d6 ffff88042279bd98
[ 1027.620626]  0000000000000000 0000000000000000 ffff8800702cba00 ffffffffc060e180
[ 1027.620629] Call Trace:
[ 1027.620634]  [<ffffffff813dbd6c>] dump_stack+0x63/0x87
[ 1027.620638]  [<ffffffff8107e9d6>] warn_slowpath_common+0x86/0xc0
[ 1027.620640]  [<ffffffff8107ea5c>] warn_slowpath_fmt+0x4c/0x50
[ 1027.620645]  [<ffffffffc0646197>] ? uvc_ioctl_enum_fmt+0xb7/0x140 [uvcvideo]
[ 1027.620650]  [<ffffffffc05f580e>] v4l_enum_fmt+0xd3e/0x10c0 [videodev]
[ 1027.620655]  [<ffffffffc05f5dea>] __video_do_ioctl+0x25a/0x2e0 [videodev]
[ 1027.620658]  [<ffffffff812202e4>] ? mntput+0x24/0x40
[ 1027.620661]  [<ffffffff8120e539>] ? path_openat+0x2e9/0x12d0
[ 1027.620666]  [<ffffffffc05f83fb>] video_usercopy+0x21b/0x590 [videodev]
[ 1027.620670]  [<ffffffffc05f5b90>] ? v4l_enum_fmt+0x10c0/0x10c0 [videodev]
[ 1027.620673]  [<ffffffff812108a2>] ? do_filp_open+0x92/0xd0
[ 1027.620678]  [<ffffffffc05f8785>] video_ioctl2+0x15/0x20 [videodev]
[ 1027.620682]  [<ffffffffc05f1644>] v4l2_ioctl+0xa4/0xc0 [videodev]
[ 1027.620685]  [<ffffffff8121396d>] do_vfs_ioctl+0x2dd/0x4c0
[ 1027.620687]  [<ffffffff8120f983>] ? putname+0x53/0x60
[ 1027.620689]  [<ffffffff811ffbb8>] ? do_sys_open+0x1b8/0x270
[ 1027.620692]  [<ffffffff81213bc9>] SyS_ioctl+0x79/0x90
[ 1027.620695]  [<ffffffff81802c76>] entry_SYSCALL_64_fastpath+0x16/0x75
[ 1027.620697] ---[ end trace 5c33dafd59dc85f9 ]---
[ 1027.621064] ------------[ cut here ]------------
[... 1625 lines cut ...]
[ 1732.480419] uvcvideo: Failed to query (SET_CUR) UVC control 8 on unit 7: -110 (exp. 2).
[ 1732.828377] uvcvideo: Failed to query (SET_CUR) UVC control 8 on unit 7: -110 (exp. 2).
[ 1733.176338] uvcvideo: Failed to query (SET_CUR) UVC control 8 on unit 7: -110 (exp. 2).
[ 1733.524339] uvcvideo: Failed to query (SET_CUR) UVC control 8 on unit 7: -110 (exp. 2).
[ 1733.872334] uvcvideo: Failed to query (SET_CUR) UVC control 8 on unit 7: -110 (exp. 2).
[ 1734.220308] uvcvideo: Failed to query (SET_CUR) UVC control 8 on unit 7: -110 (exp. 2).
[ 1734.568282] uvcvideo: Failed to query (SET_CUR) UVC control 8 on unit 7: -110 (exp. 2).
[ 1734.916284] uvcvideo: Failed to query (SET_CUR) UVC control 8 on unit 7: -110 (exp. 2).
[ 1735.264181] uvcvideo: Failed to query (SET_CUR) UVC control 8 on unit 7: -110 (exp. 2).
[ 1735.564151] uvcvideo: Failed to query (SET_CUR) UVC control 11 on unit 7: -110 (exp. 1).
[ 1735.912193] uvcvideo: Failed to query (SET_CUR) UVC control 11 on unit 7: -110 (exp. 1).
[ 1736.260106] uvcvideo: Failed to query (SET_CUR) UVC control 11 on unit 7: -110 (exp. 1).
[ 1736.608078] uvcvideo: Failed to query (SET_CUR) UVC control 11 on unit 7: -110 (exp. 1).
[ 1736.956083] uvcvideo: Failed to query (SET_CUR) UVC control 11 on unit 7: -110 (exp. 1).
[ 1737.304021] uvcvideo: Failed to query (SET_CUR) UVC control 11 on unit 7: -110 (exp. 1).
[ 1737.651994] uvcvideo: Failed to query (SET_CUR) UVC control 11 on unit 7: -110 (exp. 1).
[ 1737.999928] uvcvideo: Failed to query (SET_CUR) UVC control 11 on unit 7: -110 (exp. 1).
[ 1738.347974] uvcvideo: Failed to query (SET_CUR) UVC control 11 on unit 7: -110 (exp. 1).
[ 1738.695943] uvcvideo: Failed to query (SET_CUR) UVC control 11 on unit 7: -110 (exp. 1).
[ 1739.043910] uvcvideo: Failed to query (SET_CUR) UVC control 11 on unit 7: -110 (exp. 1).
[ 1739.391877] uvcvideo: Failed to query (SET_CUR) UVC control 11 on unit 7: -110 (exp. 1).

[ 1849.821214] systemd-hostnamed[9833]: Warning: nss-myhostname is not installed. Changing the local hostname might make it unresolveable. Please install nss-myhostname!
[ 1864.474071] usb 4-3: Failed to disable LTM before suspend
```

laofanfan, 2017-01-09:
Hi, I don't sure, but is this the patch problem?
https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md

BlazingForests, 2017-01-09:
Hi, @laofanfan 

I am using `apt-get install ros-indigo-realsense-camera` to setup system.

Them will use DKMS to build the kernel.

I don't think there have issues.

THX

mkhansen-intel, 2017-01-18:
@BlazingForests - I'm not sure why RGBD is failing for you. We haven't seen this happen on R200. We have seen RGBD failure on SR300, and we're working on a fix for that, but I don't think this is the same problem. Could you checkout the 1.6.0 version and tell me if you see the same issue?

evanbeard, 2017-02-15:
+1 I'm seeing this exact same issue, i.e. same successful run of the roslaunch but dmesg is full of the "uvcvideo: Failed to query" message.

mkhansen-intel, 2017-03-13:
@BlazingForests - this may be related to issue #207, which I'm attempting to fix with PR #209. Would you like to pull & test my PR and see if that resolves your issue?

mattcurfman, 2017-04-25:
@BlazingForests any update regarding @mkhansen-intel's possible fix with PR #209?

dorodnic, 2017-08-21:
@BlazingForests @evanbeard please let us know if you still see the problem after the code-change. Also, if Xenial and Kinetic is an option for you, it is interesting how this would behave with slightly different Kernel. 
Will do our best to help investigate