realsenseai/realsense-ros #186: ZR300 IMU driver bug: Zero measurements in linear acceleration or angular velocity

andrepfr, 2017-01-09:
### System Configuration
| Version               | Your Configuration   |
|:--------------------- |:---------------------|
| Operating System      | Ubuntu 16.04.1 LTS |
| Kernel                | 4.4.0-57-generic  |
| ROS                   | kinetic |
| ROS RealSense         | 1.7.0 |
| librealsense          | 1.12.0  |
| Camera Type-Firmware  | Name: Intel RealSense ZR300, Camera FW: 2.0.71.28, Adapter FW: 1.29.0.0, Motion Module FW: 1.25.0.0 |

---

The realsense_camera driver seems to have a bug in the ZR300 IMU part. When looking at the IMU rostopic /camera/imu/data_raw (see Screenshot below), one can see that either all angular_velocities or all linear_acceleration values are zero. The gyro and the accelerometer seem to have different frame rates (250 HZ accelerometer, 200 Hz gyro ?). Since no buffering seems to be done, the /camera/imu/data_raw topic is published at a frequency of about 430 Hz containing always either zero angular_velocity or linear_acceleration values. This can be seen in the zr300_nodelet.cc where in case an angular velocity measurement arrives, the linear_acceleration is set to zero (L595) or vice versa the angular_velocity (L604).  

I think it would be desirable to have a flag to publish gyro and accelerometer on separate topics with their full frame rate or to publish one fused topic at e.g. 200 Hz always containing non zero linear_acceleration and angular velocity.

 ##
[image]

reaganlo, 2017-01-09:
The angular velocity and linear acceleration are each received asynchronously from the ZR300 camera and they are published as and when they are received without any buffering (to avoid any kind of latency). We decided not to have them in separate topics, based on the ROS IMU message standard. Refer to the comments at http://docs.ros.org/api/sensor_msgs/html/msg/Imu.html

One can determine the type of data that the IMU topic contains by referring to the element 0 of the associated covariance matrix. This has been documented at http://wiki.ros.org/realsense_camera.

>Each IMU topic message will either contain angular velocity or linear acceleration data. The element 0 of the associated covariance matrix will be set to -1 to indicate the absence of data. E.g. If the topic message does not contain angular velocity data, then the element 0 of the angular velocity covariance matrix will be set to -1.This topic message does not contain orientation data and hence the element 0 of the orientation covariance matrix will always be -1.

andrepfr, 2017-01-10:
Thanks a lot for the fast response. As a quick fix I would suggest to create a flag which specifies if gyro and accelerometer should be published on separate topics. In case they are published on one topic it would probably be accurate enough to just take the last available accelerometer measurement (faster sensor) at the time the gyro measurement (slower sensor) arrives and publish it together. 
Let me know if I could help with the implementation.

commit 8566f453 (2017-03-27): Updated code to cache ZR300 IMU data. Fixes #186.

With this change, the IMU data_raw msg will have values for
both angular velocity and linear acceleration.
E.g when the device returns angular velocity, the IMU data_raw
msg will contain the latest value of angular velocity and
previous value of linear acceleration.
commit 58757425 (2017-03-28): Updated code to cache ZR300 IMU data. Fixes #186.

With this change, the IMU data_raw msg will have values for
both angular velocity and linear acceleration.
E.g when the device returns only angular velocity, the IMU data_raw
msg will contain the latest value of angular velocity and
previous value of linear acceleration.
closed 2017-04-24 by mattcurfman