realsenseai/realsense-ros #187: Cannot retrieve ir/ir2/depth images from R200

tkornuta, 2017-01-12:
### System Configuration
| Version               | Your Configuration   |
|:--------------------- |:---------------------|
| Operating System      | Ubuntu 14.04.5 LTS                |
| Kernel                | 4.4.0-040400-generic                |
| ROS                   | indigo                    |
| ROS RealSense         | 1.7.0                |
| librealsense          | 1.12.1                |
| Camera Type-Firmware  | Serial No: 2481018566, USB Port ID: 4-5-4, Name: Intel RealSense R200, Camera FW: 1.0.72.06            |

### Expected Behavior
rosrun image_view image_view image:=/camera/depth/image_raw
should results in showing the actual depth image 

### Actual Behavior
roslaunch realsense_camera
seems to start properly, but I can only display the color image, whereas topics ir1/ir2/depth are empty (status warning: No Image received).

Color images are retrieved properly.

### Steps to Reproduce
turtlebot@turtlebot-workstation:~/turtlebot_ws$ roslaunch realsense_camera r200_nodelet_rgbd.launch 

... logging to /home/turtlebot/.ros/log/8617f792-d876-11e6-9242-94de80a0d161/roslaunch-turtlebot-workstation-31685.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://9.1.95.21:46744/

SUMMARY

PARAMETERS
 * /camera/camera_nodelet_manager/num_worker_threads: 4
 * /camera/depth_rectify_depth/interpolation: 0
 * /camera/disparity_depth/max_range: 4.0
 * /camera/disparity_depth/min_range: 0.5
 * /camera/disparity_registered_sw/max_range: 4.0
 * /camera/disparity_registered_sw/min_range: 0.5
 * /camera/driver/base_frame_id: camera_link
 * /camera/driver/camera_type: R200
 * /camera/driver/color_fps: 30
 * /camera/driver/color_frame_id: camera_rgb_frame
 * /camera/driver/color_height: 480
 * /camera/driver/color_optical_frame_id: camera_rgb_optica...
 * /camera/driver/color_width: 640
 * /camera/driver/depth_fps: 30
 * /camera/driver/depth_frame_id: camera_depth_frame
 * /camera/driver/depth_height: 360
 * /camera/driver/depth_optical_frame_id: camera_depth_opti...
 * /camera/driver/depth_width: 480
 * /camera/driver/enable_color: True
 * /camera/driver/enable_depth: True
 * /camera/driver/enable_fisheye: False
 * /camera/driver/enable_imu: False
 * /camera/driver/enable_ir2: True
 * /camera/driver/enable_ir: True
 * /camera/driver/enable_pointcloud: False
 * /camera/driver/enable_tf: True
 * /camera/driver/ir2_frame_id: camera_ir2_frame
 * /camera/driver/ir_frame_id: camera_ir_frame
 * /camera/driver/mode: manual
 * /camera/driver/serial_no: 
 * /camera/driver/usb_port_id: 
 * /camera/points_xyzrgb_sw_registered/queue_size: 100
 * /rosdistro: indigo
 * /rosversion: 1.11.20

NODES
  /camera/
    camera_nodelet_manager (nodelet/nodelet)
    depth_metric (nodelet/nodelet)
    depth_metric_rect (nodelet/nodelet)
    depth_points (nodelet/nodelet)
    depth_rectify_depth (nodelet/nodelet)
    depth_registered_sw_metric_rect (nodelet/nodelet)
    disparity_depth (nodelet/nodelet)
    disparity_registered_sw (nodelet/nodelet)
    driver (nodelet/nodelet)
    ir_rectify_ir (nodelet/nodelet)
    points_xyzrgb_sw_registered (nodelet/nodelet)
    register_depth_rgb (nodelet/nodelet)
    rgb_debayer (nodelet/nodelet)
    rgb_rectify_color (nodelet/nodelet)
    rgb_rectify_mono (nodelet/nodelet)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[camera/camera_nodelet_manager-1]: started with pid [31703]
process[camera/driver-2]: started with pid [31704]
process[camera/rgb_debayer-3]: started with pid [31705]
process[camera/rgb_rectify_mono-4]: started with pid [31706]
process[camera/rgb_rectify_color-5]: started with pid [31714]
process[camera/ir_rectify_ir-6]: started with pid [31732]
process[camera/depth_rectify_depth-7]: started with pid [31747]
process[camera/depth_metric_rect-8]: started with pid [31759]
process[camera/depth_metric-9]: started with pid [31764]
process[camera/depth_points-10]: started with pid [31767]
process[camera/register_depth_rgb-11]: started with pid [31786]
process[camera/points_xyzrgb_sw_registered-12]: started with pid [31792]
process[camera/depth_registered_sw_metric_rect-13]: started with pid [31806]
process[camera/disparity_depth-14]: started with pid [31822]
process[camera/disparity_registered_sw-15]: started with pid [31828]
[ INFO] [1484191469.332627741]: Initializing nodelet with 4 worker threads.
[ INFO] [1484191469.789052634]: /camera/driver - Detected the following cameras:
				- Serial No: 2481018566, USB Port ID: 4-5-4, Name: Intel RealSense R200, Camera FW: 1.0.72.06
[ INFO] [1484191469.789098555]: /camera/driver - Connecting to camera with Serial No: 2481018566, USB Port ID: 4-5-4
[ INFO] [1484191470.178591469]: /camera/driver - Setting static camera options
[ INFO] [1484191470.184837443]: /camera/driver - Enabling Depth in manual mode
[ INFO] [1484191470.185064867]: /camera/driver - Enabling Color in manual mode
[ INFO] [1484191470.185299491]: /camera/driver - Enabling IR in manual mode
[ INFO] [1484191470.185380799]: /camera/driver - Enabling IR2 in manual mode
[ INFO] [1484191470.185456870]: /camera/driver - Starting camera
[ INFO] [1484191470.211437144]: /camera/driver - Publishing camera transforms (/tf_static)
[ INFO] [1484191470.211594380]: /camera/driver - Setting dynamic camera options (r200_dc_preset=5)
[ INFO] [1484191470.962114475]: /camera/driver - Initializing Depth Control Preset to 5
[ INFO] [1484191472.385715978]: /camera/driver - Setting dynamic camera options

mdhorn, 2017-01-12:
The kernel version you are reporting is not supported by the ROS librealsense package.

Please remove that kernel and use the default kernel which is installed with Ubuntu 14.04.5 LTS.

After moving the the default kernel (currently 4.4.0-47-generic), you may have to force reinstalled the ROS librealsense package to ensure the kernel is properly patched:
`sudo apt-get --reinstall install ros-indigo-librealsense`

After the update and reinstall of librealsense package, please retest.

skohlbr, 2017-01-12:
I also noted not getting IR images using the ROS driver via topics, but the depth stream and point cloud working (indicating that both streams are actually working internally). Is this what you're seeing also?

mdhorn, 2017-01-12:
@skohlbr If you are using the default launch file `roslaunch realsense_camera r200_nodelet_default.launch` then you will not get IR streams as they are disable by default since release 1.6.1. The IR streams do not need to be enabled in order to get the depth stream -- most user were no using the IR streams directly and they were causing a lot of unnecessary USB traffic.

tkornuta, 2017-01-13:
Hi Mark,

My problem resulted from the fact that I have tried to install R200 on Turtlebot, that I have recently bought. Clearpath delivered it along with an ACER laptop with Ubuntu 14.04... with kernel 3.13.xxx:] So first I have upgraded the kernel to the newest version 4.4.59, and THEN followed the instructions from
https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md
and tried to "update to 4.4 Stable Kernel" with the script:
./scripts/install_dependencies-4.4.sh

Anyway, today I've decided to reinstall the system, used Clearpath Turtlebot Ubuntu image and afterwards strictly followed your instructions (well, I have also upgraded the whole ROS after kernel upgrade) and it looks like I have succeeded - I have Turtlebot with three sensors (Kinect, Orbbec and R200) working!

For now, I did that on a stationary machine, but it is a good sign that it will also work for the ACER laptop, right? ;)

Thanks!
Tom

mdhorn, 2017-01-13:
@tkornuta Great news. Yes, following the same instructions should work on any ubuntu platform.

If you are using the RealSense cameras with ROS, then you should only ROS instructions and ignore all of the install directions from the github site for librealsense. We are maintaining a different branch now in their repo (rosdebian) to ensure cameras install properly under ROS and Ubuntu.

Now that you're working, I'm going to close this Issue.

closed 2017-01-13 by mdhorn