realsenseai/realsense-ros #188: API version mismatch: librealsense.so was compiled with API version 1.12.0 but the application was compiled with 1.11.0!  clear solution please.

Tondy, 2017-01-12:
### System Configuration
Please complete Your Configuration detail below. Refer to the BKC at [Configuration](../realsense_camera/README.md#configuration).

| Version               | Your Configuration   |
|:--------------------- |:---------------------|
| Operating System      | UBUNTU  14.04 LTS      |
| Kernel                | 3.13.0-77-generic   |
| ROS                   | indigo                  |
| ROS RealSense         | 1.7.0               |
| librealsense          |1.12.0              |
| Camera Type-Firmware  | R200           |

---
So I recently purchase a r200 realsense camera . After going through the installation 
process  and running roslaunch realsense_camera r200_nodelet_default.launch

### Actual Behavior
"
[ERROR] [1484203192.776597367]: /camera/driver - No cameras detected!
[ERROR] [1484203192.777222376]: /camera/driver - Error calling rs_create_context ( api_version:11100 ): 
API version mismatch: librealsense.so was compiled with API version 1.12.0 but the application was compiled with 1.11.0! Make sure correct version of the library is installed (make install) 

[FATAL] [1484203192.851873416]: Failed to load nodelet '/camera/driver` of type `realsense_camera/R200Nodelet` to manager `nodelet_manager'
[camera/driver-3] process has died [pid 9227, exit code 255, cmd /opt/ros/indigo/lib/nodelet/nodelet load realsense_camera/R200Nodelet nodelet_manager depth:=depth color:=color ir:=ir ir2:=ir2 fisheye:=fisheye imu:=imu __name:=driver __log:=/home/tondy/.ros/log/ea1b5066-d891-11e6-a3a5-b8aeed71526c/camera-driver-3.log].
log file: /home/tondy/.ros/log/ea1b5066-d891-11e6-a3a5-b8aeed71526c/camera-driver-3*.log"

### Steps to  Reproduce roslaunch

 "realsense_camera r200_nodelet_default.launch"

mdhorn, 2017-01-12:
Not sure which directions you used, but you need a newer kernel -- at least 4.2.x kernel which was released with 14.0.4: https://wiki.ubuntu.com/Kernel/Support#A14.04.x_Ubuntu_Kernel_Support

Please update you 14.04 distribution to have a newer kernel, then follow these directions:
http://wiki.ros.org/librealsense#Installation

mdhorn, 2017-01-24:
Was this resolved by updating the kernel?

Tondy, 2017-01-24:
Hello 

I blow a few circuits and had to order replacements. I am going to try it in the coming day or two.

Kind regards.

Tondy

Tondy, 2017-01-28:
Hello

Finally got a new power supply.

So I ran the command below to get the latest kernel as recommended http://askubuntu.com/questions/753369/latest-kernel-for-ubuntu-14-04-4

sudo apt-get install linux-generic-lts-wily

I ran 

wget -O enable_kernel_sources.sh http://bit.ly/en_krnl_src
bash ./enable_kernel_sources.sh

and

sudo apt-get update

sudo apt-get --reinstall install 'ros-indigo-librealsense'

As I thought this would make it work. I went back to ROS and ran.

roslaunch realsense_camera r200_nodelet_default.launch

pretty much same error.

[ERROR] [1485642568.173746563]: /camera/driver - No cameras detected!
[ERROR] [1485642568.174190311]: /camera/driver - Error calling rs_create_context ( api_version:11100 ): 
API version mismatch: librealsense.so was compiled with API version 1.12.0 but the application was compiled with 1.11.0! Make sure correct version of the library is installed (make install) 

[FATAL] [1485642568.292455714]: Failed to load nodelet '/camera/driver` of type `realsense_camera/R200Nodelet` to manager `nodelet_manager'
[camera/driver-3] process has died [pid 10315, exit code 255, cmd /opt/ros/indigo/lib/nodelet/nodelet load realsense_camera/R200Nodelet nodelet_manager depth:=depth color:=color ir:=ir ir2:=ir2 fisheye:=fisheye imu:=imu __name:=driver __log:=/home/tondy/.ros/log/38a8e94a-e5a9-11e6-945e-b8aeed71526c/camera-driver-3.log].
log file: /home/tondy/.ros/log/38a8e94a-e5a9-11e6-945e-b8aeed71526c/camera-driver-3*.log

Running uname -r 

Shows this.

3.13.0-77-generic

Hope you can help.

Tondy, 2017-02-01:
Hope you can give some feedback on this.

Cheers.

mdhorn, 2017-02-06:
@Tondy If `uname -r` is still showing the old 3.x kernel, you need to reboot your system.
It would be preferable to use the xenial kernel which you should just get by updating to the latest release of 14.04.
`sudo apt-get dist-upgrade`
That step should update to the latest xenial kernel after a reboot.

Tondy, 2017-02-07:
Hi

After upgrade.

uname -r

output >
4.4.0-59-generic

sudo apt-get --reinstall install 'ros-indigo-librealsense'

then > roslaunch realsense_camera r200_nodelet_default.launch

output >

[ERROR] [1486433780.723652627]: /camera/driver - No cameras detected!
[ERROR] [1486433780.723973303]: /camera/driver - Error calling rs_create_context ( api_version:11100 ): 
API version mismatch: librealsense.so was compiled with API version 1.12.0 but the application was compiled with 1.11.0! Make sure correct version of the library is installed (make install) 

[FATAL] [1486433780.913704969]: Failed to load nodelet '/camera/driver` of type `realsense_camera/R200Nodelet` to manager `nodelet_manager'
[camera/driver-2] process has died [pid 27662, exit code 255, cmd /opt/ros/indigo/lib/nodelet/nodelet load realsense_camera/R200Nodelet nodelet_manager depth:=depth color:=color ir:=ir ir2:=ir2 fisheye:=fisheye imu:=imu __name:=driver __log:=/home/tondy/.ros/log/7240172c-ecd5-11e6-84a6-d8fc930e5344/camera-driver-2.log].
log file: /home/tondy/.ros/log/7240172c-ecd5-11e6-84a6-d8fc930e5344/camera-driver-2*.log

please help

mdhorn, 2017-02-07:
@Tondy The released versions of ros-indigo-librealsense and ros-indigo-realsense-camera will always work together.

1. Ensure you have correct version of ros-indigo-realsense-camea
`sudo apt-get --reinstall install ros-indigo-realsense-camera`

2. Ensure you are ONLY source the system ROS setup.bash file, and not sourcing the setup.bash from any locally built versions of either package.
`source /opt/ros/indigo/setup.bash`
`echo $ROS_PACKAGE_PATH`
should yield
`/opt/ros/indigo/share:/opt/ros/indigo/stacks`

3. Verify that the camera is visible on the USB bus:
`lsusb  | grep 8086`
should yield
`Bus 003 Device 007: ID 8086:0a80 Intel Corp.`
though the bus number and device number will vary per system and USB port.

4. If all these are working, then try to launch the node:
`roslaunch realsense_camera r200_nodelet_default.launch`

Tondy, 2017-02-09:
Hello

Many thanks it works. 

source /opt/ros/indigo/setup.bash

Made it work. I usually use /devel/setup.bash . I need this to work with other package which now stopped
working outputting

[standalone.launch] is neither a launch file in package [bringup] nor is [bringup] a launch file name The traceback for the exception was written to the log file

If I went back to do devel/setup.bash . I believe it would work.

What can I do to have both working with source /opt/ros/indigo/setup.bash
That would be the last question.

Hope you can help
.

Cheers.

mdhorn, 2017-02-09:
@Tondy 

If you don't have a local copy of realsense_camera nor librealsense in your workspace, then you should be able to source and user your local devel/setup.bash.
My guess is that you have a version there that is older and the reason you had the version mismatch.

closed 2017-02-15 by mdhorn

Tondy, 2017-04-15:
Hello Thank you again.

So I deleted a realsense_camera folder in my src folder which was an old version git cloned a while ago . keeping the realsense camera installed by using sudo apt-get  install ros-indigo-realsense-camera  command. As most of my projects are in the src folder.

So when I source /opt/ros/indigo/setup.bash  and run. 

roslaunch realsense_camera r200_nodelet_default.launch Works fine showing that real sense is working.

My other projects in src folder stop working. giving me the error:

[robot_standalone.launch] is neither a launch file in package [chefbot_bringup] nor is [chefbot_bringup] a launch file name.The traceback for the exception was written to the log file

As a result combining my projects code with realsense is not working . 

My /src folders code works if I source devel/setup.bash but realsense will stops work.

Hope you can help.

Kind regards.

Tondy

Tondy, 2017-04-24:
Hello

Hope you might still be able to help

Cheers.