realsenseai/realsense-ros #190: Depth and IR distortion parameters inverse

fdone, 2017-01-13:
### System Configuration

| Version               | Your Configuration   |
|:--------------------- |:---------------------|
| Operating System      | Ubuntu 16.04                |
| Kernel                |   4.4.0-59-generic  |
| ROS                   | kinetic       |
| ROS RealSense         |  1.7.0 |
| librealsense          | master 0f618cd4dcf1c8805487e80544652625e7043a34 Thu Jan 5 14:07:30    |
| Camera Type-Firmware  | F200   2.60.0.0  |

### Expected Behavior
Viewing the /ir/image_rect_ir and /depth/image_rect topics of a F200 camera. I expect the IR and depth image to be undistorted.

Expected behavior from a different program:
[image]

[image]

### Actual Behavior
Looking at the /ir/image_rect_ir and /depth/image_rect of a F200 camera. It seems that the IR and depth images are more distorted than the raw images.

After a bit of investigation it turns out the distortion matrix D in /ir/camera_info and /depth/camera_info is publishing the inverse distortion parameters.

Parameters read directly from the cameras:
`
 distc[5] forwad distortion parameters of the ir camera
  -0.148355 -0.029984 -0.003011 0.000410 -0.077241

 invdistc[5] the inverse distortion parameters of the irt camera
  0.151655 0.026701 0.003516 -0.000494 0.144765
`

Parameters published on ir/depth camera_info:
`
header: 
  seq: 275
  stamp: 
    secs: 1484266588
    nsecs: 567394693
  frame_id: camera_ir_optical_frame
height: 480
width: 640
distortion_model: plumb_bob
D: [0.15165534615516663, 0.026701051741838455, 0.003515728749334812, -0.000494183914270252, 0.14476454257965088]
K: [477.1851806640625, 0.0, 328.41302490234375, 0.0, 477.18505859375, 245.83731079101562, 0.0, 0.0, 1.0]
R: [1.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 1.0]
P: [477.1851806640625, 0.0, 328.41302490234375, 0.0, 0.0, 477.18505859375, 245.83731079101562, 0.0, 0.0, 0.0, 1.0, 0.0]
binning_x: 0
binning_y: 0
roi: 
  x_offset: 0
  y_offset: 0
  height: 0
  width: 0
  do_rectify: False

`

Screenshot with current behavior:
[image]

I am not quite sure if the problem is in librealsense or ros realsense. But it would be nice to have the correct distortion parameters published so other standard nodes behave properly.

### Steps to Reproduce
roslaunch realsense_camera f200_array_rgbd.test and look at the rectified topics and camera_info topics.

r3n33, 2017-01-17:
I can confirm if you invert the coefficients the F200's distortion model is corrected. Thank you @fdone 

https://github.com/intel-ros/realsense/blob/indigo-devel/realsense_camera/src/base_nodelet.cpp#L741

Before
[image]
After
[image]

Isabelp, 2017-07-11:
Hi,

I am having the same issue, the ir rectified image presents distortion, therefore I assume the point clouds I get also will have distortion. Could you tell me how did you read the parameters directly from the camera? 
I would like to compare them with the ones published in camera_info too.

RealSenseCustomerSupport, 2019-03-07:
----------------------------------------------

[Please Ignore - RealSense system comment]

RealSenseCustomerSupport, 2019-03-08:
----------------------------------------------

@ghost F200 is EOL for long time. Please change to the latest supported D400 cameras. Thanks!

closed 2019-03-19 by RealSenseCustomerSupport