realsenseai/realsense-ros #192: Unable to lower fps on R200

hakanrobo, 2017-01-16:
### System Configuration

| Version               | Your Configuration   |
|:--------------------- |:---------------------|
| Operating System      | Ubuntu 16.04                |
| Kernel                | ?-?-?                |
| ROS                   | Kinetic                    |
| ROS RealSense         | 1.7.0                |
| librealsense          | v1.11.0                |
| Camera Type-Firmware  | R200-1.0.72.10            |

---

### Expected Behavior

I would like to run the driver for the R200 at a lower fps, preferably 10 instead of the default 30. However, it seems like librealsense does not support this. The behavior is also verified at:
https://github.com/IntelRealSense/librealsense/blob/master/doc/supported_video_formats.pdf

Is this really true? What makes librealsense limit the fps? I would prefer not to downsample through my application, it would be alot nicer just to make the driver spit of data at a predefined frequency.

### Actual Behavior
`[ERROR] [1484576390.032735761]: /camera/driver - Verify camera firmware version and/or calibration data!
[ERROR] [1484576390.032806853]: /camera/driver - Error calling rs_get_stream_intrinsics ( device:0x6730e0, stream:DEPTH, intrin:0x7fe678b890 ): 
uvc subdevice 1 cannot provide 480x360:Z16@10Hz DEPTH`

### Steps to Reproduce

Change fps in the launch file

kevincwells, 2017-01-19:
Hi @hakanrobo,

The short answer is yes, that minimum frame rate is a limitation enforced by the librealsense library itself, not the ROS package exposing it. More specifically, this limitation is hard-coded in the [device mode definitions](https://github.com/IntelRealSense/librealsense/blob/master/src/ds-device.cpp) in the base class and camera-specific class definitions.  For a more in-depth explanation as to _why_ these limitations are enforced, or to request that other frame rates be supported in the future, I encourage you to file a separate issue on the [librealsense issues page](https://github.com/IntelRealSense/librealsense/issues).

While a hardware solution would be desirable for overall system efficiency, I would like to point you to the [topic_tools throttle node](http://wiki.ros.org/topic_tools/throttle) as a possible solution to your problem.  It is designed to efficiently throttle high-bandwidth streams either by bandwidth or message rate, allowing your application to simply subscribe to the downsampled stream.  People commonly use it with camera feeds, especially if that feed is going to be broadcast over WiFi or the like.

Cheers,
Kevin

closed 2017-02-02 by kevincwells

djiglesias, 2017-11-22:
I just added a software filter for throttling the frame rate on Realsense R200 camera streams in ROS. I forked Intel's Realsense repository and added the change to branch `software-throttle` [HERE](https://github.com/djiglesias/realsense). You can access the throttle in two ways.

## Dynamic Reconfigure
You can launch the R200 using either the default or rgbd launch file and then launch the `rqt_reconfigure` tool in a new window. Navigate to `driver` in the explorer window and enable throttling by selecting `enable_throttle` and then select the desired rate (0-60 fps) on the desired stream.
```
$ roslaunch realsense_camera r200_nodelet_default.launch 
$ rosrun rqt_reconfigure rqt_reconfigure
```
## ROS Launch
When launching the file, add `enable_throttle:=true` to enable the throttling behaviour (off by default), and then select the frame rate for the {color/depth/infra/fisheye} streams using `throttle_{stream}:={rate}`. Note that the rate is an integer.
```
$ roslaunch realsense_camera r200_nodelet_default.launch enable_throttle:=true throttle_color:=5
```
## Compiling Locally
Pull the repo into your Catkin workspace and compile. I am running ROS Kinetic Ubuntu 16.04 and ROS Indigo on Ubuntu 14.04 without problems. Let me know if you hit any snags. Builds and runs from terminal and Eclipse IDE.
```
$ cd ~/catkin_ws/src
$ git clone https://github.com/djiglesias/realsense
$ cd ~/catkin_ws
$ catkin_make
```