realsenseai/realsense-ros #194: SR3000 nodelet stuck on "/camera/driver - Setting dynamic camera options"

dbolkensteyn, 2017-01-23:
### System Configuration
| Version               | Your Configuration   |
|:--------------------- |:---------------------|
| Operating System      | Ubuntu 16.04.1 LTS                |
| Kernel                | 4.4.0-59-generic                |
| ROS                   | kinetic                    |
| ROS RealSense         | 1.7.0                |
| librealsense          | 1.12.1                |
| Camera Type-Firmware  | Name: Intel RealSense SR300, Camera FW: 3.21.0.0            |

### Expected Behavior
The RealSense ROS node should successfully initialize the camera and startup.

### Actual Behavior
```
SUMMARY
========

PARAMETERS
 * /camera/camera_nodelet_manager/num_worker_threads: 4
 * /camera/depth_rectify_depth/interpolation: 0
 * /camera/disparity_depth/max_range: 4.0
 * /camera/disparity_depth/min_range: 0.5
 * /camera/disparity_registered_sw/max_range: 4.0
 * /camera/disparity_registered_sw/min_range: 0.5
 * /camera/driver/base_frame_id: camera_link
 * /camera/driver/camera_type: SR300
 * /camera/driver/color_fps: 30
 * /camera/driver/color_frame_id: camera_rgb_frame
 * /camera/driver/color_height: 480
 * /camera/driver/color_optical_frame_id: camera_rgb_optica...
 * /camera/driver/color_width: 640
 * /camera/driver/depth_fps: 30
 * /camera/driver/depth_frame_id: camera_depth_frame
 * /camera/driver/depth_height: 480
 * /camera/driver/depth_optical_frame_id: camera_depth_opti...
 * /camera/driver/depth_width: 640
 * /camera/driver/enable_color: True
 * /camera/driver/enable_depth: True
 * /camera/driver/enable_fisheye: False
 * /camera/driver/enable_imu: False
 * /camera/driver/enable_ir2: True
 * /camera/driver/enable_ir: True
 * /camera/driver/enable_pointcloud: False
 * /camera/driver/enable_tf: True
 * /camera/driver/ir2_frame_id: camera_ir2_frame
 * /camera/driver/ir_frame_id: camera_ir_frame
 * /camera/driver/mode: manual
 * /camera/driver/serial_no: 
 * /camera/driver/usb_port_id: 
 * /camera/points_xyzrgb_sw_registered/queue_size: 100
 * /rosdistro: kinetic
 * /rosversion: 1.12.6

NODES
  /camera/
    camera_nodelet_manager (nodelet/nodelet)
    depth_metric (nodelet/nodelet)
    depth_metric_rect (nodelet/nodelet)
    depth_points (nodelet/nodelet)
    depth_rectify_depth (nodelet/nodelet)
    depth_registered_sw_metric_rect (nodelet/nodelet)
    disparity_depth (nodelet/nodelet)
    disparity_registered_sw (nodelet/nodelet)
    driver (nodelet/nodelet)
    ir_rectify_ir (nodelet/nodelet)
    points_xyzrgb_sw_registered (nodelet/nodelet)
    register_depth_rgb (nodelet/nodelet)
    rgb_debayer (nodelet/nodelet)
    rgb_rectify_color (nodelet/nodelet)
    rgb_rectify_mono (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [7645]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to c788df16-e190-11e6-a842-7c5cf8c484a7
process[rosout-1]: started with pid [7658]
started core service [/rosout]
process[camera/camera_nodelet_manager-2]: started with pid [7675]
process[camera/driver-3]: started with pid [7676]
process[camera/rgb_debayer-4]: started with pid [7677]
process[camera/rgb_rectify_mono-5]: started with pid [7690]
process[camera/rgb_rectify_color-6]: started with pid [7703]
process[camera/ir_rectify_ir-7]: started with pid [7716]
process[camera/depth_rectify_depth-8]: started with pid [7724]
process[camera/depth_metric_rect-9]: started with pid [7745]
[ INFO] [1485192262.754945313]: Initializing nodelet with 4 worker threads.
process[camera/depth_metric-10]: started with pid [7780]
process[camera/depth_points-11]: started with pid [7789]
process[camera/register_depth_rgb-12]: started with pid [7791]
process[camera/points_xyzrgb_sw_registered-13]: started with pid [7804]
process[camera/depth_registered_sw_metric_rect-14]: started with pid [7822]
process[camera/disparity_depth-15]: started with pid [7850]
process[camera/disparity_registered_sw-16]: started with pid [7866]
[ INFO] [1485192263.403944165]: /camera/driver - Detected the following cameras:
				- Serial No: 612205003278, USB Port ID: 6-2-1, Name: Intel RealSense SR300, Camera FW: 3.21.0.0
[ INFO] [1485192263.404075553]: /camera/driver - Connecting to camera with Serial No: 612205003278, USB Port ID: 6-2-1
[ INFO] [1485192263.714832995]: /camera/driver - Setting static camera options
[ INFO] [1485192263.716269810]: /camera/driver - Enabling Depth in manual mode
[ INFO] [1485192263.716474639]: /camera/driver - Enabling Color in manual mode
[ INFO] [1485192263.716727637]: /camera/driver - Enabling IR in manual mode
[ INFO] [1485192263.716904901]: /camera/driver - Starting camera
[ INFO] [1485192263.733345093]: /camera/driver - Publishing camera transforms (/tf_static)
[ INFO] [1485192263.733725827]: /camera/driver - Setting dynamic camera options
```

It stays stuck on {{/camera/driver - Setting dynamic camera options}} forever.

Note that it used to work a few days ago, I don't really know what changed.
Also note that the Raw Stream example from the RealSense SDK is correctly working on Windows using the same camera, same USB port and cable.

### Steps to Reproduce
{{roslaunch realsense_camera sr300_nodelet_rgbd.launch}}

mdhorn, 2017-01-23:
@dbolkensteyn Why do you think it is "stuck"?
Are the topics not being published?

dbolkensteyn, 2017-01-23:
Wow, indeed my bad @mdhorn !

I was so convinced it was stuck that I didn't even launch rviz to check, I even built librealsense to see if the examples their were working...

Sorry for the noise

closed 2017-01-23 by dbolkensteyn

whyou5945, 2017-07-23:
Hey, I meet the same problem on R200 while I run "roslaunch realsense_camera r200_nodelet_default.launch". And it stuck on  /camera/driver - Setting dynamic camera options. Since I m a totally newbee, so I still don't know how to fix it after I see this issue, so could you please tell me how to fix it? 
Thank you very much!