realsenseai/realsense-ros #195: Registered xyzrgb point cloud stops working after couple of secs + Compiling the source code is not working

mehdish89, 2017-01-23:
### System Configuration

| Version                            | Your Configuration                            |
|:-------------------------------- |:-------------------------------------------------|
| Operating System           | Ubuntu 14.04.5 LTS                          |
| Kernel                             | 4.4.0-040400-generic                        |
| ROS                                | indigo                                                |
| ROS RealSense             | 1.6.0                                                  |
| librealsense                    | 1.11.1                                                |
| Camera Type-Firmware  | Intel RealSense SR300 - 3.10.10.0  |

---

### Expected Behavior

### Steps to Reproduce

Installed ros-indigo-librealsense and ros-indigo-realsense-camera. Running the launch file "realsense_camera/sr300_nodelet_rgbd.launch", the camera works fine and rgb and depth stream are working. 

### Actual Behavior

Only the registered point cloud(xyzrgb) topic stops getting messages after a while, like 20 secs, even though the point cloud is still getting the messages and working fine. 

I found a similar issue, addressed in #25 , was resolved by some changes in the 'realsense_camera' source code that extrapolates the point cloud message time-stamps into the future. I tried to do so, compiling the package from source, and after it compiled successfully, running the launch file "realsense_camera/sr300_nodelet_rgbd.launch", I get this error that says nodelet manager fails to load the SR300Nodelet.

> [ERROR] [1485194503.220707883]: Failed to load nodelet [/driver] of type [realsense_camera/SR300Nodelet] even after refreshing the cache: MultiLibraryClassLoader: Could not create object of class type realsense_camera::SR300Nodelet as no factory exists for it. Make sure that the library exists and was explicitly loaded through MultiLibraryClassLoader::loadLibrary()
[ERROR] [1485194503.220751435]: The error before refreshing the cache was: Failed to load library /home/lab/catkin_ws/devel/lib//librealsense_camera_nodelet.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = /home/lab/catkin_ws/devel/lib//librealsense_camera_nodelet.so: undefined symbol: _ZN2cv6String10deallocateEv)

dbolkensteyn, 2017-01-24:
I am facing the same issue.

The amount of seconds I am getting messages on `/camera/depth_registered/points` seems to depend on the `queue_size:=10000` argument passed on to `sr300_nodelet_rgbd.launch`. With the default of 100, I am getting about 600 messages. With a value of 10000, I am beyond 6000 received messages (so it seems to be a good workaround) however it is just a matter of time before it gets stuck.

I also reproduce this without rviz, simply using `rostopic echo /camera/depth_registered/points`.

mdhorn, 2017-01-24:
@mehdish89 Build from source should be working. Please take a look at the tutorial about building from source: http://wiki.ros.org/realsense_camera/Tutorials

mdhorn, 2017-01-24:
The problem with the `/camera/depth_registered/points` topic seems like it might be related to either the timestamps on the input topics, or maybe the synchronization filter used. 
We have seen increasing the queue_size as "work around", but also setting the synchronization filter to be `exact_sync` seems to help.

```
index 084390222b4e..e6ca7097340f 100644
--- a/realsense_camera/launch/sr300_nodelet_rgbd.launch
+++ b/realsense_camera/launch/sr300_nodelet_rgbd.launch
@@ -32,7 +32,9 @@
        If the depth_registered pointcloud topic generated by the depth_image_proc package does not contain data,
        increase this value further. -->
   <arg name="queue_size"       default="100" />
-  <param name="$(arg camera)/points_xyzrgb_sw_registered/queue_size" type="int" value="$(arg queue_size)" />
+  <arg name="exact_sync"       default="true" />
+  <param name="$(arg camera)/points_xyzrgb_sw_registered/queue_size" type="int"  value="$(arg queue_size)" />
+  <param name="$(arg camera)/points_xyzrgb_sw_registered/exact_sync" type="bool" value="$(arg exact_sync)" />
 
   <!-- Optionally suppress loading the driver nodelet. Useful while playing back recorded raw data from a bag.-->
   <arg name="load_driver"   default="true" />
```

In general, systems with less resources (RAM and cores) seem to stop publishing sooner.

This still needs more investigation. Inputs, insights, and additional data welcome.

dbolkensteyn, 2017-02-03:
FYI I've just tested the R200 camera (the older model), and it does not seem to be affected by this issue. I received over 70'000 registered depth messages from it with its default queue_size argument value of 100.

dbolkensteyn, 2017-02-03:
Adding `exact_sync` isn't helping much. I also see message transmission getting interrupted for a while. For example it successfully receives 37 messages, then stalls for 30-60 seconds, then another 30 messages are received, etc. They seem to come in batch of ~33.

dbolkensteyn, 2017-02-03:
@mdhorn Is this issue related to #47 and #89? Will the pointcloud generation code be moved into a separate ROS node (and Git repo) and communicate with this package only through ROS topics?

jonpol01, 2017-02-06:
We actually encountered this problem(20~30mins to die) on R200 on intel joule environment.
If the said low spec env only have this issues..
Then I am hoping someone can answer for a workaround.

arpit15, 2017-03-24:
@mdhorn Is there any update on this issue? Compiling from source create errors for me.

mdhorn, 2017-03-24:
@arpit15 Can you provide details of the platform you are using to compile: OS, version of ROS, etc.
What compile error are you seeing?

The code on indigo-devel always compiles for the default ROS platforms, so I'm curious what problem you are seeing.

arpit15, 2017-03-24:
OS: Ubuntu 14.04
ROS version: Indigo
The code is able to compile. However I get runtime errors

[ERROR] [1490328431.571617457]: Failed to load nodelet [/camera/driver] of type [realsense_camera/SR300Nodelet] even after refreshing the cache: MultiLibraryClassLoader: Could not create object of class type realsense_camera::SR300Nodelet as no factory exists for it. Make sure that the library exists and was explicitly loaded through MultiLibraryClassLoader::loadLibrary()
[ERROR] [1490328431.571663617]: The error before refreshing the cache was: Failed to load library /home/arpit/ros_wss/exp_ws/devel/lib//librealsense_camera_nodelet.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = /home/arpit/ros_wss/exp_ws/devel/lib//librealsense_camera_nodelet.so: undefined symbol: _ZN2cv6String10deallocateEv)
[FATAL] [1490328431.571806916]: Failed to load nodelet '/camera/driver` of type `realsense_camera/SR300Nodelet` to manager `camera_nodelet_manager'
[camera/driver-3] process has died [pid 8069, exit code 255, cmd /opt/ros/indigo/lib/nodelet/nodelet load realsense_camera/SR300Nodelet camera_nodelet_manager depth:=depth color:=rgb ir:=ir ir2:=ir2 __name:=driver __log:=/home/arpit/.ros/log/5a6d8b94-1047-11e7-95fe-4ccc6a67057c/camera-driver-3.log].
log file: /home/arpit/.ros/log/5a6d8b94-1047-11e7-95fe-4ccc6a67057c/camera-driver-3*.log

mdhorn, 2017-03-24:
@arpit15 Which were of gcc are you using?
 `gcc -v`

Does the standard ROS release package correctly on your system?
`sudo apt-get update ; sudo apt-get install ros-indigo-realsense-camera`

Try clearing your build environment and rebuild from scratch:
`cd /home/arpit/ros_wss/exp_ws`
`rm -rf build*/ devel*/ logs/ .catkin_*`
` catkin_make -DCMAKE_BUILD_TYPE=RelWithDebInfo`

arpit15, 2017-03-25:
@mdhorn 
gcc specs are as follows

Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/4.8/lto-wrapper
Target: x86_64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Ubuntu 4.8.4-2ubuntu1~14.04.3' --with-bugurl=file:///usr/share/doc/gcc-4.8/README.Bugs --enable-languages=c,c++,java,go,d,fortran,objc,obj-c++ --prefix=/usr --program-suffix=-4.8 --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --with-gxx-include-dir=/usr/include/c++/4.8 --libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --enable-gnu-unique-object --disable-libmudflap --enable-plugin --with-system-zlib --disable-browser-plugin --enable-java-awt=gtk --enable-gtk-cairo --with-java-home=/usr/lib/jvm/java-1.5.0-gcj-4.8-amd64/jre --enable-java-home --with-jvm-root-dir=/usr/lib/jvm/java-1.5.0-gcj-4.8-amd64 --with-jvm-jar-dir=/usr/lib/jvm-exports/java-1.5.0-gcj-4.8-amd64 --with-arch-directory=amd64 --with-ecj-jar=/usr/share/java/eclipse-ecj.jar --enable-objc-gc --enable-multiarch --disable-werror --with-arch-32=i686 --with-abi=m64 --with-multilib-list=m32,m64,mx32 --with-tune=generic --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu
Thread model: posix
gcc version 4.8.4 (Ubuntu 4.8.4-2ubuntu1~14.04.3) 

I cleared build devel logs .catkin_* and then compile everything using 
`catkin build -DCMAKE_BUILD_TYPE=RelWithDebInfo`
The compilation was successful as before. However I got the same error as well.

mdhorn, 2017-03-28:
Does the standard ROS release package correctly on your system?
`sudo apt-get update ; sudo apt-get install ros-indigo-realsense-camera`

arpit15, 2017-03-28:
Yes, the standard ROS release package is working correctly. However I am facing the problem of this thread i.e. XYZRGB pointcloud stops streaming after sometime. Therefore I wanted to try building from source.

mdhorn, 2017-03-28:
@arpit15 what hardware are you using?
CPU, memory?

mdhorn, 2017-03-28:
Researching that error you listed for the undefined symbol looks more like a possible OpenCV issue.

Does the system have any OpenCV3 libraries installed?

What is the output of:
`dpkg -l '*opencv*'`

This might be helpful as well:
https://github.com/introlab/rtabmap_ros/issues/33#issuecomment-137886650

arpit15, 2017-03-28:
the output of `dpkg -l '*opencv*'`

Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name                                          Version                     Architecture                Description
+++-=============================================-===========================-===========================-===============================================================================================
ii  libopencv-calib3d-dev:amd64                   2.4.8+dfsg1-2ubuntu1        amd64                       development files for libopencv-calib3d
ii  libopencv-calib3d2.4:amd64                    2.4.8+dfsg1-2ubuntu1        amd64                       computer vision Camera Calibration library
ii  libopencv-contrib-dev:amd64                   2.4.8+dfsg1-2ubuntu1        amd64                       development files for libopencv-contrib
ii  libopencv-contrib2.4:amd64                    2.4.8+dfsg1-2ubuntu1        amd64                       computer vision contrib library
ii  libopencv-core-dev:amd64                      2.4.8+dfsg1-2ubuntu1        amd64                       development files for libopencv-core
ii  libopencv-core2.4:amd64                       2.4.8+dfsg1-2ubuntu1        amd64                       computer vision core library
ii  libopencv-dev                                 2.4.8+dfsg1-2ubuntu1        amd64                       development files for opencv
ii  libopencv-features2d-dev:amd64                2.4.8+dfsg1-2ubuntu1        amd64                       development files for libopencv-features2d
ii  libopencv-features2d2.4:amd64                 2.4.8+dfsg1-2ubuntu1        amd64                       computer vision Feature Detection and Descriptor Extraction library
ii  libopencv-flann-dev:amd64                     2.4.8+dfsg1-2ubuntu1        amd64                       development files for libopencv-flann
ii  libopencv-flann2.4:amd64                      2.4.8+dfsg1-2ubuntu1        amd64                       computer vision Clustering and Search in Multi-Dimensional spaces library
ii  libopencv-gpu-dev:amd64                       2.4.8+dfsg1-2ubuntu1        amd64                       development files for libopencv-gpu2.4
ii  libopencv-gpu2.4:amd64                        2.4.8+dfsg1-2ubuntu1        amd64                       computer vision GPU library
ii  libopencv-highgui-dev:amd64                   2.4.8+dfsg1-2ubuntu1        amd64                       development files for libopencv-highgui
ii  libopencv-highgui2.4:amd64                    2.4.8+dfsg1-2ubuntu1        amd64                       computer vision High-level GUI and Media I/O library
ii  libopencv-imgproc-dev:amd64                   2.4.8+dfsg1-2ubuntu1        amd64                       development files for libopencv-imgproc
ii  libopencv-imgproc2.4:amd64                    2.4.8+dfsg1-2ubuntu1        amd64                       computer vision Image Processing library
ii  libopencv-legacy-dev:amd64                    2.4.8+dfsg1-2ubuntu1        amd64                       development files for libopencv-legacy
ii  libopencv-legacy2.4:amd64                     2.4.8+dfsg1-2ubuntu1        amd64                       computer vision legacy library
ii  libopencv-ml-dev:amd64                        2.4.8+dfsg1-2ubuntu1        amd64                       development files for libopencv-ml
ii  libopencv-ml2.4:amd64                         2.4.8+dfsg1-2ubuntu1        amd64                       computer vision Machine Learning library
ii  libopencv-objdetect-dev:amd64                 2.4.8+dfsg1-2ubuntu1        amd64                       development files for libopencv-objdetect
ii  libopencv-objdetect2.4:amd64                  2.4.8+dfsg1-2ubuntu1        amd64                       computer vision Object Detection library
ii  libopencv-ocl-dev:amd64                       2.4.8+dfsg1-2ubuntu1        amd64                       development files for libopencv-ocl2.4
ii  libopencv-ocl2.4:amd64                        2.4.8+dfsg1-2ubuntu1        amd64                       computer vision OpenCL support library
ii  libopencv-photo-dev:amd64                     2.4.8+dfsg1-2ubuntu1        amd64                       development files for libopencv-photo2.4
ii  libopencv-photo2.4:amd64                      2.4.8+dfsg1-2ubuntu1        amd64                       computer vision computational photography library
ii  libopencv-stitching-dev:amd64                 2.4.8+dfsg1-2ubuntu1        amd64                       development files for libopencv-stitching2.4
ii  libopencv-stitching2.4:amd64                  2.4.8+dfsg1-2ubuntu1        amd64                       computer vision image stitching library
ii  libopencv-superres-dev:amd64                  2.4.8+dfsg1-2ubuntu1        amd64                       development files for libopencv-superres2.4
ii  libopencv-superres2.4:amd64                   2.4.8+dfsg1-2ubuntu1        amd64                       computer vision Super Resolution library
ii  libopencv-ts-dev:amd64                        2.4.8+dfsg1-2ubuntu1        amd64                       development files for libopencv-ts2.4
ii  libopencv-ts2.4:amd64                         2.4.8+dfsg1-2ubuntu1        amd64                       computer vision ts library
[... cut ...]
ii  ros-indigo-opencv-candidate                   0.2.4-0trusty-20170313-0936 amd64                       Experimental OpenCV functions
ii  ros-indigo-opencv3                            3.1.0-1trusty-20170313-0356 amd64                       OpenCV 3.0
ii  ros-indigo-vision-opencv                      1.11.15-0trusty-20170313-09 amd64                       Packages for interfacing ROS with OpenCV, a library of programming functions for real time comp

Is there a way to compile just realsense camera against opencv3 without uninstalling everything and compiling from source?

mdhorn, 2017-03-29:
I would guess that the opencv3 package you have installed is the reason the code you build from source doesn't work. Can you try removing that package and rebuilt, and test the code?
`sudo apt-get remove  ros-indigo-opencv3 `

arpit15, 2017-03-31:
@mdhorn  I removed opencv3 using the above command. However I am still facing the same error.

mdhorn, 2017-03-31:
Strange, if you remove the opencv3 and rebuilt I would not expect to still see that.
There is something else on your system.

can you provide the output of 
```
ldd /home/arpit/ros_wss/exp_ws/devel/lib//librealsense_camera_nodelet.so
objdump -p /home/arpit/ros_wss/exp_ws/devel/lib//librealsense_camera_nodelet.so
```

arpit15, 2017-04-01:
linux-vdso.so.1 =>  (0x00007ffc78fc1000)
	/usr/lib/libtcmalloc_minimal.so.4 (0x00007f4c0739e000)
	libcv_bridge.so => /opt/ros/indigo/lib/libcv_bridge.so (0x00007f4c0717b000)
	libopencv_core.so.2.4 => /usr/lib/x86_64-linux-gnu/libopencv_core.so.2.4 (0x00007f4c06d0b000)
	libimage_transport.so => /opt/ros/indigo/lib/libimage_transport.so (0x00007f4c06a97000)
	libdynamic_reconfigure_config_init_mutex.so => /opt/ros/indigo/lib/libdynamic_reconfigure_config_init_mutex.so (0x00007f4c06890000)
	libnodeletlib.so => /opt/ros/indigo/lib/libnodeletlib.so (0x00007f4c06636000)
	libclass_loader.so => /opt/ros/indigo/lib/libclass_loader.so (0x00007f4c06417000)
	libtf.so => /opt/ros/indigo/lib/libtf.so (0x00007f4c061e5000)
	libtf2_ros.so => /opt/ros/indigo/lib/libtf2_ros.so (0x00007f4c05f45000)
	libroscpp.so => /opt/ros/indigo/lib/libroscpp.so (0x00007f4c05be0000)
	libroscpp_serialization.so => /opt/ros/indigo/lib/libroscpp_serialization.so (0x00007f4c059dc000)
	librosconsole.so => /opt/ros/indigo/lib/librosconsole.so (0x00007f4c057b0000)
	librostime.so => /opt/ros/indigo/lib/librostime.so (0x00007f4c05586000)
	libboost_system.so.1.54.0 => /usr/lib/x86_64-linux-gnu/libboost_system.so.1.54.0 (0x00007f4c05381000)
	libboost_thread.so.1.54.0 => /usr/lib/x86_64-linux-gnu/libboost_thread.so.1.54.0 (0x00007f4c0516b000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f4c04f4d000)
	libconsole_bridge.so.0.2 => /usr/lib/x86_64-linux-gnu/libconsole_bridge.so.0.2 (0x00007f4c04d42000)
	librealsense.so.1 => /home/arpit/ros_wss/exp_ws/devel/.private/librealsense/lib/librealsense.so.1 (0x00007f4c04a54000)
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f4c04750000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f4c04449000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f4c04233000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f4c03e6b000)
	libopencv_imgproc.so.2.4 => /usr/lib/x86_64-linux-gnu/libopencv_imgproc.so.2.4 (0x00007f4c039da000)
	libopencv_highgui.so.2.4 => /usr/lib/x86_64-linux-gnu/libopencv_highgui.so.2.4 (0x00007f4c0378f000)
	libopencv_contrib.so.2.4 => /usr/lib/x86_64-linux-gnu/libopencv_contrib.so.2.4 (0x00007f4c034ac000)
	libboost_regex.so.1.54.0 => /usr/lib/x86_64-linux-gnu/libboost_regex.so.1.54.0 (0x00007f4c031a4000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007f4c02f8b000)
	librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007f4c02d83000)
	libGL.so.1 => /usr/lib/nvidia-367/libGL.so.1 (0x00007f4c02af3000)
	libtbb.so.2 => /usr/lib/libtbb.so.2 (0x00007f4c028bf000)
	libmessage_filters.so => /opt/ros/indigo/lib/libmessage_filters.so (0x00007f4c026ba000)
	libroslib.so => /opt/ros/indigo/lib/libroslib.so (0x00007f4c024a7000)
	libboost_filesystem.so.1.54.0 => /usr/lib/x86_64-linux-gnu/libboost_filesystem.so.1.54.0 (0x00007f4c02291000)
	libtinyxml.so.2.6.2 => /usr/lib/x86_64-linux-gnu/libtinyxml.so.2.6.2 (0x00007f4c0207c000)
	libbondcpp.so => /opt/ros/indigo/lib/libbondcpp.so (0x00007f4c01e4b000)
	libPocoFoundation.so.9 => /usr/lib/libPocoFoundation.so.9 (0x00007f4c01aff000)
	libtf2.so => /opt/ros/indigo/lib/libtf2.so (0x00007f4c018ce000)
	libactionlib.so => /opt/ros/indigo/lib/libactionlib.so (0x00007f4c016b1000)
	libxmlrpcpp.so => /opt/ros/indigo/lib/libxmlrpcpp.so (0x00007f4c01492000)
	libcpp_common.so => /opt/ros/indigo/lib/libcpp_common.so (0x00007f4c0128a000)
	librosconsole_log4cxx.so => /opt/ros/indigo/lib/librosconsole_log4cxx.so (0x00007f4c01076000)
	librosconsole_backend_interface.so => /opt/ros/indigo/lib/librosconsole_backend_interface.so (0x00007f4c00e73000)
	liblog4cxx.so.10 => /usr/lib/liblog4cxx.so.10 (0x00007f4c00a8b000)
	/lib64/ld-linux-x86-64.so.2 (0x0000560f2340d000)
	libusb-1.0.so.0 => /lib/x86_64-linux-gnu/libusb-1.0.so.0 (0x00007f4c00873000)
	libjpeg.so.8 => /usr/lib/x86_64-linux-gnu/libjpeg.so.8 (0x00007f4c0061e000)
	libpng12.so.0 => /lib/x86_64-linux-gnu/libpng12.so.0 (0x00007f4c003f7000)
	libtiff.so.5 => /usr/lib/x86_64-linux-gnu/libtiff.so.5 (0x00007f4c00185000)
	libjasper.so.1 => /usr/lib/x86_64-linux-gnu/libjasper.so.1 (0x00007f4bfff2e000)
	libIlmImf.so.6 => /usr/lib/x86_64-linux-gnu/libIlmImf.so.6 (0x00007f4bffc7e000)
	libHalf.so.6 => /usr/lib/x86_64-linux-gnu/libHalf.so.6 (0x00007f4bffa3b000)
	libgtk-x11-2.0.so.0 => /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0 (0x00007f4bff3fe000)
	libgdk-x11-2.0.so.0 => /usr/lib/x86_64-linux-gnu/libgdk-x11-2.0.so.0 (0x00007f4bff14a000)
	libgobject-2.0.so.0 => /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0 (0x00007f4bfeef9000)
	libglib-2.0.so.0 => /lib/x86_64-linux-gnu/libglib-2.0.so.0 (0x00007f4bfebf1000)
	libgtkglext-x11-1.0.so.0 => /usr/lib/libgtkglext-x11-1.0.so.0 (0x00007f4bfe9ec000)
	libgdkglext-x11-1.0.so.0 => /usr/lib/libgdkglext-x11-1.0.so.0 (0x00007f4bfe788000)
	libdc1394.so.22 => /usr/lib/x86_64-linux-gnu/libdc1394.so.22 (0x00007f4bfe514000)
	libv4l1.so.0 => /usr/lib/x86_64-linux-gnu/libv4l1.so.0 (0x00007f4bfe30d000)
	libavcodec.so.54 => /usr/lib/x86_64-linux-gnu/libavcodec.so.54 (0x00007f4bfd5b6000)
	libavformat.so.54 => /usr/lib/x86_64-linux-gnu/libavformat.so.54 (0x00007f4bfd294000)
	libavutil.so.52 => /usr/lib/x86_64-linux-gnu/libavutil.so.52 (0x00007f4bfd06e000)
	libswscale.so.2 => /usr/lib/x86_64-linux-gnu/libswscale.so.2 (0x00007f4bfce27000)
	libopencv_features2d.so.2.4 => /usr/lib/x86_64-linux-gnu/libopencv_features2d.so.2.4 (0x00007f4bfcb83000)
	libopencv_calib3d.so.2.4 => /usr/lib/x86_64-linux-gnu/libopencv_calib3d.so.2.4 (0x00007f4bfc8ed000)
[... cut ...]
	libXau.so.6 => /usr/lib/x86_64-linux-gnu/libXau.so.6 (0x00007f4bee09d000)
	libXdmcp.so.6 => /usr/lib/x86_64-linux-gnu/libXdmcp.so.6 (0x00007f4bede96000)
	libgraphite2.so.3 => /usr/lib/x86_64-linux-gnu/libgraphite2.so.3 (0x00007f4bedc70000)
	libdatrie.so.1 => /usr/lib/x86_64-linux-gnu/libdatrie.so.1 (0x00007f4beda68000)
	libSM.so.6 => /usr/lib/x86_64-linux-gnu/libSM.so.6 (0x00007f4bed860000)
	libICE.so.6 => /usr/lib/x86_64-linux-gnu/libICE.so.6 (0x00007f4bed644000)
	libgpg-error.so.0 => /lib/x86_64-linux-gnu/libgpg-error.so.0 (0x00007f4bed43e000)

mdhorn, 2017-04-03:
@arpit15 I'm at a loss of what causing this problem for you.

Can you remove that local copy of librealsense?
librealsense.so.1 => /home/arpit/ros_wss/exp_ws/devel/.private/librealsense/lib/librealsense.so.1 (0x00007f4c04a54000)

The only places that missing symbol seem to reside are in opencv3 libraries.
If you've removed opencv3 packages (or libraries from /usr/local), rebooted, and tried build in a new, clean workspace I'm not sure how this still a problem.

The last things I can suggest is install the 'lddtree' tool and use it to find each shared library and it's dependencies. Then look through all of these libraries to find which one has the offending symbol.

```
sudo apt-get install pax-utils
lddtree /home/arpit/ros_wss/exp_ws/devel/lib//librealsense_camera_nodelet.so
```

You could try something like this to collect the list of libraries and search them:
```
lddtree devel/lib/librealsense_camera_nodelet.so   | awk -F'=> ' '{print $NF}' | grep '/' | tee all_libs.txt
for l in `cat all_libs.txt`; do grep _ZN2cv6String10deallocateEv $l ; done
```

arpit15, 2017-04-04:
@mdhorn 
Checking all the following libraries seems to be a lot of work. I will try to install from source on a new computer with all the new libraries to see if it builds on that. Is there a way I could eliminate some of the libraries because the realsense camera works with precompiled libraries. 
The whole motive of me trying to compile from source is to get the stream continously. Is there a way to solve the issue in this thread without compiling from source?

`lddtree devel/lib/librealsense_camera_nodelet.so   | awk -F'=> ' '{print $NF}' | grep '/' | tee all_libs.txt
for l in `cat all_libs.txt`; do grep _ZN2cv6String10deallocateEv $l ; done`

outputs the following thing

scanelf: rpath_security_checks(): Security problem NULL DT_RPATH in devel/lib/librealsense_camera_nodelet.so
scanelf: rpath_security_checks(): Security problem NULL DT_RPATH in devel/lib/librealsense_camera_nodelet.so
scanelf: rpath_security_checks(): Security problem NULL DT_RPATH in devel/lib/librealsense_camera_nodelet.so
scanelf: rpath_security_checks(): Security problem NULL DT_RPATH in devel/lib/librealsense_camera_nodelet.so
scanelf: rpath_security_checks(): Security problem NULL DT_RPATH in devel/lib/librealsense_camera_nodelet.so
scanelf: rpath_security_checks(): Security problem NULL DT_RPATH in devel/lib/librealsense_camera_nodelet.so
scanelf: rpath_security_checks(): Security problem NULL DT_RPATH in devel/lib/librealsense_camera_nodelet.so
scanelf: rpath_security_checks(): Security problem NULL DT_RPATH in devel/lib/librealsense_camera_nodelet.so
scanelf: rpath_security_checks(): Security problem NULL DT_RPATH in devel/lib/librealsense_camera_nodelet.so
scanelf: rpath_security_checks(): Security problem NULL DT_RPATH in devel/lib/librealsense_camera_nodelet.so
scanelf: rpath_security_checks(): Security problem NULL DT_RPATH in devel/lib/librealsense_camera_nodelet.so
scanelf: rpath_security_checks(): Security problem NULL DT_RPATH in devel/lib/librealsense_camera_nodelet.so
scanelf: rpath_security_checks(): Security problem NULL DT_RPATH in devel/lib/librealsense_camera_nodelet.so
scanelf: rpath_security_checks(): Security problem NULL DT_RPATH in devel/lib/librealsense_camera_nodelet.so
scanelf: rpath_security_checks(): Security problem NULL DT_RPATH in devel/lib/librealsense_camera_nodelet.so
scanelf: rpath_security_checks(): Security problem NULL DT_RPATH in devel/lib/librealsense_camera_nodelet.so
scanelf: rpath_security_checks(): Security problem NULL DT_RPATH in devel/lib/librealsense_camera_nodelet.so
scanelf: rpath_security_checks(): Security problem NULL DT_RPATH in devel/lib/librealsense_camera_nodelet.so
scanelf: rpath_security_checks(): Security problem NULL DT_RPATH in devel/lib/librealsense_camera_nodelet.so
scanelf: rpath_security_checks(): Security problem NULL DT_RPATH in devel/lib/librealsense_camera_nodelet.so
scanelf: rpath_security_checks(): Security problem NULL DT_RPATH in devel/lib/librealsense_camera_nodelet.so
/opt/ros/indigo/lib/libcv_bridge.so
/usr/lib/x86_64-linux-gnu/libopencv_imgproc.so.2.4
/usr/lib/libtbb.so.2
/usr/lib/x86_64-linux-gnu/libopencv_highgui.so.2.4
/usr/lib/nvidia-367/libGL.so.1
/lib/i386-linux-gnu/libdl.so.2
/lib/i386-linux-gnu/ld-linux.so.2
/usr/lib/nvidia-367/libGLX.so.0
/usr/lib/i386-linux-gnu/libX11.so.6
/usr/lib/i386-linux-gnu/libxcb.so.1
/usr/lib/i386-linux-gnu/libXau.so.6
/usr/lib/i386-linux-gnu/libXdmcp.so.6
/usr/lib/i386-linux-gnu/libXext.so.6
/usr/lib/nvidia-367/libGLdispatch.so.0
/usr/lib/x86_64-linux-gnu/libjpeg.so.8
/lib/i386-linux-gnu/libpng12.so.0
/lib/i386-linux-gnu/libz.so.1
/usr/lib/x86_64-linux-gnu/libtiff.so.5
/lib/x86_64-linux-gnu/liblzma.so.5
/usr/lib/x86_64-linux-gnu/libjbig.so.0
/usr/lib/x86_64-linux-gnu/libjasper.so.1
/usr/lib/x86_64-linux-gnu/libIlmImf.so.6
/usr/lib/x86_64-linux-gnu/libIex.so.6
/usr/lib/x86_64-linux-gnu/libIlmThread.so.6
/usr/lib/x86_64-linux-gnu/libHalf.so.6
/usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0
/usr/lib/x86_64-linux-gnu/libgmodule-2.0.so.0
/usr/lib/x86_64-linux-gnu/libpangocairo-1.0.so.0
/usr/lib/i386-linux-gnu/libfreetype.so.6
/usr/lib/i386-linux-gnu/libXfixes.so.3
/usr/lib/x86_64-linux-gnu/libatk-1.0.so.0
/usr/lib/x86_64-linux-gnu/libcairo.so.2
/usr/lib/x86_64-linux-gnu/libpixman-1.so.0
/lib/x86_64-linux-gnu/ld-linux-x86-64.so.2
/usr/lib/x86_64-linux-gnu/libxcb-shm.so.0
/usr/lib/x86_64-linux-gnu/libxcb-render.so.0
/usr/lib/i386-linux-gnu/libXrender.so.1
/lib/i386-linux-gnu/librt.so.1
/usr/lib/x86_64-linux-gnu/libgdk_pixbuf-2.0.so.0
/usr/lib/x86_64-linux-gnu/libgio-2.0.so.0
[... 82 lines cut ...]
/usr/lib/x86_64-linux-gnu/libboost_filesystem.so.1.54.0
/usr/lib/x86_64-linux-gnu/libtinyxml.so.2.6.2
/opt/ros/indigo/lib/libdynamic_reconfigure_config_init_mutex.so
/opt/ros/indigo/lib/libnodeletlib.so
/opt/ros/indigo/lib/libclass_loader.so
/usr/lib/libPocoFoundation.so.9
/opt/ros/indigo/lib/libtf.so
/opt/ros/indigo/lib/libtf2_ros.so
/opt/ros/indigo/lib/libroscpp.so
/opt/ros/indigo/lib/libroscpp_serialization.so
/opt/ros/indigo/lib/librosconsole.so
/usr/lib/liblog4cxx.so.10
/usr/lib/x86_64-linux-gnu/libaprutil-1.so.0
/lib/i386-linux-gnu/libcrypt.so.1
/usr/lib/x86_64-linux-gnu/libapr-1.so.0
/opt/ros/indigo/lib/librostime.so
/usr/lib/x86_64-linux-gnu/libboost_system.so.1.54.0
/usr/lib/x86_64-linux-gnu/libboost_thread.so.1.54.0
/lib/i386-linux-gnu/libpthread.so.0
/usr/lib/x86_64-linux-gnu/libconsole_bridge.so.0.2
/home/arpit/ros_wss/exp_ws/devel/.private/librealsense/lib/librealsense.so.1
/usr/lib/x86_64-linux-gnu/libstdc++.so.6
/lib/i386-linux-gnu/libm.so.6
/lib/i386-linux-gnu/libgcc_s.so.1
/lib/i386-linux-gnu/libc.so.6

mdhorn, 2017-04-04:
@arpit15 Okay, let's drop the discussion of building from source on your system.

As for fixing the streaming problem for RGBD, please try the latest release 1.8.0.
It has been released to the ROS Build Farm, but just missed the last package sync yesterday so it will be about 2 weeks before it is in the public ROS repository. 

1.8.0 is available in [Shadow Repository](http://wiki.ros.org/ShadowRepository), but I'm attaching the package to this Issue so you can just download and install it without switching to the Shadow Repository.
[ros-indigo-realsense-camera-180.zip](https://github.com/intel-ros/realsense/files/894172/ros-indigo-realsense-camera-180.zip)
First, unzip the file:
`unzip ros-indigo-realsense-camera-180.zip`
Then install the newer release of realsense-camera: 
`sudo dpkg -i ros-indigo-realsense-camera_1.8.0-0trusty-20170403-142612-0700_amd64.deb`

arpit15, 2017-04-06:
@mdhorn Thank you. Everything seems to be fine with the new code version.

mdhorn, 2017-04-06:
@arpit15 very good to hear!

closed 2017-04-06 by mdhorn

arpit15, 2017-04-07:
@mdhorn The stream is freezing again. I am obtaining the following error.
`[ERROR] [1491598159.515880658]: /camera/driver - Error calling rs_wait_for_frames ( device:0x31f8018 ): 
Timeout waiting for frames.`

AndyZe, 2017-05-08:
I was having the same problem of an R200 crashing after about 20 minutes when the queue size was 10k. It crashed sooner with a smaller queue size. But the exact_sync option that Mark mentioned seems to have solved it. I am using a recent debian (installed ~4/30/2017).

arpit15, 2017-06-01:
@AndyZe I can't find the exact_sync option in file sr300_nodelet_rgbd.launch. I am using realsense_camera version 1.8.0 on ROS indigo. Kindly let me know where should I put this parameter in.

Thanks 
Arpit

arpit15, 2017-06-01:
@AndyZe @mdhorn It seems that the ram requirement for the nodelet keeps on increasing until my computer runs out of ram. This leads to closing of the roslaunch file. I was able to run the node longer(110k frames received in rviz) by closing everything else. 
Kindly suggest a way around this. I don't understand why does the RAM requirements keeps on increasing for the nodelet. 

Thanks
Arpit