realsenseai/realsense-ros #200: Kinetic debian package does not publish (but advertises) IR images

skohlbr, 2017-01-26:
### System Configuration
Please complete Your Configuration detail below. Refer to the BKC at [Configuration](../realsense_camera/README.md#configuration).

| Version               | Your Configuration   |
|:--------------------- |:---------------------| 
| Operating System      | "Ubuntu 16.04.1 LTS"          |
| Kernel                | 4.4.0-59-generic              |
| ROS                   |  kinetic      |
| ROS RealSense         | 1.6.1               |
| librealsense          | 1.11.1                |
| Camera Type-Firmware  | Name: Intel RealSense R200, Camera FW: 1.0.72.06            |

### Expected Behavior

After starting
<pre>
roslaunch realsense_camera r200_nodelet_default.launch
</pre>
receive images when subscribing to the  `/camera/ir/image_raw` or `/camera/ir2/image_raw` topics.

### Actual Behavior

No data is received when subscribing to the topics.

### Steps to Reproduce

Install `ros-kinetic-realsense-camera` from .debs:
<pre>
sudo apt-get install ros-kinetic-realsense-camera
</pre>
Run standard launch file:
<pre>
roslaunch realsense_camera r200_nodelet_default.launch
</pre>
Try subscribing to the above mentioned topics. Same holds true for the associated camera_info topics. RGB and depth streams work.

skohlbr, 2017-01-26:
Ok, so starting
<pre>
roslaunch realsense_camera r200_nodelet_ rgbd.launch
</pre>
provides ir images. It still seems weird that the default setup advertises them but silently refuses to publish them.

mdhorn, 2017-01-26:
We had a request from the hardware team to disable the IR streams by default starting in release 1.6.1. For RGB example launch file they are enable, but by default IR streams are disabled.

There is a new 'enable_ir' parameter as of 1.6.1 release, but I see that the ROS Wiki page was missing this -- just added it.

Let me know if this resolves your question.

skohlbr, 2017-01-27:
Ok, that is good to know. I think it would be much less confusing if the Camera Publishers would also be disabled when the IR streams are disabled, perhaps by creating publishers only in [advertiseTopics() ](https://github.com/intel-ros/realsense/blob/031d407c354c7d926b431aefeaad10470ea7cf87/realsense_camera/src/base_nodelet.cpp#L360) depending on the corresponding (non dynamically reconfigurable) `enable_[STREAM]` params for the different streams. I'm not sure if this would have side effects though.

mdhorn, 2017-01-27:
I recall the initial version of the code did not advertise disable streams, but we ran into some issues or compatibility problems and change the design to always advertise even if the topic wasn't published due to the stream being disabled.

skohlbr, 2017-01-27:
Ok, this is cleared up then, thanks for the quick replies :)

closed 2017-01-27 by skohlbr