realsenseai/realsense-ros #201: ROS published registered point cloud has significantly higher registration error than ccp-example cloud

skohlbr, 2017-01-27:
### System Configuration

| Version               | Your Configuration   |
|:--------------------- |:---------------------| 
| Operating System      | "Ubuntu 16.04.1 LTS"          |
| Kernel                | 4.4.0-59-generic              |
| ROS                   |  kinetic      |
| ROS RealSense         | 1.6.1               |
| librealsense          | 1.11.1                |
| Camera Type-Firmware  | Name: Intel RealSense R200, Camera FW: 1.0.72.06            |

---

### Expected Behavior

The ROS published point cloud on the `/camera/depth_registered/points` topic should have comparable registration error to the one that is shown in the `cpp-pointcloud` example

### Actual Behavior

The cloud published on the `/camera/depth_registered/points` has visibly higher registration error than the one provided by the example code. See provided example images below. The ones with blue background are from the cpp-pointcloud example, the ones with grey background from rviz. Fingers are colored correctly for the example, while they are mostly white when holding them horizontal for the ROS cloud. This indicates a registration offset along the vertical axis between depth and rgb image.

[image]
[image]

[image]
[image]

### Steps to Reproduce

#### ROS cloud:
Install realsense camera from .debs
<pre>
sudo apt-get install ros-kinetic-realsense-camera
</pre>
Run launch file
<pre>
roslaunch realsense_camera r200_nodelet_rgbd.launch 
</pre>
Look at point cloud in rviz

#### cpp-poincloud example cloud

Clone librealsense repo and build with example as per https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md

Navigate to `build/devel/lib/librealsense` and start `./cpp-pointcloud`

skohlbr, 2017-01-27:
Looks like #184 might be the culprit here.

mdhorn, 2017-02-22:
We are looking into #184

skohlbr, 2017-08-07:
@mattcurfman @mdhorn  is this on Intel's radar still? It's been 6 months and multiple people reporting calibration issues with the R200 and ROS drivers even before that (this issue, #233, #184, #30).

skohlbr, 2017-08-12:
So I did some more digging. The back-projected point cloud in the RGB image using the rviz Camera Display (both image_color and image_rect_color) has the color of points exactly match up. This actually is bad though, as the registration error between color and depth  means that the depth image (and thus point cloud) is off by the ~9 pixels reported previously.

You can clearly see the side of the box I pointed the camera at being color red (see cloud at bottom of the screenshot), which it should not be.
[image]

For comparison, the same scene running `cpp-tutorial-3-pointcloud ` which essentially looks perfect:
[image]

skohlbr, 2017-08-12:
So I reactivated the native points stream and registration of color vs. points looks good (as expected). Reprojection into the RGB image shows an offset though, so what is broken is very likely the calibration values for the RGB camera:

[image]

From https://github.com/IntelRealSense/librealsense/blob/v1.12.1/doc/projection.md#r200 we know that the **rectified** color stream has no rotation wrt to the depth frames. But in the driver the plain **non-rectified** color stream is used so far, while all rotation components used in calibration are just set to identity (see https://github.com/intel-ros/realsense/blob/indigo-devel/realsense_camera/src/base_nodelet.cpp#L1061, https://github.com/intel-ros/realsense/blob/indigo-devel/realsense_camera/src/base_nodelet.cpp#L735).

Working theory is that two approaches should be feasible:

- Use the rectified color stream. In that case, the identity rotation is correct.

- Continue using the non-rectified stream. In that case, the frame rotations need to be set correctly. Also, for this to work, rectification of the RGB stream has to be done in "ROS world". I'm not sure if the modified calibration model used for the R200 RGB camera is compatible with the cv camera model used by the nodelets.

skohlbr, 2017-08-12:
So I managed to vastly improve things by using the rectified stream calibration instead of the standard unrectified stream one. Here's a screenshot using this approach:

[image]

Created a PR for this change (#261). This is still somewhat inconsistent as the rectified camera stream is not used consistently, but the results are vastly better than the previously completely unacceptable 9 pixel offset.

icarpis, 2018-01-31:
Fixed by #261

closed 2018-01-31 by icarpis