realsenseai/realsense-ros #203: R200 camera nodelet manager process has died

y27choi, 2017-02-08:
### System Configuration
Please complete Your Configuration detail below. Refer to the BKC at [Configuration](../realsense_camera/README.md#configuration).

| Version               | Your Configuration   |
|:--------------------- |:---------------------|
| Operating System      | Ubuntu 14.04.4 LTS |
| Kernel                | 4.4.0-47-generic|
| ROS                   | indigo                    |
| ROS RealSense         | 1.5.0               |
| librealsense          | 1.11.0               |
| Camera Type-Firmware  | R200-1.0.72.06            |

### Expected Behavior
/camera/driver can successfully update the dynamic settings of the camera.

### Actual Behavior
It doesn't happen every time, but I've seen it happen often enough to notice. When I disable depth_camera using dynamic_reconfigure service call ("set_parameter"), and enable it back right after, it throws out the error message (see below) and the nodelet manager crashes.
```
[ INFO] [1486525217.725395391]: /camera/driver - Stopping camera
[ INFO] [1486525218.253660335]: /camera/driver - Enabling Depth in manual mode
[ INFO] [1486525218.253774217]: /camera/driver - Enabling IR in manual mode
[ INFO] [1486525218.253866257]: /camera/driver - Enabling IR2 in manual mode
[ INFO] [1486525218.253936991]: /camera/driver - Starting camera
[ INFO] [1486525218.326540724]: /camera/driver - Setting dynamic camera options
[ERROR] [1486525223.303488718]: /camera/driver - Error calling rs_wait_for_frames ( device:0x133c088 ): 
Timeout waiting for frames. 
terminate called after throwing an instance of 'boost::exception_detail::clone_impl<boost::exception_detail::error_info_injector<boost::lock_error> >'
  what():  boost: mutex lock failed in pthread_mutex_lock: Invalid argument
[camera_nodelet_manager-1] process has died [pid 4689, exit code -6, cmd /opt/ros/indigo/lib/nodelet/nodelet manager __name:=camera_nodelet_manager __log:=/tmp/90b3746a-ed84-11e6-968d-b8aeedea3ef7/camera_nodelet_manager-1.log].
```

### Steps to Reproduce
1) Launch a single nodelet manager for two cameras.
2) Enable/disable the depth camera by calling a ROS service ("set_parameter")
3) After repeating step 2) for a while, the nodelet manager will crash.

mdhorn, 2017-02-08:
Please update to the latest releases and retest.
`sudo apt-get update`
`sudo apt-get upgrade`
librealsense should be 1.12.0 for Indigo
realsense_camera should be 1.7.0  for Indigo

mdhorn, 2017-03-23:
Closing due to no response.

closed 2017-03-23 by mdhorn

mga0, 2017-07-03:
I have exactly the same issue with two SR300 in a nodelet.

### System Configuration

| Version               | Your Configuration   |
|:--------------------- |:---------------------|
| Operating System      | Ubuntu 16.04            |
| Kernel                | 4.4.0-83-generic               |
| ROS                   | kinetic                    |
| ROS RealSense         |     1.8.0            |
| librealsense          | 1.12.1               |
| Camera Type-Firmware  | SR300 3.15.0.0            |

---

### Expected Behavior

Switching off and on the IR projector of two Realsense SR300 using dynamic reconfiguration

### Actual Behavior

Nodelet crashes after 3-5 successful switch off/on procedures.

> [ INFO] [1499072586.160897330]: /camera/driver - Setting dynamic camera options
> [ INFO] [1499072586.216745451]: /camera/driver - Stopping camera
> [ INFO] [1499072586.226933855]: /camera/driver - Enabling Depth in manual mode
> [ INFO] [1499072586.227052865]: /camera/driver - Starting camera
> [ INFO] [1499072588.137289066]: /camera/driver - Setting dynamic camera options
> [ERROR] [1499072591.233984291]: /camera/driver - Error calling rs_wait_for_frames ( device:0x1ea3820 ): 
> Timeout waiting for frames. 

### Steps to Reproduce

1. Launch two SR300s with the following file: roslaunch 

<launch>

  <arg name="camera_1"              default="camera_L" />
  <arg name="camera_1_camera_type"  default="SR300"    /> 
  <arg name="camera_1_serial_no"    default="617205002089"   />
  <arg name="camera_1_usb_port_id"  default=""        /> 

  <arg name="camera_2"              default="camera_C" />
  <arg name="camera_2_camera_type"  default="SR300"    /> 
  <arg name="camera_2_serial_no"    default="620202003712"   />
  <arg name="camera_2_usb_port_id"  default=""        /> 

  <arg name="camera_3"              default="camera_R" />
  <arg name="camera_3_camera_type"  default="SR300"    /> 
  <arg name="camera_3_serial_no"    default="553201000655"  />
  <arg name="camera_3_usb_port_id"  default=""        /> 

  <arg name="camera_4"              default="camera_B" />
  <arg name="camera_4_camera_type"  default="SR300"    /> 
  <arg name="camera_4_serial_no"    default="617203001912"  /> 
  
  <arg name="camera_4_usb_port_id"  default=""        /> 

  <arg name="enable_depth"      default="false" />
  <arg name="enable_color"      default="true" />
  <arg name="color_fps"         default="10" />
  <arg name="color_width"       default="1280" />
  <arg name="color_height"      default="720" />
  <arg name="depth_fps"         default="10" />
  <arg name="depth_width"       default="640" />
  <arg name="depth_height"      default="480" />
  <arg name="mode"              default="manual" />
  <arg name="enable_tf"         default="false" />

  <param name="$(arg camera_1)/driver/enable_depth"      type="bool" value="$(arg enable_depth)" />
[... 41 lines cut ...]

  <arg name="manager" value="camera_nodelet_manager" /> 
  <node pkg="nodelet" type="nodelet" name="$(arg manager)" args="manager" output="screen"/>

  

 <group ns="$(arg camera_1)">
    <include file="$(find realsense_camera)/launch/includes/nodelet.launch.xml">
      <arg name="manager"      value="/$(arg manager)" />
      <arg name="camera"       value="$(arg camera_1)" />
      <arg name="camera_type"  value="$(arg camera_1_camera_type)" />
      <arg name="serial_no"    value="$(arg camera_1_serial_no)" />
      <arg name="usb_port_id"  value="$(arg camera_1_usb_port_id)" />
    </include>
  </group>

 <group ns="$(arg camera_2)">
    <include file="$(find realsense_camera)/launch/includes/nodelet.launch.xml">
      <arg name="manager"      value="/$(arg manager)" />
      <arg name="camera"       value="$(arg camera_2)" />
      <arg name="camera_type"  value="$(arg camera_2_camera_type)" />
      <arg name="serial_no"    value="$(arg camera_2_serial_no)" />
      <arg name="usb_port_id"  value="$(arg camera_2_usb_port_id)" />
    </include>
  </group>

delmottea, 2017-07-04:
I got the same problem (also with two sr300).
This problem happen only with ros, I have no problem switching on and off the laser when using directly librealsense outside ros.
The only solution we found until now was to use a small servo-motor to occlude mechanically the laser projector