realsenseai/realsense-ros #205: Add a script for restarting the USB device

AndyZe, 2017-02-22:
We are using our RealSense on a mobile robot platform, which means long cables and lots of motion. The cameras often crash, so we found it convenient to write a C program that finds and restarts all R200 cameras without power cycling them. (It might be more convenient if it were a Python script, but I was in a hurry to get this done and found a decent C template online.) Let me know if this is of interest and where to put it and I'll do a PR. Otherwise, hopefully it inspires something similar.

[reset_r200.c.txt](https://github.com/intel-ros/realsense/files/795138/reset_r200.c.txt)

AndyZe, 2017-02-22:
To test it, compile:
$ cc reset_r200.c -o reset_r200

Make it executable:
$ chmod +x reset_r200

AndyZe, 2017-03-03:
Adding a line to kill all zombie R200 processes, based on the process name "nodelet."
[reset_r200.c.txt](https://github.com/intel-ros/realsense/files/815434/reset_r200.c.txt)

mattcurfman, 2017-04-24:
Hi @AndyZe, it is not normal for R200 cameras to crash, but acknowledge that USB3 devices are more sensitive to long cables, etc.  I'd recommend including your reset utility as a .cpp file in the 'src' directory, and adding the required commands to the CMakeLists.txt file to build it as a standalone reset utility.

closed 2018-04-06 by jlgarcia75