realsenseai/realsense-ros #207: Possible regression in version 1.7 -- drifiting timestamps for R200 depth images

mirzashah, 2017-03-02:
Recently we started having an issue where over time, the timestamps of the depth images generated by the realsense driver drift into the past from what they're supposed to be. This causes our point clouds to use the wrong transform and we end up seeing point cloud trails -- which is very obvious as we rotate our robots in place.

We were able to confirm this by overwriting the timestamp of our depth images with current system time (```ros::Time::now()```) before feeding them to the processing nodelets. This appears to project the clouds into the correct frame.

We think this commit might be the culprit:
dbb3983ae0f5ac43c55106d3271b7abea1bdf3aa

Reverting to 1.6.1 appears to fix the issue for us.

mkhansen-intel, 2017-03-13:
@mirzashah - would you like to try my PR#209 and see if that fixes this issue for you?

mirzashah, 2017-03-24:
@mkhansen-intel will attempt to when I get the chance, thanks for the fix

mdhorn, 2017-04-03:
Resolved in release 1.8.0
Available now by pulling tag 1.8.0 and building from sources.
Released to ROS Build Farm.  Available in [Shadow Repository](http://wiki.ros.org/ShadowRepository) once [PR 14443 for Indigo](https://github.com/ros/rosdistro/pull/14443) and [PR 14444 for Kinetic](https://github.com/ros/rosdistro/pull/14444l) are merged.
Public ROS repository packages available after next sync (1-3 weeks) by [OSRF](https://www.osrfoundation.org/osrf-projects/).

closed 2017-04-03 by mdhorn