realsenseai/realsense-ros #208: SR300 lost camera calibration

MarqRazz, 2017-03-02:
### System Configuration
| Version               | Your Configuration   |
|:--------------------- |:---------------------|
| Operating System      | Ubuntu 14.04.5 LTS              |
| Kernel                | 4.4.0-64-generic                |
| ROS                   | indigo                    |
| ROS RealSense         | 1.7.1                |
| librealsense          | 1.12.1                |
| Camera Type-Firmware  | SR300 - 3.15.0.0            |

### Expected Behavior
After launching camera I expect it to return a non-zero distortion matrix
`rostopic echo /camera/color/camera_info`
below are the values I get back from my r200 camera (Camera FW: 1.0.72.06)
```
header: 
  seq: 47
  stamp: 
    secs: 1488496395
    nsecs: 884896012
  frame_id: camera_rgb_optical_frame
height: 1080
width: 1920
distortion_model: plumb_bob
D: [-0.09274807572364807, 0.07124239206314087, -0.00021547084907069802, 0.002587695373222232, 0.0]
K: [1408.539306640625, 0.0, 980.18115234375, 0.0, 1411.0552978515625, 499.4236145019531, 0.0, 0.0, 1.0]
R: [1.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 1.0]
P: [1408.539306640625, 0.0, 980.18115234375, 0.0, 0.0, 1411.0552978515625, 499.4236145019531, 0.0, 0.0, 0.0, 1.0, 0.0]
binning_x: 0
binning_y: 0
roi: 
  x_offset: 0
  y_offset: 0
  height: 0
  width: 0
  do_rectify: False
---
```

### Actual Behavior
 distortion matrix is zero
```
header: 
  seq: 88
  stamp: 
    secs: 1488496084
    nsecs: 770626796
  frame_id: camera_rgb_optical_frame
height: 480
width: 640
distortion_model: plumb_bob
D: [0.0, 0.0, 0.0, 0.0, 0.0]
K: [617.9534912109375, 0.0, 306.2196350097656, 0.0, 617.95361328125, 248.8450469970703, 0.0, 0.0, 1.0]
R: [1.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 1.0]
P: [617.9534912109375, 0.0, 306.2196350097656, 0.0, 0.0, 617.95361328125, 248.8450469970703, 0.0, 0.0, 0.0, 1.0, 0.0]
binning_x: 0
binning_y: 0
roi: 
  x_offset: 0
  y_offset: 0
  height: 0
  width: 0
  do_rectify: False
---
```

### Steps to Reproduce
`roslaunch realsense_camera sr300_nodelet_default.launch`
`rostopic echo /camera/color/camera_info`

FYI this is a new camera, up to this point I have been doing my tests on a r200. Is is possible to rewrite the camera_info to the camera? From what I have read this is not supported yet.

I do get a warning about the camera firmware. Could this cause the problem?
```
[ WARN] [1488496075.059926094]: /camera/driver - Detected unvalidated firmware:
- 612201003586's current camera firmware is 3.15.0.0, Validated camera firmware is 3.10.10.0
```

Thanks,
-Marq

mdhorn, 2017-03-03:
I believe this is because the R200 has distortion for the color stream, but the SR300 and F200 do not.  See https://github.com/IntelRealSense/librealsense/blob/master/doc/projection.md

MarqRazz, 2017-03-06:
Thanks for the link mdhorn!

closed 2017-03-06 by MarqRazz