realsenseai/realsense-ros #210: SR300 has fixed time stamp

MarqRazz, 2017-03-03:
### System Configuration
| Version               | Your Configuration   |
|:--------------------- |:---------------------|
| Operating System      | Ubuntu 14.04.5 LTS              |
| Kernel                | 4.4.0-64-generic                |
| ROS                   | indigo                    |
| ROS RealSense         | 1.7.1                |
| librealsense          | 1.12.1                |
| Camera Type-Firmware  | SR300 - 3.15.0.0            |

### Expected Behavior
After launching camera I expect published images to have a time stamp that updates as the node runs

### Actual Behavior
 all images received have the same time stamp from when the node was started. 

### Steps to Reproduce
`roslaunch realsense_camera sr300_nodelet_default.launch`
For a quick check I have been looking at the camera_info topic because it uses the same time stamp
`rostopic echo /camera/color/camera_info`

Removing the static declaration from the variable `last_common_stamp` fixed the problem for me
```
  /*
   * Determine the timestamp for the publish topic. -- overrides base class
   */
  ros::Time SR300Nodelet::getTimestamp(rs_stream stream_index, double frame_ts)
  {
    static ros::Time last_common_stamp = ros::Time::now();

    if (stream_index == fastest_stream_)
    {
      last_common_stamp = ros::Time::now();
    }

    return last_common_stamp;
  }
```
If this is a reasonable fix I can issue a pull request.

Thanks,
-Marq

mdhorn, 2017-03-03:
The problem was introduced when we switch from polling to callbacks. To help align the color and depth using callback we were only updating the timestamp with via the fastest stream. What was not accounted for is if the faster stream (by default depth) doesn't have subscribers, the code which updates the timestamp is never called.

This will be fixed in the next major release as we are rolling back to polling for SR300/F200/R200 camera as the callbacks aren't fully functional of the older cameras.

One possible work around here is to have subscribers on both the color and depth stream.

The other recommendation would be pulling the code from the 1.6.1 release tag and building locally.

MarqRazz, 2017-03-06:
Thanks for pointing me in the right direction mdhorn! Any ideas on when the next release will be out?

mdhorn, 2017-03-06:
It will likely take 4+ weeks before the fix is pushed and tested; then the random 2-8 weeks before the next ROS package sync.

For now, I'd recommend building 1.6.1 from sources, and watch the repo for a patch.

mkhansen-intel, 2017-03-13:
@MarqRazz - I have a PR #209  that I think fixes this issue, would you like to clone it and try it out?

MarqRazz, 2017-03-16:
Sorry it took me a few days to get to this.

@mkhansen-intel - I cloned #209 and still have the same issue.

MarqRazz, 2017-03-16:
@mkhansen-intel 

I ran some more tests today and can confirm that PR #209 does fix this issue.

Thanks for your help!

mdhorn, 2017-04-03:
Resolved in release 1.8.0
Available now by pulling tag 1.8.0 and building from sources.
Released to ROS Build Farm.  Available in [Shadow Repository](http://wiki.ros.org/ShadowRepository) once [PR 14443 for Indigo](https://github.com/ros/rosdistro/pull/14443) and [PR 14444 for Kinetic](https://github.com/ros/rosdistro/pull/14444l) are merged.
Public ROS repository packages available after next sync (1-3 weeks) by [OSRF](https://www.osrfoundation.org/osrf-projects/).

closed 2017-04-03 by mdhorn