realsenseai/realsense-ros #213: SR300 - Exposure control

jamessergeant, 2017-03-07:
### System Configuration

```
| Version               | Your Configuration   |
|:--------------------- |:---------------------|
| Operating System      | Ubuntu 14.04.5 LTS  |
| Kernel                |4.4.0-64-generic                |
| ROS                   | indigo                   |
| ROS RealSense         | 1.7.1                |
| librealsense          | 1.12.1              |
| Camera Type-Firmware  | 3.17.0.0           |
```

### Expected Behavior
Ability to disable auto exposure and control exposure setting. The librealsense example cpp-enumerated-devices lists COLOR_EXPOSURE and COLOR_ENABLE_AUTO_EXPOSURE as supported options. See below (streams ommitted).

### Actual Behavior
Dynamic reconfigure's get_configuration does not return exposure settings as an option.

```
Device 0 - Intel RealSense SR300:
 Serial number: 614205004058
 Firmware version: 3.17.0.0
 USB Port ID: 2-2-1
 Camera info: 
    DEVICE_NAME         : 	Intel RealSense SR300
    DEVICE_SERIAL_NUMBER: 	614205004058
    CAMERA_FIRMWARE_VERSION: 	3.17.0.0
 Supported options:                                    min        max       step  default  
    COLOR_BACKLIGHT_COMPENSATION                       : 0    ... 4           1     0         
    COLOR_BRIGHTNESS                                   : -64  ... 64          1     0         
    COLOR_CONTRAST                                     : 0    ... 100         1     50        
    COLOR_EXPOSURE                                     : 39   ... 10000       1     156       
    COLOR_GAIN                                         : 0    ... 128         1     64        
    COLOR_GAMMA                                        : 100  ... 500         1     300       
    COLOR_HUE                                          : -180 ... 180         1     0         
    COLOR_SATURATION                                   : 0    ... 100         1     64        
    COLOR_SHARPNESS                                    : 0    ... 100         1     50        
    COLOR_WHITE_BALANCE                                : 2800 ... 6500        10    4600      
    COLOR_ENABLE_AUTO_EXPOSURE                         : 0    ... 1           1     1         
    COLOR_ENABLE_AUTO_WHITE_BALANCE                    : 0    ... 1           1     1         
    F200_LASER_POWER                                   : 0    ... 16          1     16        
    F200_ACCURACY                                      : 1    ... 3           1     1         
    F200_MOTION_RANGE                                  : 0    ... 220         1     9         
    F200_FILTER_OPTION                                 : 0    ... 7           1     5         
    F200_CONFIDENCE_THRESHOLD                          : 0    ... 15          1     3         
    SR300_AUTO_RANGE_ENABLE_MOTION_VERSUS_RANGE        : 0    ... 2           1     -1        
    SR300_AUTO_RANGE_ENABLE_LASER                      : 0    ... 1           1     -1        
    SR300_AUTO_RANGE_MIN_MOTION_VERSUS_RANGE           : -32768... 32767       1     -1        
    SR300_AUTO_RANGE_MAX_MOTION_VERSUS_RANGE           : -32768... 32767       1     -1        
    SR300_AUTO_RANGE_START_MOTION_VERSUS_RANGE         : -32768... 32767       1     -1        
    SR300_AUTO_RANGE_MIN_LASER                         : -32768... 32767       1     -1        
    SR300_AUTO_RANGE_MAX_LASER                         : -32768... 32767       1     -1        
    SR300_AUTO_RANGE_START_LASER                       : -32768... 32767       1     -1        
    SR300_AUTO_RANGE_UPPER_THRESHOLD                   : 0    ... 65535       1     -1        
    SR300_AUTO_RANGE_LOWER_THRESHOLD                   : 0    ... 65535       1     -1        
    FRAMES_QUEUE_SIZE                                  : 1    ... 20          1     20        
    HARDWARE_LOGGER_ENABLED                            : 0    ... 1           1     0 
```

mdhorn, 2017-03-21:
Fixed by #214 
Will be in release candidate 1.8.0

mdhorn, 2017-03-21:
Reopening until release is made

mdhorn, 2017-04-03:
Resolved in release 1.8.0
Available now by pulling tag 1.8.0 and building from sources.
Released to ROS Build Farm.  Available in [Shadow Repository](http://wiki.ros.org/ShadowRepository) once [PR 14443 for Indigo](https://github.com/ros/rosdistro/pull/14443) and [PR 14444 for Kinetic](https://github.com/ros/rosdistro/pull/14444l) are merged.
Public ROS repository packages available after next sync (1-3 weeks) by [OSRF](https://www.osrfoundation.org/osrf-projects/).

closed 2017-04-03 by mdhorn