realsenseai/realsense-ros #217: cob_people detection for realsense camera driver

zubair123khan, 2017-03-10:
### System Configuration
Please complete Your Configuration detail below. Refer to the BKC at [Configuration](../realsense_camera/README.md#configuration).

| Version               | Your Configuration   |
|:--------------------- |:---------------------|
| Operating System      | ubuntu                |
| Kernel                | 16.04               |
| ROS                   | kinect                   |
| ROS RealSense         | realsense_camera               |
| librealsense          | ?.?.?                |
| Camera Type-Firmware  | R200           |

hi all , 

i am now trying to use cob_people detection instead of realsense_person, and it seems there is no error in launching the file, i just need to link cob_people_detection with my camera driver, for it to transport images to the cob_people_detection. 
i am not able to load images from my realsense camera driver to cob_people_detection. 
any one please help on what changes i need to do in the people detection launch file . !!~~

~thanks

zubair123khan, 2017-03-13:
guys how do i get color image topic and point cloud topic for realsense camera R200 ??? where this info will be ??

please help 

thanks

mdhorn, 2017-03-13:
The ROS module is documented on the ROS wiki:
http://wiki.ros.org/realsense_camera

zubair123khan, 2017-03-13:
got it working,, but still i am not able to work with realsense_people ,, hope they get back with the update soon ,
thanks

mattcurfman, 2017-03-13:
I will post an update when there is new information to share with realsense_ros_person node

zubair123khan, 2017-03-13:
@mattcurfman  when at the soonest will it be possible ??
it will be a great update ...

closed 2017-03-13 by zubair123khan

zubair123khan, 2017-03-22:
@mattcurfman  any update on realsense_person??

mattcurfman, 2017-04-05:
@zubair123khan check out https://github.com/IntelRealSense/realsense_samples_ros released today

zubair123khan, 2017-04-06:
thanks really a good one ..