realsenseai/realsense-ros #218: rgb point cloud in r200 nodelt.default

zubair123khan, 2017-03-14:
### System Configuration
Please complete Your Configuration detail below. Refer to the BKC at [Configuration](../realsense_camera/README.md#configuration).

| Version               | Your Configuration   |
|:--------------------- |:---------------------|
| Operating System      | ubuntu            |
| Kernel                | ?-?-?                |
| ROS                   | kinect                 |
| ROS RealSense         | realsense_camera              |
| librealsense          | ?.?.?                |
| Camera Type-Firmware  | R200       |

hi, i want an rgb point cloud output which is actually the output for launch file r200_nodelet_rgbd.launch , but i want to run launch file r200_nodelet_ddefault.launch . and i want this rgb point cloud to be the output of nodelet_default.launch . 

can anyone help me on how to merge these two launch files to get the point cloud rgb from nodelet_default.launch 

thanks

mdhorn, 2017-03-14:
I might be helpful for you to first read through the roslaunch XML wiki:
http://wiki.ros.org/roslaunch/XML 

Then the realsense_camera Tutorials:
http://wiki.ros.org/realsense_camera/Tutorials

so that you understand how the launch files work that you may update them yourself.

zubair123khan, 2017-03-15:
why am i not getting the colored point cloud while using rgbd.launch ,, i am just getting depth point cloud 

any suggestions ?

mdhorn, 2017-03-15:
The default `r200_nodelet_rgbd.launch` launch file should publish a color point cloud.

`roslaunch realsense_camera r200_nodelet_rgbd.launch
`

We will need to you to provide some more details about your environment, what command and launch file you are using, and attach the launch file if you have made any modifications.
Otherwise we would just be making guesses.

| Version               | Method |
|:--------------------- |:------------ |
| Operating System      | `grep DISTRIB_DESCRIPTION /etc/*elease*` or `lsb_release -d` |
| Kernel                | `uname -r` |
| ROS                   | `rosversion -d` |
| ROS RealSense         | `rosversion realsense_camera` |
| librealsense          | `rosversion librealsense` |
| Camera Type-Firmware  | View the ROS log from running nodelet |

---

### Expected Behavior

### Actual Behavior

### Steps to Reproduce

zubair123khan, 2017-03-16:
@mattcurfman  please ping me once the realsense_person is out with update,, 
~thanks

zubair123khan, 2017-03-16:
@mdhorn i was using realsense_camera r200_nodelet_default.launch but that doesnt contain the color point cloud so then i used realsense_camera r200_nodelet_rgbd.launch which have /camera/depth_registered/points as rgb point cloud if i am not wrong,,

zubair123khan, 2017-03-16:
@mdhorn 
[ WARN] [1489683236.227971950]: /camera/driver - Detected unvalidated firmware:
				- 2391012929's current camera firmware is 1.0.72.10, Validated camera firmware is 1.0.72.06
this is the warning i am getting,, how to validate my firmware ?? any suggestion

mdhorn, 2017-03-16:
You should ONLY need to run:
`roslaunch realsense_camera r200_nodelet_rgbd.launch`

You need to provide the information I requested or I can't really provide you any direction.  We need to know the configuration and the hardware would be good too.

Please do not say "it doesn't work", that is not anything I can assist you with. 
Provide the fail message, or the fail signature.  If there is a problem with the image, then attach examples of the images.

mdhorn, 2017-03-16:
@zubair123khan That is a warning, nothing that will prevent functionality.
It is saying that the version of FW on your camera is new than the version of FW used to validate the realsense_camera package you are using.

zubair123khan, 2017-03-16:
sure I will include from next time @mdhorn

mdhorn, 2017-03-16:
@zubair123khan Next time?  So this problem is resolved?  Please close if so.

zubair123khan, 2017-03-16:
Yes I am getting my cloud properly..
But I don't know what is happening in between when I am running it again
it's not showing the cloud
And same thing with no corrections .. it's showing the point cloud now.

zubair123khan, 2017-03-17:
[image]

@mdhorn  in this image the output of camera nodelet manager should be atleast a color image and a color point cloud, i am using r200_nodelet_rgbd.launch ...
and hence th output of sensor message gateway should be color image and point cloud,,

what can be the flaw from the camera driver or nodelet ..???

please help

zubair123khan, 2017-03-20:
@mattcurfman  any update on realsense_person ???

mdhorn, 2017-03-23:
@zubair123khan I thought we resolved the color point cloud issue last week.

I'm not sure I follow your last comment to me.
What exactly is the problem you are having?

closed 2017-04-13 by zubair123khan