realsenseai/realsense-ros #222: ERROR TF re-parenting contention

severin-lemaignan, 2017-03-27:
I'm running 2 SR300 cameras at the same time ([launch file here](https://github.com/severin-lemaignan/freeplay-sandbox-ros/blob/master/launch/dual_sr300.launch) ). Everything looks to work fine, but `roswtf` return the following error (curisouly enough, only for one camera):

```
ERROR TF re-parenting contention:
 * reparenting of [camera_ir_optical_frame] to [camera_2_ir_frame] by [/camera_nodelet_manager]
```

Maybe something to investigate?

### System Configuration

| ROS RealSense         | 1.7.2                |

mdhorn, 2017-03-28:
@severin-lemaignan Using this stripped down launch file [sr300_nodelet_multiple_cameras.launch](https://github.com/intel-ros/realsense/files/874170/sr300_nodelet_multiple_cameras.launch.txt)  (based on r200_nodelet_multiple_cameras.launch) I'm not seeing any errors reported:

```
$ roswtf
Loaded plugin tf.tfwtf
No package or stack in context
================================================================================
Static checks summary:

No errors or warnings
================================================================================
Beginning tests of your ROS graph. These may take awhile...
analyzing graph...
... done analyzing graph
running graph rules...
... done running graph rules
running tf checks, this will take a second...
... tf checks complete

Online checks summary:

No errors or warnings
```

| Version               | Method |
|:--------------------- |:------------ |
| Operating System      | Ubuntu 16.04.2 LTS |
| Kernel                | 4.4.0-59-generic |
| ROS                   | kinetic |
| ROS RealSense         | 1.7.2 |
| librealsense          | 1.12.1 |
| Camera Type-Firmware  | 3.10.10.0 |

Also tested with realsense_camera version 1.7.1 and the release candidate 1.8.0 built from sources.

severin-lemaignan, 2017-03-29:
Hum... I've not been able to reproduce the issue reliably... Sometimes it occurs, sometimes it does not. I'll put that on an invalid TF cache for now, and close the issue until I can reproduce it reliably.

closed 2017-03-29 by severin-lemaignan