realsenseai/realsense-ros #224: SR300 depth image and point cloud data is flickering in rviz

vacky11, 2017-03-29:
### System Configuration
Please complete Your Configuration detail below. Refer to the BKC at [Configuration](../realsense_camera/README.md#configuration).

| Version               | Your Configuration   |
|:--------------------- |:---------------------|
| Operating System      | ubuntu 14.04 |
| Kernel                        | 4.4.0-040400-generic |
| ROS                           | indigo |
| ROS RealSense         | 1.7.1 |
| librealsense                | 1.12.0 |
| Camera Type-Firmware  | SR300 |

### Expected Behavior
Get depth image and point cloud data in rviz

### Actual Behavior
Depth image and point cloud data is flickering

### Steps to Reproduce
```
roslaunch realsense_camera sr300_nodelet_rgbd.launch
rviz
```
Add /camera/depth_registered/points as PointCloud2
Add /camera/depth/image as Depth Image

mdhorn, 2017-03-29:
Please update to the one of the released ubuntu kernels by using the steps documented on the ROS wiki:
http://wiki.ros.org/librealsense#Installation_Prerequisites
The "4.4.0-040400-generic" is not supported in the ROS environment -- that was something the librealsense team had documented a while back.

vacky11, 2017-03-30:
Thanks for replying.
I followed the steps mentioned in the link you provided. But the ubuntu kernel version didn't change. Can you please tell me how can I update to one of the released ubuntu kernels?

mdhorn, 2017-03-30:
@vacky11 After the new kernel install, did you reboot to switch to the new kernel?
What kernel are you running now?

You basically need to remove that XXXX kernel image that you downloaded and installed.
Use `dpkg -l | grep 4.4.0-040400` to find the packages for the old kernel, then remove those packages with `sudo apt-get purge <packages>`.
Here is an example from askubuntu:
http://askubuntu.com/questions/654094/how-to-roll-back-or-downgrade-kernel-on-14-04-re-wine-stopped-working

mdhorn, 2017-03-31:
@vacky11 I see today that the step to actually update to the new kernel was not on the wiki.
Please complete that step:
http://wiki.ros.org/librealsense#Update_to_the_Xenial_kernel
Update to the Xenial kernel
```
sudo apt-get install linux-image-generic-lts-xenial
sudo reboot

```

vacky11, 2017-04-24:
Updating the kernel has solved the problem.
@mdhorn  Thanks a lot.

closed 2017-04-24 by vacky11

nlyubova, 2017-05-23:
Hello, 
I had exactly the same issue for both Ubuntu 14.04 and 16.04. 
@vacky11 which kernel is working for you on Ubuntu 14? After upgrading 14.04 to 4.4.0-78 kernel, I cannot start my Ubuntu anymore...

@mdhorn do you know which kernel is needed for Ubuntu 16? I have 4.4.15 and depth image is flickering like in this issue

vacky11, 2017-05-24:
@nlyubova 
For ubuntu 14.04 with kernel 4.4.0-78, it worked for me.

AndyZe, 2017-09-21:
A hacky workaround for this type of thing is to turn up the "Decay Time" in RViz.