realsenseai/realsense-ros #225: Realsense driver fails to build on OS X

helenol, 2017-04-03:
### System Configuration
| Version               | Your Configuration   |
|:--------------------- |:---------------------|
| Operating System      | OS X 10.11.2                |
| Kernel                | ?-?-?                |
| ROS                   | indigo                    |
| ROS RealSense         | 1.8.0                |
| librealsense          |    1.12.1             |
| Camera Type-Firmware  | zr300 (but doesn't matter)            |

---

### Expected Behavior
Can build driver under OS X.

### Actual Behavior
Compilation fails with an error:
```
/Users/helen/catkin_ws/src/realsense/realsense_camera/src/base_nodelet.cpp:1277:30: error: use of undeclared identifier 'environ'
      execvpe(argv[0], argv, environ);
```

Upon fixing the environ error (by adding:
```C++
#define _GNU_SOURCE
#include <unistd.h>
extern char** environ;
```
)

I get a new error:
```
/Users/helen/catkin_ws/src/realsense/realsense_camera/src/base_nodelet.cpp:1275:7: error: use of undeclared identifier 'execvpe'; did you mean 'execve'?
      execvpe(argv[0], argv, environ);
      ^~~~~~~
      execve
/usr/include/unistd.h:444:6: note: 'execve' declared here
int      execve(const char *, char * const *, char * const *) __WATCHOS_PROHIBITED __TVOS_PROHIBITED;
         ^
```

It appears that execvpe() isn't defined on OS X -- http://stackoverflow.com/questions/28313542/executing-a-command-with-execvpe-in-c#comment44979436_28313542 

It seems like that part of the code is only for switching the PID of the process -- is this necessary in all cases? Or can this part just be skipped on OS X?

Any help is appreciated.
Thanks,
Helen

mattcurfman, 2017-04-24:
Are you trying to build the ROS camera node itself under Mac OS?  We've only used Ubuntu 14.04 and 16.04 with ROS to date.  Would love to learn more about ROS on Mac.

helenol, 2017-04-25:
Yes, installed using Mike Purvis's instructions: https://github.com/mikepurvis/ros-install-osx
After just commenting out the offending line, the node mostly works under OS X except for the IMU on the ZR300, which gets 2-3 callbacks in and then stops (while everything works fine with the same driver on 14.04). Camera and depth data come in without issue though.

mattcurfman, 2017-04-25:
Yes, I've seen the same behavior with the ZR300 camera on Mac, for IMU data stream.  I'd love to see if anybody could debug that that submit a PR to http://github.com/IntelRealSense/librealsense.  I haven't had time myself, and it is not an area the librealsense team is looking at as a priority.

I'd like to keep the master branch clean for ROS on Ubuntu/Debian, but would be happy to host a experimental branch for Mac support and accept any PR's you might choose to submit for support.

closed 2019-03-26 by RealSenseCustomerSupport