realsenseai/realsense-ros #233: Vertical misalignment of registered(+rectified) depth image compared to rectified RGB image

mindThomas, 2017-04-12:
### System Configuration
| Version               | Your Configuration   |
|:--------------------- |:---------------------|
| Operating System      | Yocto               |
| Kernel                | 4.4.3-yocto-standard                |
| ROS                   | indigo                    |
| ROS RealSense         | 1.7.2                |
| librealsense          | 1.12.1   (system lib)                |
| Camera Type-Firmware  | R200-1.0.71.06            |

### Expected Behavior
I would expect /camera/rgb/image_rect_color and /camera/depth_registered/sw_registered/image_rect_raw to be aligned.

### Actual Behavior
Please download the pictures below and try to match them (with transparancy).
[image]
[image]

Doing a transparent matching of the above test images results in the following horizontally mismatched picture.
[image]
Notice how the registered depth image has a 9 pixel tall black area in the top and a 20 pixel black bar in the right side. If the registered depth image is just moved 9 pixels upwards the color and depth image seem to match much better.
[image]

### Steps to Reproduce
Launch the rgbd nodelet.
`roslaunch realsense_camera r200_nodelet_rgbd.launch`
Capture images with rqt or similar

### Further Questions
Can someone elaborate on the difference between all the topics being published by the RGBD nodelet?
```
/camera/depth/camera_info
/camera/depth/disparity
/camera/depth/image
/camera/depth/image_raw
/camera/depth/image_rect
/camera/depth/image_rect_raw
/camera/depth/points
/camera/depth_registered/disparity
/camera/depth_registered/points
/camera/depth_registered/sw_registered/camera_info
/camera/depth_registered/sw_registered/image_rect
/camera/depth_registered/sw_registered/image_rect_raw
/camera/ir/camera_info
/camera/ir/image_raw
/camera/ir/image_rect_ir
/camera/ir2/camera_info
/camera/ir2/image_raw
/camera/rgb/camera_info
/camera/rgb/image_color
/camera/rgb/image_mono
/camera/rgb/image_raw
/camera/rgb/image_rect_color
/camera/rgb/image_rect_mono
```
What is the difference between raw and rectified in the depth image (there doesn't seem to be any difference, which could be explained from the fact that the depth image does not need distortion correction)?
What is the difference between raw and rectified RGB image? The only difference there seem to be is that the rectified RGB image is non-distorted. Is this correct? No conversion using intrinsics or extrinsics, eg. moving into a different frame?
What is the registered depth image? Is that simply each pixel of the depth image transformed into the optical RGB frame (image plane) and hence matching with the rectified RGB image?

mbelluzzo, 2017-04-22:
@mindThomas have you took the information on https://github.com/IntelRealSense/librealsense/blob/v1.12.1/doc/projection.md#r200 into consideration?

Specially the parts about [extrinsic parameters](https://github.com/IntelRealSense/librealsense/blob/v1.12.1/doc/projection.md#extrinsic-camera-parameters) and
[R200](https://github.com/IntelRealSense/librealsense/blob/v1.12.1/doc/projection.md#r200). Maybe if you apply the info there you can fix the alignment.

mindThomas, 2017-04-22:
@mbelluzzo I have indeed considered this.
I even decided to implement the registration myself using the available intrinsic (/.../camera_info topic) and extrinsics (/tf_static) with great success. Doing the registration manually (looping through all pixels, which is processing intensive) seem to work and gives me a depth image aligned to the RGB image.
But why does the software registered depth image, that comes out of the RGBD nodelet (/camera/depth_registered/sw_registered/image_rect_raw), not seem to be properly aligned?

I thought the point in registration was to align one image into another - including both intrinsics and extrinsics - such that I could avoid doing any other post-processing.
What I need is simply two aligned images from where I can take one pixel in the RGB image to get the color and from the same pixel location take a pixel in the depth image to get the depth.
This does not seem to be possible with the current RGBD nodelet implementation?

skohlbr, 2017-05-05:
This is appears to be the same bug as for instance reported in #201 (and further related issues linked there). Somewhere in the ROS pipeline intrinsic calibration data is not used correctly, leading to the observed offsets. We've been using this fork with a hack that fixed things for us: https://github.com/zhang-xin/realsense/commits/hack (originally linked in #30 I believe)

Given that people report these issues since more than a year I certainly hope that the maintainers have a look. Who knows how many people abandoned the R200 already because the registration looked crappy for them ;)

arpit15, 2017-06-29:
@mindThomas I am using sr300. I have tried to use the intrinsics and extrinsics from the camera to reproject depth image to rgb image frame. However I still see misalignment in the images. 

[image]

Could you kindly share your script for doing the projection so that I can recheck my code and get the desired result?
I initially thought that the factory settings are not good enough. However I haven't had any success after re-calibration as well. 
[image]

It would be great if the authors could mark this issue as a bug. This way people won't waste time to calibrate the sensor again.

mindThomas, 2017-06-30:
@arpit15 We did not have to recalibrate the camera. Our algorithm simply use the factory calibrated parameters but does the rectification and alignment manually.
This process happens in the following lines of our ROS node: https://github.com/davidromanos/intel_aero_rtf_gr871/blob/master/src/depthgrab.cpp#L251-L270

arpit15, 2017-06-30:
@mindThomas Thanks for posting the link to the code. From [sr300 projection information](https://github.com/IntelRealSense/librealsense/blob/v1.12.1/doc/projection.md#f200-and-sr300) it seems that sr300 doesn't require any distortion correction. Did you check if the problem was in rectification or intrinsics not used properly in projecting from depth to rgb frame? I think you can use /camera/rgb/image_rect_color and /camera/depth/image_rect to get the rectified images and then do your transform to get the images in the same frame. 
I tried to do the above with no avail. I get alignment as posted above for sr300. Can you post the results which you getting with your code for r200?

Thanks
Arpit

mindThomas, 2017-07-01:
We did not check nor investigate why the RealSense node did not supply us the correct images in /camera/rgb/image_rect_color and rectified depth. Based on tests we quickly concluded that these two images were not aligned so we had to do something ourselves. As the frames only have a translation difference along the horizontal axis the rectified images should at least align vertically, which we experienced they didn't.

I agree with your observations that the SR200 is not working and modelled the same way as the R200 so you can not use our code right away. Please notice that the R200 camera uses the Modified Brown-Conrady distortion model on the color images whereof the SR200 uses the Inverse Brown-Conrady distortion model on the depth images.

arpit15, 2017-07-01:
@mindThomas Thanks for pointing out the distortion model used in SR300. I tried to create a file just like R200 using rs.h and pyrealsense. The code could be found [here](https://gist.github.com/arpit15/2c2628b4124a2d4e411f831bb0d53043). However I don't get a good output. 
[image]
[image]
[image]

Was this something which you obtained in your initial trials? I am trying to debug the cause. Any help is appreciated.
Thanks

icarpis, 2018-01-31:
Fixed by #261

closed 2018-01-31 by icarpis