realsenseai/realsense-ros #234: depth image has a wierd shape

arpit15, 2017-04-12:
### System Configuration
Please complete Your Configuration detail below. Refer to the BKC at [Configuration](../realsense_camera/README.md#configuration).

| Version               | Your Configuration   |
|:--------------------- |:---------------------|
| Operating System      | ubuntu 14.04                |
| Kernel                | 4.4.0-66-generic                |
| ROS                   | Indigo                  |
| ROS RealSense         | 1.8.0               |
| librealsense          | 1.12.1               |
| Camera Type-Firmware  | SR300 3.15.0.0            |

---
### Expected Behavior
Image when nothing is on the table
[image]
Corresponding rgb image
[image]

I expect the depth image to give values upto certain range at all times. For ex: in the above case depth image I would assume that the pixels where the depth image is black are outside of its range. However that doesn't seem to be the case.
[image]
Corresponding rgb image
[image]
As can be seen in the this case if I place an object the sensor starts giving values.
Another case
[image]

[image]

I don't understand this behaviour. This happens for any object for ex: my hand and toy
[image]

### Actual Behavior
The depth image gives values when there is an object present and not otherwise. Like in the above cases I would expect the depth image to have values in the circle in the first image and zero everywhere else no matter what happens there as it is out of range.

### Steps to Reproduce
roslaunch realsense_camera sr300_nodelet_rgbd.launch

Thank you
Arpit

mindThomas, 2017-04-14:
Can you please take a look at the IR streams (/camera/ir/image_raw and /camera/ir2/image_raw) as this behaviour might be a result from saturated or non-present IR texture in the upper corners.
Keep in mind that for a textureless surface like your white table, the generated IR pattern is a necessity to perform matching and calculate a disparity map from the two stereo images.

arpit15, 2017-04-14:
@mindThomas 
[this](https://drive.google.com/open?id=0B62cBQ948cfVaFN0Vy1oUUNQZjQ) is the link to just the ir stream.  [This](https://drive.google.com/open?id=0B62cBQ948cfVNlo1RTAzbmRkdzA) is the stream with the corresponding depth stream. I don't understand how the ir stream can get patterns on the same table part when the toy is present. Kindly help me to understand the issue and find a way around it.
Thank you

mindThomas, 2017-04-16:
We can't see these pictures.
Can you please upload them here in the thread?

arpit15, 2017-06-01:
@mindThomas  I have changed the permissions of the video. It will be accessible by link now.

closed 2019-03-26 by RealSenseCustomerSupport