realsenseai/realsense-ros #236: depth image is not proper and have weired pattern or noise

zubair123khan, 2017-04-18:
### System Configuration
Please complete Your Configuration detail below. Refer to the BKC at [Configuration](../realsense_camera/README.md#configuration).

| Version               | Your Configuration   |
|:--------------------- |:---------------------|
| Operating System      | ubuntu              |
| Kernel                | 4.4.0-72-generic             |
| ROS                   | kinect                   |
| ROS RealSense         | realsense_camera             |
| Camera Type-Firmware  | r200           |

---

### Expected Behavior
there should now be any holes in the depth image . 

### Actual Behavior
the depth image is weired and seems not to be working as it should be ,, please help diagnose this problem as i am not able to use depth value properly because of this issue 
please take a look at the image below 

[image]

zubair123khan, 2017-04-20:
@mdhorn

mattcurfman, 2017-04-24:
the 'cpp-capture' utility shows R200 previews running at 640x480, which means that the R200 minimal depth it can see is 72cm (reference: https://github.com/IntelRealSense/librealsense/blob/master/doc/camera_specs.md).  What does it look like with your hand 72cm or farther?

At 320x240, the minimal depth is 32cm

zubair123khan, 2017-04-27:
its better now , it was the same result with 72 cm or far but i did some setting change and luckily i got it to original depth clarity.. 
thanks  @mattcurfman

closed 2017-04-27 by zubair123khan