realsenseai/realsense-ros #237: Unable to connect SR300 - LIBUSB_ERROR_ACCESS

vacky11, 2017-04-24:
### System Configuration
Please complete Your Configuration detail below. Refer to the BKC at [Configuration](../realsense_camera/README.md#configuration).

| Version               | Your Configuration   |
|:--------------------- |:---------------------|
| Operating System      | 14.04  |
| Kernel                | 4.4.0-72-generic   |
| ROS                   |Indigo   |
| ROS RealSense         |1.8.0  |
| librealsense          | 1.12.1     |
| Camera Type-Firmware  | SR300        |

Getting following error after - roslaunch realsense_camera sr300_nodelet_rgbd.launch

### Expected Behavior
Camera should connect 

### Actual Behavior
[ERROR] [1493042283.953908649]: /camera/driver - No cameras detected!
[ERROR] [1493042283.954012152]: /camera/driver - Error calling rs_create_context ( api_version:11201 ): 
libusb_open(...) returned LIBUSB_ERROR_ACCESS

mattcurfman, 2017-04-24:
Can you confirm that ros-kinetic-librealsense package installed correctly?  ros-kinetic-librealsense uses DPMS to ensure your kernel is properly patched, and the correct UDEV rules are installed under /etc/udev/rules.d.  

The documentation at http://wiki.ros.org/librealsense#Installation_Prerequisites will provide you more information about the installation of ros-kinetic-librealsense package

vacky11, 2017-04-24:
@mattcurfman  Thanks for replying. 
I have followed all the installation instructions given by librealsense package. Is there a way to ensure that my librealsense package is installed properly?

mattcurfman, 2017-04-24:
I would recommended running the installation verification script @ https://github.com/IntelRealSense/librealsense/blob/master/scripts/gt_setup_verification.sh

vacky11, 2017-04-25:
I get the following error when I run the script. I am not able to understand why it is trying to find ZR300 camera even though I have connected SR300 camera.

Error: No ZR300 cameras were found,
Make sure that the device is firmly connected to USB3 port and rerun the script...

mattcurf, 2017-04-27:
@vacky11 agreed, should have nothing to do with ZR300 camera since you have SR300.  I will check further

mmseines, 2017-05-18:
@vacky11  I have a simmilar issue as well. LIBUSB_ACESS_ERROR, followed by all the nodelets failing to launch.

Opening the gt_verification script i changed the part detecting the ZR300 camera (see change below)
```
#Verify a single ZR300 is properly connected
DS=$(lsusb | grep -i "8086:0aa5" | wc -l)  #changed from 0ab0 i think...
```
where 8086:0aa5 is the device code i get when i use lsusb | grep 8086 with my sr300 connected. 

in my case the script returns with:
`
ZR300 Camera was found - Ok

Verify formats required by ZR300 Camera: 

Y8I	-	Ok
Y12I	-	Ok
Y16	-	Ok
Z16	-	Ok
GREY	-	Ok
YUYV	-	Ok
RW16 	-	Fail

Unrecognized formats: 

The formats patch has not been applied correctly! 

Done...
`

I do not know if this is enough to identify the problem since all i did was hack the detection of a ZR300 camera...

mmseines, 2017-05-19:
@vacky11 
I figured it out. 
The udev permission gives you permission if your user is part of the "plugdev" group I was not a part of this group by default.

`sudo adduser "username" plugdev`

should fix it if we have the same problem

closed 2017-08-21 by dorodnic