realsenseai/realsense-ros #240: ZR300 Unable to get point cloud data or IR data

zubinp, 2017-05-18:
### System Configuration

| Version               | Your Configuration   |
|:--------------------- |:---------------------|
| Operating System      | Ubuntu 16.04.2 LTS                 |
| Kernel                | 4.4.0-78-generic               |
| ROS                   | Kinetic                    |
| ROS RealSense         | 1.8.0               |
| librealsense          | 1.12.1                |
| Camera Type-Firmware  |Intel Realsense ZR300 2.0.71.28   |

---

### Expected Behavior

On launching zr300_nodelet_rgdb.launch file, registered pointclouds should have been published

### Actual Behavior

I do not get any messages on /camera/depth/points, /camera/depth_registered/points, /camera/ir/image_raw , /camera/ir2/image_raw topics. I only get messages on the rgb image topic at 30 HZ. 

My dmesg log looks like this after attaching the camera:
```
[  368.461606] usb 3-2: new high-speed USB device number 2 using xhci_hcd
[  368.592534] usb 3-2: New USB device found, idVendor=0451, idProduct=8027
[  368.592541] usb 3-2: New USB device strings: Mfr=0, Product=0, SerialNumber=1
[  368.592545] usb 3-2: SerialNumber: BE0C008160B7
[  368.593346] hub 3-2:1.0: USB hub found
[  368.593761] hub 3-2:1.0: 2 ports detected
[  368.701828] usb 4-2: new SuperSpeed USB device number 2 using xhci_hcd
[  368.720215] usb 4-2: New USB device found, idVendor=0451, idProduct=8025
[  368.720222] usb 4-2: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[  368.721114] hub 4-2:1.0: USB hub found
[  368.721613] hub 4-2:1.0: 2 ports detected
[  368.993992] usb 4-2.2: new SuperSpeed USB device number 3 using xhci_hcd
[  369.011481] usb 4-2.2: LPM exit latency is zeroed, disabling LPM.
[  369.013417] usb 4-2.2: New USB device found, idVendor=8086, idProduct=0ad0
[  369.013423] usb 4-2.2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  369.013427] usb 4-2.2: Product: CX3-UVC
[  369.013431] usb 4-2.2: Manufacturer: Cypress
[  369.015010] uvcvideo: Found UVC 1.10 device CX3-UVC (8086:0ad0)
[  369.418074] usb 4-2.1: new SuperSpeed USB device number 4 using xhci_hcd
[  369.439397] usb 4-2.1: New USB device found, idVendor=8086, idProduct=0acb
[  369.439404] usb 4-2.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  369.439408] usb 4-2.1: Product: Intel RealSense 3D Camera ZR300
[  369.439412] usb 4-2.1: Manufacturer: Intel Corp
[  369.439415] usb 4-2.1: SerialNumber: SN_3481801072
[  369.441317] uvcvideo: Found UVC 1.10 device Intel RealSense 3D Camera ZR300 (8086:0acb)
[  369.445501] uvcvideo: Found UVC 1.10 device Intel RealSense 3D Camera ZR300 (8086:0acb)
[  369.448067] uvcvideo: Unable to create debugfs 4-4 directory.
[  369.449491] uvcvideo: Found UVC 1.10 device Intel RealSense 3D Camera ZR300 (8086:0acb)
[  369.452066] uvcvideo: Unable to create debugfs 4-4 directory.
[  369.452352] input: Intel RealSense 3D Camera ZR300 as /devices/pci0000:00/0000:00:1c.6/0000:0f:00.0/usb4/4-2/4-2.1/4-2.1:1.4/input/input18

```

And after launching zr300_nodelet_rgbd.launch

```
[  480.909780] uvcvideo: Failed to query (SET_CUR) UVC control 1 on unit 7: -32 (exp. 2).
[  480.960397] uvcvideo: Failed to query (SET_CUR) UVC control 1 on unit 7: -32 (exp. 2).
[  481.011041] uvcvideo: Failed to query (SET_CUR) UVC control 1 on unit 7: -32 (exp. 2).
[  481.085243] uvcvideo: Failed to query (SET_CUR) UVC control 5 on unit 2: -32 (exp. 4).
[  481.136152] uvcvideo: Failed to query (SET_CUR) UVC control 5 on unit 2: -32 (exp. 4).
[  481.186908] uvcvideo: Failed to query (SET_CUR) UVC control 5 on unit 2: -32 (exp. 4).
[  481.237599] uvcvideo: Failed to query (SET_CUR) UVC control 5 on unit 2: -32 (exp. 4).
[  481.288742] uvcvideo: Failed to query (SET_CUR) UVC control 5 on unit 2: -32 (exp. 4).
[  481.339420] uvcvideo: Failed to query (SET_CUR) UVC control 5 on unit 2: -32 (exp. 4).
[  481.390232] uvcvideo: Failed to query (SET_CUR) UVC control 5 on unit 2: -32 (exp. 4).
[  481.441259] uvcvideo: Failed to query (SET_CUR) UVC control 5 on unit 2: -32 (exp. 4).
[  481.492104] uvcvideo: Failed to query (SET_CUR) UVC control 5 on unit 2: -32 (exp. 4).
[  481.542776] uvcvideo: Failed to query (SET_CUR) UVC control 5 on unit 2: -32 (exp. 4).
[  481.593636] uvcvideo: Failed to query (SET_CUR) UVC control 5 on unit 2: -32 (exp. 4).
[  481.644426] uvcvideo: Failed to query (SET_CUR) UVC control 5 on unit 2: -32 (exp. 4).
[  481.695275] uvcvideo: Failed to query (SET_CUR) UVC control 5 on unit 2: -32 (exp. 4).
[  481.745991] uvcvideo: Failed to query (SET_CUR) UVC control 5 on unit 2: -32 (exp. 4).
[  481.796773] uvcvideo: Failed to query (SET_CUR) UVC control 5 on unit 2: -32 (exp. 4).
[  481.847566] uvcvideo: Failed to query (SET_CUR) UVC control 5 on unit 2: -32 (exp. 4).
[  481.898306] uvcvideo: Failed to query (SET_CUR) UVC control 5 on unit 2: -32 (exp. 4).
[  481.949186] uvcvideo: Failed to query (SET_CUR) UVC control 5 on unit 2: -32 (exp. 4).
[  482.000037] uvcvideo: Failed to query (SET_CUR) UVC control 5 on unit 2: -32 (exp. 4).
[  482.050701] uvcvideo: Failed to query (SET_CUR) UVC control 5 on unit 2: -32 (exp. 4).
[  482.101588] uvcvideo: Failed to query (SET_CUR) UVC control 5 on unit 2: -32 (exp. 4).
[  482.103151] uvcvideo: Failed to query (SET_CUR) UVC control 5 on unit 2: -32 (exp. 4).
[  482.153803] uvcvideo: Failed to query (SET_CUR) UVC control 5 on unit 2: -32 (exp. 4).
[  482.204796] uvcvideo: Failed to query (SET_CUR) UVC control 5 on unit 2: -32 (exp. 4).
[  482.255827] uvcvideo: Failed to query (SET_CUR) UVC control 5 on unit 2: -32 (exp. 4).
[  482.306536] uvcvideo: Failed to query (SET_CUR) UVC control 5 on unit 2: -32 (exp. 4).
...
...
```

### Steps to Reproduce

To install librealsense, I followed the steps highlighted in this tutorial : http://wiki.ros.org/librealsense#Installation_Prerequisites 

Then I cloned and compiled the realsense_camera ros package.

ashishraste, 2017-05-26:
Hi @zubinp .. I think the feature of registering RGB to depth information isn't available with ZR300 based nodelets. I am looking for the same information to be published and I see that R200/SR300 (via something like realsense_r200_rgbd.launch) has this information.

zubinp, 2017-05-26:
@ashishraste there is a zr300_nodelet_rgbd.launch file in this repository. I even got it to work in one of my colleague's laptop. This issue is probably hardware related. My computer's usb 3 might have an older firmware version. I'm still investigating the issue and might post back here once I zero in on the exact issue.

ashishraste, 2017-05-27:
@zubinp Thanks for pointing it out :+1: I was using an old ROS distributed realsense_camera package and missed out.
USB3 did create some issues in the beginning. I assume you'd have tried externally powered USB3 adapter, if not it might work.

closed 2018-04-06 by jlgarcia75