realsenseai/realsense-ros #242: SR300 point cloud displays weirdly on rviz

dastez00, 2017-06-06:
### System Configuration
|Version | Your Configuration | 
|:----------|:----------| 
|Operating System | Ubuntu 14.04.5 LTS|
|Kernel | 4.4.0-78-generic|
|ROS | indigo| 
|ROS RealSense | 1.8.0|
|librealsense | 1.12.1|
|Camera Type | Intel RealSense SR300|
|Camera Firmware | 3.21.0.0|

### Expected Behavior
Displaying the registered XYZRGB points stream from topic /camera/depth_registered/points on rviz.

### Actual Behavior
The displayed point cloud is globally very flat in the direction of the optical axis, excepted some parts that pop nonsensically out of the cloud. Moreover, there seems to be 2 "symmetrical" point clouds overlapping.

[image]
[image]
[image]
[image]

### Steps to Reproduce
Tab 1:
`roslaunch realsense_camera sr300_nodelet_rgbd.launch`
Tab 2:
`rosrun tf2_ros static_transform_publisher 0 0 5 0 0 0 world camera_link`
Tab 3:
`rosrun rviz rviz`
In rviz:
- Select "world" as Fixed Frame
- Add a display by topic -> /camera/depth_registered/points (type PointCloud2)
- Set the Decay Time to 0.25 at least (otherwise for me the displayed point cloud flickers a lot)

To solve this problem, I have already:
- tested the camera with the SDK samples provided by Intel on Windows (e.g. 3D scan) and everything is working quite well.
- updated to the last firmware (even though this may have been a bad idea, but the problem was already there before).
- displayed pretty much every other topic provided from the camera. The point cloud without RGB (/camera/depth/points) gives me the same result.
- changed the depth_fps and color_fps to other possible values (following this link: https://github.com/IntelRealSense/librealsense/blob/master/doc/supported_video_formats.pdf) via a custom launch file, with no change. Sometimes the fps does not even follow my values.
- Something noticeable is that I am unable to display any ir image (it says "no image" or displays a black image)

Is there something that I am missing ?
Does the point cloud need some other kind of transformation before being displayed on rviz ?
Or is it simply a graphical glitch, which has nothing to do with the actual data ?
Would that be a better idea then to treat raw depths and colors separately via another library ?

Actually my goal is just to be sure that the data sent by this topic are ok, before continuing post-processing it with the Point Cloud Library. I indeed need 6D (XYZ + RGB) points to perform recognition tasks.

johngardenier, 2017-06-14:
I solved this issue by increasing the depth fps to 30. At 10 fps the point cloud would sometimes be flat like you describe, sometimes it would jump around. Note that a flickering point cloud and depth image is solved by installing the uvcvideo driver kernel patches.

RaphSchru, 2017-06-14:
Hi John, thanks for your reply.

Both my depth_fps and rgb_fps were already set to 30 by default. I've tried to lower it to 10 fps by changing the arguments in a custom launch file, with no change for the shape of my point cloud.

Apart from that, in the installation instructions of librealsense (http://wiki.ros.org/librealsense#uvcvideo_Driver_Patches),
it says very briefly that a patch to uvcvideo is needed for other cameras than the R200.
Does someone know how I can download/install that patch ? (I've already done the Xenial kernel update as suggested).

Thanks.

closed 2017-06-17 by dastez00