realsenseai/realsense-ros #243: RGBD stream stopping after 1hour with increasing ram requirement

arpit15, 2017-06-08:
### System Configuration
Please complete Your Configuration detail below. Refer to the BKC at [Configuration](../realsense_camera/README.md#configuration).

| Version               | Your Configuration   |
|:--------------------- |:---------------------|
| Operating System      | ubuntu 14.04                |
| Kernel                | 4.4.0-66-generic                |
| ROS                   | Indigo                  |
| ROS RealSense         | 1.8.0               |
| librealsense          | 1.12.1               |
| Camera Type-Firmware  | SR300 3.15.0.0            |

---

### Expected Behavior
The depth stream from topic /camera/depth_registered/points is obtained at all times. 

### Actual Behavior
The ram requirements of the process keeps increasing until it reaches to 95% of the total ram which in my case is 32GB. After eating up the ram the sensor stops publishing the topic. 
I have seen following messages after the node is shutdown

1. rs_wait_for_frames
2. [camera/camera_nodelet_manager-1] process has died [pid 10166, exit code -9, cmd /opt/ros/indigo/lib/nodelet/nodelet manager __name:=camera_nodelet_manager 
3. Without any message
I could receive more 100k frames in rviz before the node crashes.
I can't understand why the requirement of memory keeps increasing as the nodes don't store anything while processing input data. 
Kindly let me know what could be a way to run the sensor for longer durations.

### Steps to Reproduce
roslaunch realsense_camera sr300_nodelet_rgbd.launch

Thank you
Arpit

citay, 2017-07-30:
I tried to reproduce it on my HP Z220 Core-i7 with Ubuntu 16.04 Kernel 4.4.0-77.
I used 'htop' to monitor the RAM usage.
It ran more than 3 hours without increasing the RAM.

arpit15, 2017-07-30:
@citay  is it possible for you to try out on 14.04? I have also had crashes for other reasons. I am also facing calibration errors between rgb and depth images with sr300. Therefore for now I have given up on this camera. I will try again on 16.04 ubuntu and let you know if I am facing the same problem.

closed 2018-01-31 by icarpis