realsenseai/realsense-ros #247: Once stopped, cannot find the camera again

nlyubova, 2017-06-13:
### System Configuration

| Version               | Your Configuration   |
|:--------------------- |:---------------------|
| Operating System      | Ubuntu 16.04                |
| Kernel                | 4.4.15-tegra on NVIDIA Jetson TX2              |
| ROS                   | Kinetic                    |
| ROS RealSense         | 1.8.0                |
| librealsense          | 1.12.1                |
| Camera Type-Firmware  | ZR300 2.0.71.28, R200, SR300           |

### Expected Behavior
ROS launch file should be able to start any time without disconnecting the camera from USB 

### Actual Behavior
Once the ROS node is started and stopped, we cannot start it again (it does not find the camera) without disconnecting the camera from USB port and connecting it again. Tested with ZR300, R200, SR300, all cameras show the same behavior. Does it mean that the camera does not stop correctly? 

### Steps to Reproduce
1) start the launch file
roslaunch realsense_camera zr300_nodelet_rgdb.launch
2) stop it
Ctrl+C
3) start again and it does not find the camera anymore
roslaunch realsense_camera zr300_nodelet_rgdb.launch
[ERROR] [1497343628.338504027]: /camera/driver - No cameras detected!
[ERROR] [1497343628.338619424]: /camera/driver - Error calling rs_create_context ( api_version:11201 ): 
UVCIOC_CTRL_QUERY:UVC_SET_CUR error 5, Input/output error

Please, help to fix it.

citay, 2017-07-23:
R200 and ZR300 have a known firmware limitation, not properly handling abnormal exit flow.
SR300 seem to work fine on my HP Z220 Core-i7 with Ubuntu 16.04, Kernel 4.4.0-77-generic.
I will try to repeat the experiment on the Jetson. Did you see this behavior with SR300 on another systems?

dniewinski, 2017-09-29:
I'm seeing exactly the same issue with 2 R200s and Ubuntu 14.04.5 LTS.  First launch works, subsequent launches fail saying there are no cameras.

icarpis, 2017-10-31:
Hello @dniewinski, please see my previous comment regarding R200 known firmware limitation.

closed 2017-10-31 by icarpis