realsenseai/realsense-ros #249: Build fails on Mac OS X

5yler, 2017-06-17:
### System Configuration

| Version               | Method |
|:--------------------- |:------------ |
| Operating System      | `Mac OS X 10.9.5` |
| Kernel                | `13.4.01` |
| ROS                   | `indigo` |
| ROS RealSense         | `1.8.0` |
| librealsense          | `1.12.0` |
---

### Expected Behavior

Using `catkin_make` would successfully compile the package. I successfully installed `librealsense` in the ROS environment beforehand.

### Actual Behavior

```
21:29 𝜓 ~/realsense_ws catkin_make
Base path: ~/realsense_ws
Source space: ~/realsense_ws/src
Build space: ~/realsense_ws/build
Devel space: ~/realsense_ws/devel
Install space: ~/realsense_ws/install
####
#### Running command: "cmake ~/realsense_ws/src -DCATKIN_DEVEL_PREFIX=~/realsense_ws/devel -DCMAKE_INSTALL_PREFIX=~/realsense_ws/install -G Unix Makefiles" in "~/realsense_ws/build"
####
-- The C compiler identification is AppleClang 6.0.0.6000057
-- The CXX compiler identification is AppleClang 6.0.0.6000057
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using CATKIN_DEVEL_PREFIX: ~/realsense_ws/devel
-- Using CMAKE_PREFIX_PATH: /usr/local/Cellar;/opt/ros/indigo;~/indigo_desktop_full_ws/devel
-- This workspace overlays: /opt/ros/indigo
-- Found PythonInterp: /usr/local/bin/python (found version "2.7.5") 
-- Using PYTHON_EXECUTABLE: /usr/local/bin/python
-- Using default Python package layout
-- Found PY_em: /usr/local/lib/python2.7/site-packages/em.pyc  
-- Using empy: /usr/local/lib/python2.7/site-packages/em.pyc
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: ~/realsense_ws/build/test_results
-- Found gtest: gtests will be built
-- Using Python nosetests: /usr/local/bin/nosetests-2.7
-- catkin 0.6.18
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 1 packages in topological order:
-- ~~  - realsense_camera
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'realsense_camera'
-- ==> add_subdirectory(realsense/realsense_camera)
-- Using these message generators: gencpp;genlisp;genpy
-- realsense_camera: 1 messages, 5 services
-- Configuring done
-- Generating done
-- Build files have been written to: ~/realsense_ws/build
####
#### Running command: "make -j4 -l4" in "~/realsense_ws/build"
####
[... 185 lines cut ...]
clang: error: unknown argument: '-z'
clang: error: unknown argument: '-z'
clang: error: unknown argument: '-z'
clang: error: no such file or directory: 'noexecstack'
clang: error: no such file or directory: 'relro'
clang: error: no such file or directory: 'now'
make[2]: *** [~/realsense_ws/devel/lib/realsense_camera/tests_camera_core] Error 1
make[1]: *** [realsense/realsense_camera/CMakeFiles/tests_camera_core.dir/all] Error 2
make: *** [all] Error 2
make: INTERNAL: Exiting with 5 jobserver tokens available; should be 4!
Invoking "make -j4 -l4" failed
```

### Steps to Reproduce

```
mkdir -p ~/realsense_ws/src
cd ~/realsense_ws/src
catkin_init_workspace
git clone https://github.com/intel-ros/realsense.git
git checkout git checkout -b stable-branch stable
cd ~/realsense_ws
catkin_make
```
(This is the easy part, getting ROS installed on a Mac is the hard part!)

5yler, 2017-06-17:
Got it to compile by commenting out the `CMAKE_EXE_LINKER_FLAGS` in `CMakeLists.txt` and removing `execvpe(argv[0], argv, environ);` from `base_nodelet.cpp`. Tested and working with SR300 camera:
[image]

icarpis, 2017-12-27:
Hello @5yler, thank you for your helping. I'm glad to hear that it works on OSX.
I'll add this fix in the next release.

closed 2019-03-26 by RealSenseCustomerSupport