realsenseai/realsense-ros #253: Use image encoding 'mono8' instead of '8UC1'

NikolausDemmel, 2017-07-11:
The camera driver publishes image messages for the grayscale fisheye camera of the ZR300 with encoding "8UC1". While this is not wrong, there is a more specific encoding typically used, namely "mono8" which not only implies that the data is encoded like "8UC1", but also, that it contains a grayscale ("monochromatic") image of intensities (as opposed to for example depth values).

Having the driver publish "8UC1" instead of "mono8" means that the ROS opencv bridge conversion functions choke on it if you say something like

```
img_data = np.array(CVB.imgmsg_to_cv2(msg, desired_encoding="mono8"))
```

See https://github.com/ros-perception/vision_opencv/issues/175#issuecomment-314099859 for a discussion of this.

Therefore I propose to change the driver to publish images for the fisheye camera with encoding "mono8". I would also suggest to do the same for the IR cameras, since those are also intensity images and the [Realssense encoding is "intensity" anyway](https://github.com/intel-ros/realsense/blob/92083cf8772ddc34e455e47cdcb771a72eba18d7/realsense_camera/src/zr300_nodelet.cpp#L59-L82).

If the maintainers agree to this change, I can also provide a PR.

See also http://docs.ros.org/jade/api/sensor_msgs/html/image__encodings_8h_source.html

artifactz, 2018-03-29:
Any news on this?
`cv_bridge` conversion of IR images still doesn't work straightaway...

pavloblindnology, 2018-08-09:
Ping. It seems so easy to fix to save tons of users from forking the repo to solve such child problems.

ibaranov-cp, 2019-03-04:
:+1:  This is a minor issue that would make a serious Quality of life improvement for users.

doronhi, 2019-03-05:
@NikolausDemmel ,
I would like to have your PR if it's not too much trouble.
Thanks

NikolausDemmel, 2019-03-05:
I haven't been using this for a while and I don't think I'll get round to providing a PR right now, sorry, but it should boil down to replacing one string ("8UC1") with another ("mono8"), possibly in multiple places.

doronhi, 2019-03-05:
Thanks. I'll find the time and handle it. It sounds like a good idea.

ibaranov-cp, 2019-03-06:
Did a pull request here :)

https://github.com/intel-ros/realsense/pull/656

RealSenseCustomerSupport, 2019-03-07:
[Please Ignore - RealSense system comment]

eirikaso, 2019-03-12:
I have recorded rosbag files that use 8UC1. It would be a real hassle having to record them again. Does anyone have a workaround that I can use on already recorded video?

ibaranov-cp, 2019-03-12:
An easy way would be to play back the rosbag, use a python subscriber to listen to the data, modify the encoding string, and publish it back out again :)

Relevant links:
http://wiki.ros.org/ROS/Tutorials/WritingPublisherSubscriber%28python%29
http://docs.ros.org/melodic/api/sensor_msgs/html/msg/Image.html

NikolausDemmel, 2019-03-12:
Even easier might be writing a simple python script that directly manipulates the messages in the bag, see http://wiki.ros.org/rosbag/Cookbook . Note that it will loose information about latched topics, but so will replaying.

doronhi, 2019-03-13:
Thank you @ibaranov-cp and all of you guys.
I merged the PR. It's available in the development branch and will be included in the next release so I close this issue now.

closed 2019-03-13 by doronhi

eirikaso, 2019-03-13:
> An easy way would be to play back the rosbag, use a python subscriber to listen to the data, modify the encoding string, and publish it back out again :)
>
> Relevant links: [...]

Thank you. For my use case it worked to just remove the 2nd argument (desired_encoding). I will keep your solution in mind if I need it in the future