realsenseai/realsense-ros #254: questions on realsense SR300, examples & can't launch under ROS indigo

Thompson104, 2017-07-13:
### System Configuration
Please complete Your Configuration detail below. Refer to the BKC at [Configuration](../realsense_camera/README.md#configuration).

| Version               | Your Configuration   |
|:--------------------- |:---------------------|
| Operating System      | Ubuntu 14.04               |
| Kernel                | 4.4.0-81-generic                |
| ROS                   | indigo                  |
| ROS RealSense         | SR300                |
| librealsense          | I just installed it               |
| Camera Type-Firmware  | 3.15.0.0        |

---

---

Hey all, 

I have two questions for you guys and hope you can help me with them. I'm a graduate student at mechanical engineering. I have been working with ROS for more than a year. But camera is new to me.

1. I wonder how can you run tutorials under librealsense/examples? I don't know how to do this in linux...I feel terrible for me programming skills. Is there like a tutorial on this? I tried to google it but nothing useful comes up. 

2. I tried `roslaunch realsense_camera sr300_nodelet_rgbd.launch
`
and 
`roslaunch realsense_camera sr300_nodelet_default.launch
`
But it gets stuck on Setting dynamic camera options, 

> [ INFO] [1499915780.552897441]: /camera/driver - Setting dynamic camera options

and nothing happens after that for a long time. I wonder what may cause this problem? 

I took computer vision this semester but I haven't had the chance to use PCL with real camera, I wonder if you could kindly share some useful links? I'm the only one working in my lab on this part. I would really appreciate any help sincerely.

### Steps to Reproduce

omershalev, 2017-07-27:
I don't think this question is related to realsense-ROS but rather to librealsense.
Please refer to the [librealsene](https://github.com/IntelRealSense/librealsense) repository.
Specifically, follow the [installation guide](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md). You don't have to use QTCreator. Make sure to run cmake with `DBUILD_EXAMPLES=true`.
Then you can simply launch any of the examples - the executable files are located under `/usr/local/bin`.

dorodnic, 2017-08-08:
Hi @Thompson104 
Did you managed to get the examples running? 
I'm closing the issue for now, but please re-open it if you have any additional questions (ROS or librealsense related). We will be happy to assist :)

closed 2017-08-08 by dorodnic