realsenseai/realsense-ros #256: questions about imu information of ZR300, how to calibrate the raw imu data

cza1019, 2017-07-21:
### System Configuration
| Version               | Your Configuration   |
|:--------------------- |:---------------------|
| Operating System      | Ubuntu 14.04.5              |
| Kernel                | 4.4.0-79-generic              |
| ROS                   | indigo                   |
| ROS RealSense         | latest              |
| librealsense          | latest               |
| Camera Type-Firmware  | latest           |

I am trying to use ZR300 to run VINS-Mono(https://github.com/HKUST-Aerial-Robotics/VINS-Mono). I find that the output of the imu date topic:/camera/imu/data_raw has not been calibrated. 
Then I get the calibration information of IMU data from another ZR300 ROS driver(https://github.com/IntelRealSense/realsense_samples_ros/tree/kinetic-devel/realsense_ros_camera), the output of topic /camera/accel/imu_info is like this: 
data: [1.0236904621124268, 0.0, 0.0, -0.0455622673034668, 0.0, 1.0147671699523926, 0.0, 0.14746201038360596, 0.0, 0.0, 0.9979372620582581, -0.41241949796676636]
noise_variances: [0.14709974825382233, 0.14709974825382233, 0.14709974825382233]
bias_variances: [9.999999747378752e-05, 9.999999747378752e-05, 9.999999747378752e-05]

I think this 12-dimensions matrix is the calibration info. It can be rewritten like this:
data: [  1.0236904621124268, 0.0, 0.0, -0.0455622673034668, 
            0.0, 1.0147671699523926, 0.0, 0.14746201038360596, 
            0.0, 0.0, 0.9979372620582581, -0.41241949796676636]
It looks like a Homogeneous transformation matrix. So I use it to calibrate the accel data[accel_x, accel_y,accel_z]'.  I'm not sure how to do it.  I try a reasonable way:  
calibrated_accel_x= (accel_x-(-0.0455622673034668))*1.0236904621124268
the other two axes are calibrated in the same way.   But it doesn't work. Actually, I have tried all possible combinations to calibrate the raw data. None of them provide a reasonable result.

So is this calibration information of IMU reliable? Is there any SDK to calibrate the raw imu data?

icarpis, 2017-09-13:
Hi @cza1019 , this matrix is the IMU calibration info. the values are as follows:

    /* Scale X        cross axis        cross axis      Bias X */
    /* cross axis     Scale Y           cross axis      Bias Y */
    /* cross axis     cross axis        Scale Z         Bias Z */

Only the scale and bias values should used to calibrate the accel/gyro data. the rest of values should hold zeros.
Please use the following formula to calibrate the gyro and accel values
> calibrated_accel_x = (raw_accel_x) * (Scale X) - (Bias X)
this formula is valid also for the other axes.

caoxudong0513, 2017-09-21:
请问你解决了吗？

BruceLun, 2017-12-28:
It seems simple enough to _get_ the parameter information, but how is it set? Just pass yaml parameters when the camera nodlet is started?

icarpis, 2017-12-28:
The calibration parameters are used to fix the gyro/accel data manually by the formula above.

BruceLun, 2017-12-28:
My question is how to change the calibration  values in the matrix mentioned above.

icarpis, 2017-12-28:
You cannot change the calibration values. These values calculated during the calibration process and written into the camera.

BruceLun, 2017-12-29:
So the "calibration process" is performed at the factory and hard-coded. The published raw IMU data is un-calibrated and the only way to use it is to have a node subscribe the raw IMU data, and re-publish it applying the calibration values?

icarpis, 2017-12-29:
Yes, exactly.

closed 2018-04-06 by jlgarcia75