realsenseai/realsense-ros #257: R200 nodelet stuck on "/camera/driver - Setting dynamic camera options"

whyou5945, 2017-07-23:
### System Configuration
Please complete Your Configuration detail below. Refer to the BKC at [Configuration](../realsense_camera/README.md#configuration).

| Version               | Your Configuration   |
|:--------------------- |:---------------------|
| Operating System      |    Ubuntu 14.04.5 LTS        |
| Kernel                |       4.4.0-87-generic      |
| ROS                   |         indigo        |
| ROS RealSense         |      1.8.0          |
| librealsense          |    1.12.1            |
| Camera Type-Firmware  |  Intel RealSense R200, Camera FW: 1.0.71.01       |

---

### Expected Behavior
The RealSense ROS node should successfully initialize the camera and startup.

### Actual Behavior
I just install the Ros-realsense from sources according to http://wiki.ros.org/realsense_camera/Tutorials/Building_from_Sources. But at the final step, I run roslaunch realsense_camera r200_nodelet_default.launch, It was stuck on [ INFO] [1500801034.375389904]: /camera/driver - Setting dynamic camera options.
I have googled this and find the similar problem on issue#194 https://github.com/intel-ros/realsense/issues/194 . It seems it was solved, but I still don't know how to fix it.
### Steps to Reproduce
roslaunch realsense_camera r200_nodelet_default.launch

omershalev, 2017-07-27:
Can you check if you see anything on the topics?
First launch r200_nodelet_default.launch. In another terminal: first execute `rostopic list`, then choose one of the camera topics and execute `rostopic echo <name of camera topic>`.
If you see the data printed to the console it means that realsense node works as expected.

whyou5945, 2017-07-30:
@omershalev Thank you very much! I tried what u said, then I can see many dynamic data. It seems it works as expected. But i still can't see anything by launch the rviz. I will try to write an subscriber to check it. 
Anyway, thank you for your help!

omershalev, 2017-07-30:
You can find rviz tutorials in [http://wiki.ros.org/rviz/Tutorials](http://wiki.ros.org/rviz/Tutorials)

whyou5945, 2017-07-31:
Thank you very much!

closed 2017-08-08 by dorodnic